espressif/esp-zigbee-sdk #495: How can I send data of type int32 or larger from esp32-h2 module to another esp32-he module

mehmetbozhaydar, 2024-11-28:
### Question

I am new to esp-idf. Which library should I use to send int32 type data with Zigbee?

xieqinan, 2024-11-29:
Yes, if you expect to use the Zigbee to achieve your purpose, I think you can refer to the [docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_custom.html) first.

mehmetbozhaydar, 2024-11-30:
@xieqinan I looked at the documentation. I tried to create a custom cluster but I get the error below.
How can I solve it please help

`I (392) app_start: Starting scheduler on CPU0
I (397) main_task: Started on CPU0
I (397) main_task: Calling app_main()
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42012cde  RA      : 0x420120e2  SP      : 0x40810e90  GP      : 0x4080a410
--- 0x42012cde: zb_bufpool_storage_allocate at ??:?
0x420120e2: allocate_buffer_unsafe at zb_bufpool_mult.c.obj:?

TP      : 0x40805828  T0      : 0x4000af8c  T1      : 0xffffffe0  T2      : 0x00010401
--- 0x40805828: xQueueReceiveFromISR at C:/Users/Haydar/Documents/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2182
0x4000af8c: multi_heap_internal_unlock in ROM

S0/FP   : 0x00000001  S1      : 0x4080be20  A0      : 0x00000000  A1      : 0x00000050
A2      : 0x4080be20  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000001  A7      : 0x00000002  S2      : 0x00000000  S3      : 0x40810ef8
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000001  T5      : 0x4084dc98  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000
--- 0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
40810e90: 0x00000040 0x4084dc54 0x00000001 0x420120e2 0x4084dcf0 0x4084dc6c 0x00000000 0x40810ef8      
--- 0x420120e2: allocate_buffer_unsafe at zb_bufpool_mult.c.obj:?

40810eb0: 0x00000000 0x00000000 0x40810ef8 0x42012176 0x4084daa8 0x0000ffff 0x4201b650 0x420121ce      
--- 0x42012176: process_delayed_buf_entry_unsafe at zb_bufpool_mult.c.obj:?
0x4201b650: zb_zcl_init_periodic_activities at zcl_common.c.obj:?
0x420121ce: zb_get_buf_delayed at zb_bufpool_mult.c.obj:?

40810ed0: 0x00030003 0x4084dcf0 0x40840003 0x00000000 0x00000000 0x4080e000 0x4201b650 0x420122d0      
--- 0x4201b650: zb_zcl_init_periodic_activities at zcl_common.c.obj:?
0x420122d0: zb_buf_get_out_delayed_func at ??:?

40810ef0: 0x00000000 0x00000000 0x4201b650 0x00010000 0x00000000 0x00000000 0x4084dca8 0x42021012      
--- 0x4201b650: zb_zcl_init_periodic_activities at zcl_common.c.obj:?
0x42021012: zb_af_register_device_ctx at ??:?

40810f10: 0x00000000 0x4080e000 0x4084dca8 0x4200a51e 0x00000000 0x00000000 0x00000000 0x00000000      
--- 0x4200a51e: esp_zb_device_register at ??:?

40810f30: 0x00000000 0x4080e000 0x4080e000 0x42008e06 0x00000000 0x0001e240 0xf2010401 0x000000ff      
--- 0x42008e06: app_main at C:/Users/Haydar/Music/HA_on_off_light/main/server.c:142

40810f50: 0x42043218 0x00000000 0x42043000 0x420392e2 0x00000000 0x00001388 0x00000001 0x00000000      
--- 0x420392e2: main_task at C:/Users/Haydar/Documents/esp/v5.1/esp-idf/components/freertos/app_startup.c:209 (discriminator 13)

40810f70: 0x00000000 0x00000000 0x00000000 0x40806e10 0x00000000 0x00000000 0x00000000 0x00000000      
--- 0x40806e10: vPortTaskWrapper at C:/Users/Haydar/Documents/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

40810f90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40810fb0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x40810f10 0x4080e124 0x4080b360 0x4080b360      
40810fd0: 0x40810fc0 0x4080b358 0x00000018 0xfdbf57db 0xd1b2afd8 0x40810fc0 0x00000000 0x00000001      
40810ff0: 0x4080ffbc 0x6e69616d 0xf5f9f100 0x7fbffaab 0x001a007f 0x00000000 0x40810fb0 0x00000001      
40811010: 0x00000000 0x00000000 0x00000000 0x00000000 0x4080eac4 0x4080eb2c 0x4080eb94 0x00000000      
40811030: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x40011484 0x00000000 0x00000000      
--- 0x40011484: _cleanup_r in ROM

40811050: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000      
40811070: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000      
40811090: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408110b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000      
408110d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000      
[... 154 lines cut ...]

// Custom request handler
static esp_err_t zb_custom_request_handler(const esp_zb_zcl_custom_cluster_command_message_t *message) {
    ESP_LOGI(TAG, "Received custom command");
    ESP_LOGI(TAG, "Payload size: %d", message->data.size);

    if (message->data.size == sizeof(uint32_t)) {
        uint32_t received_value;
        memcpy(&received_value, message->data.value, sizeof(uint32_t));
        ESP_LOGI(TAG, "Received int32 value: %ld", received_value);
    }
    return ESP_OK;
}

// General action handler
static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message) {
    switch (callback_id) {
    case ESP_ZB_CORE_CMD_CUSTOM_CLUSTER_REQ_CB_ID:
        return zb_custom_request_handler((esp_zb_zcl_custom_cluster_command_message_t *)message);
    default:
        ESP_LOGW(TAG, "Unhandled callback ID: 0x%x", callback_id);
        return ESP_OK;
    }
}
`

xieqinan, 2024-12-02:
Hi,

Here has an [example](https://github.com/espressif/esp-zigbee-sdk/issues/202#issuecomment-2005690377) maybe meet your requirement, could you please also refer to it?

mehmetbozhaydar, 2024-12-02:
@xieqinan  hi Thank you, it helped a lot. You are my life saver.

closed 2024-12-02 by mehmetbozhaydar