espressif/esp-zigbee-sdk #496: Level cluster attribute 'CurrentLevel' limited to values from 0 to 200

OOHehir, 2024-11-28:
### Question

I'm using a ESP32-C6 as an end-device, library versions 1.6.

I've been using the level cluster (0x0008) & have noticed that the CurrentLevel (identifier 0x000) is limited to values from 0 to 200 (inclusive). I've tried with 2 different coordinators. 

It seems standard practice for Zigbee devices as far as I can tell, but undocumented. Perhaps it could be set at compile time or documented?

xieqinan, 2024-11-29:
@OOHehir,

> It seems standard practice for Zigbee devices as far as I can tell, but undocumented. Perhaps it could be set at compile time or documented?

I don’t think so. The esp-zigbee-sdk does not impose any limitations on the `current level`. Is the `current level` controlled by the coordinator? If so, could you share how you are using the coordinator?

OOHehir, 2024-12-09:
@xieqinan I'll try and come up with a MRE to demonstrate

OOHehir, 2024-12-10:
@xieqinan You're quite correct, my mistake. In my code I'd set ESP_ZB_ZCL_ATTR_LEVEL_CONTROL_MAX_LEVEL_ID & that was the issue. Closing, thanks.

closed 2024-12-10 by OOHehir