espressif/esp-zigbee-sdk #497: Getting time from Coordinator

asbachb, 2024-12-02:
### Question

Hello,

I'm currently try to get the time from my coordinator (Sonoff ZBDonge-E).

After the network steering I try to request the time:

```c
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGI(TAG, "Device rebooted");
                
                esp_zb_gasmeter_request_time();
            }
        } else {
            /* commissioning failed */
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
        }
        break;
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK) {
            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            ESP_LOGI(TAG, "Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",
                     extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                     extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel(), esp_zb_get_short_address());
                     
            esp_zb_gasmeter_request_time();
        } else {
            ESP_LOGI(TAG, "Network steering was not successful (status: %s)", esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;
```

```c
void esp_zb_gasmeter_request_time() {
	ESP_LOGI(TAG, "Requesting time from coordinator...");

    uint16_t attributes[] = 
    {
        ESP_ZB_ZCL_ATTR_TIME_LOCAL_TIME_ID
    };

    esp_zb_zcl_read_attr_cmd_t read_req;
    read_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    read_req.attr_field = attributes;
    read_req.attr_number = sizeof(attributes) / sizeof(uint16_t);
    read_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TIME;
    read_req.zcl_basic_cmd.dst_endpoint = 1;
    read_req.zcl_basic_cmd.src_endpoint = 1;
    read_req.zcl_basic_cmd.dst_addr_u.addr_short = 0x0000;  //coordinator
    esp_zb_zcl_read_attr_cmd_req(&read_req);
}
```

But actually nothing is happening after sending the request:

```
I (505) ESP_ZB_GASMETER: Recived Zigbee signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (505) ESP_ZB_GASMETER: Recived Zigbee signal: ZDO Skip Start Up (0x1), status: ESP_OK
I (505) ESP_ZB_GASMETER: Initialize Zigbee stack
I (2915) ESP_ZB_GASMETER: Recived Zigbee signal: BDB Device Reboot (0x6), status: ESP_OK
I (2915) ESP_ZB_GASMETER: Device started up in non factory-reset mode
I (2915) ESP_ZB_GASMETER: Device rebooted
I (2925) ESP_ZB_GASMETER: Requesting time from coordinator...
```

Is that the correct way to the time from the zigbee network?

xieqinan, 2024-12-02:
@asbachb 

> Is that the correct way to fetch the time from the zigbee network?

I think so. Have you registered the `zb_action_handler` using `esp_zb_core_action_handler_register()`? please refer to the [link](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L288). If this is already done, could you check whether the `read_response` command has been sent by the coordinator?

asbachb, 2024-12-02:
> `esp_zb_core_action_handler_register`

Yes. That's part of my initialization code:
```c
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();

   ...

   static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message) {
    ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
``` 

It seems I don't get a reply at all. Where should I expect the `read_response` command?

xieqinan, 2024-12-03:
> It seems I don't get a reply at all. Where should I expect the read_response command?

It is better to use the [Zigbee sniffer](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark) to capture it from the air. If the coordinator without response anything, the `zb_action_handler()` will not be triggered.

asbachb, 2025-05-30:
> Hi [@asbachb](https://github.com/asbachb) ,
> 
> Any updates?

Not really. I tried to flash a sonoff stick as sniffer but was not successful. So currently I'm not able to sniff the traffic.