espressif/esp-zigbee-sdk #498: ED_KEEP_ALIVE gpio and ulp_lp_core_wakeup_main_processor();

t1gerCo2, 2024-12-06:
### Is your feature request related to a problem?

today I use ED_KEEP_ALIVE every 10 minutes to  update data produced from ULP i2c sensor, and the main cpu is waken up from light sleep. it would be more battery efficient to wake upp the main zigbee loop from the ulp_lp_core_wakeup_main_processor(); only when the data is changed not every time intervall.
 

### Describe the solution you'd like.

If i understand correctly, the main process can be woken with either timer(ED_KEEP_ALIVE) or gpio, i guess adding ulp_lp_core_wakeup_main_processor();  would be a minor change if i understand correctly , but would do a lot of good to reduce power consumption

chshu, 2024-12-24:
It's doable, please refer to the example code here: https://github.com/espressif/esp-idf/blob/master/examples/system/ulp/lp_core/lp_i2c/main/lp_core/main.c#L39.

Note that not all of the ESP SoCs support ULP, among the Zigbee enabled SoCs, ESP32-C6 supports ULP, but ESP32-H2 doesn't.

chshu, 2024-12-24:
Closing.. Please file new issues under https://github.com/espressif/esp-idf/issues, if you have any follow up question on ULP usage.

closed 2024-12-24 by chshu