espressif/esp-zigbee-sdk #499: Persistent config

cosmo84, 2024-12-09:
### Question

Hello,
I have played with the "All Device Types App Example" and successfully created the device I need. When I reboot my ESP32, it does not reconnect to the network and so I assume that all configuration that is done within the CLI is lost after reboot or at least another start of the zigbee stack is needed.
Is there any way to make the config persistent?

Best,
Stephan

xieqinan, 2024-12-10:
@cosmo84 ,

> at least another start of the zigbee stack is needed.

Yes, you can input the `bdb_comm start steer` to make the CLI device rejoin.

cosmo84, 2024-12-10:
I tried to do it this way, by adding these lines to the **esp_zigbee_all_device_types_app.c** file:

`bool factory_new = esp_zb_bdb_is_factory_new(); `
`if (!factory_new) { `
`        ESP_LOGI(TAG, "Device is not factory new. Starting Zigbee network steering."); `
`        ESP_ERROR_CHECK(esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING)); `
`}`
`else { `
`        ESP_LOGI(TAG, "Device is factory new.");`
`}`

But every time I reboot the device (after it was successfully joined to a network) it is reported as factory new.

xieqinan, 2024-12-10:
Where are these codes placed? Please follow the steps below to test:

- Device A
```
factoryreset
role zc
channel 11
network type c
bdb_comm start form
```

- Device B
```
factoryreset
role zed
channel 11
network type c
bdb_comm start steer
reboot
bdb_comm start steer
```
The below log will be presented `ESP_ZB_CONSOLE_APP: Device started up in non factory-reset mode`

cosmo84, 2024-12-10:
Okay, that made it work. Thanks.

closed 2024-12-10 by cosmo84