espressif/esp-zigbee-sdk #500: esp_zb_basic_cluster_add_attr void* value_p should be const void*?

damian-kurek-wizzdev, 2024-12-11:
### Question

When using C++ compiler it complains that it can not convert const char* to void*
error: invalid conversion from 'const void*' to 'void*' [-fpermissive]
I'm not 100% sure but I think that adding const void* to this parameter should fix the issue 
const_cast to char* and then using it lets the compiler do the job, but than I will get static analysis warning to not use const_cast

xieqinan, 2024-12-12:
This issue arises because C++ is more strict about type conversions than C. Specifically, converting from `const char*` to `void*` without an explicit cast is not allowed in C++. Here's an analysis of your situation and some potential solutions:

---

### **Root Cause**
- The error occurs because `const char*` (a pointer to a constant character array) cannot be implicitly converted to `void*` (a generic pointer type) in C++.
- In C++, implicit conversions that drop the `const` qualifier are not permitted, as they violate const-correctness.

---

### **Options to Fix the Issue**

#### **1. Update the Function Prototype**
If the function can accept a `const void*`, modify its parameter type to preserve const-correctness.

Example:
```cpp
void myFunction(const void* data);
```

This change is preferred because it does not compromise const-correctness and avoids the need for casting. 

#### **2. Use `const_cast` Carefully**
If you must call a function that does not accept `const void*` (e.g., a legacy C API), you can use `const_cast` to remove the `const` qualifier explicitly.

Example:
```cpp
const char* str = "example";
void* ptr = const_cast<char*>(str);
myFunction(ptr); // Pass the casted pointer
```

However:
- This should only be done if you are sure the function does not modify the data.
- It may trigger static analysis warnings, as you noted.

#### **3. Modify the Called Function (If Possible)**
If you have access to the implementation of the function that requires `void*`, update it to take a `const void*` instead. 

Example:
Before:
```cpp
void myFunction(void* data);
```

After:
```cpp
void myFunction(const void* data);
```

This is ideal for maintaining const-correctness across the codebase.

#### **4. Suppress Compiler Errors with `-fpermissive` (Not Recommended)**
You can use the `-fpermissive` compiler flag to relax the type-checking rules. However, this is not a good practice because it may lead to unintended behavior or bugs.

---

damian-kurek-wizzdev, 2024-12-12:
These functions are inside esp-zigbee lib esp_zigbee_attribute.h

xieqinan, 2024-12-12:
Could you please give an example ?

xieqinan, 2024-12-24:
@damian-kurek-wizzdev ，

Thank you for your explanation. I don’t think this is a blocking issue for you. We already have plans to optimize the parameters for the SDK's API, but this will take some time. In the meantime, could you please use the above workaround for testing?

By the way, could you share why you prefer using a `const *` type as the parameter for `esp_zb_basic_cluster_add_attr()`? We can confirm that this function only copies `value_p` to the stack without modifying it. If possible, please provide more background on your reasoning and share how you're using it in your project, including a code example.

damian-kurek-wizzdev, 2024-12-30:
I want tot store all default values as constexpr const value for example it includes module identifier 

constexpr const char* MODEL_IDENTIFIER = "\x10"
                                         "Panel gen4"; /* Customized model identifier */
                                         
esp_zb_basic_cluster_add_attr(
        pBasicCluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, const_cast<char*>(MODEL_IDENTIFIER));