espressif/esp-zigbee-sdk #502: Zigbee Scene switch and Zigbee 2-Gang (2 Level Clusters endpoints) Dimmer Switch.

mundevx, 2024-12-12:
### Question

For 4-Gang switch, when i add all 4 lights in scene switch via app, it is working fine, i am getting EP1, EP2, EP3, EP4, i.e 4 times scene store callback with different endpoints and on scene devices are added and scene store command is working OK, Turning all ON or OFF as per settings in app.

But if add level control cluster with on off cluster, the behavior distracted.

I have an issue with 2-Gang Dimmer switch. While adding dimmer switch with EP1 & EP2 ON in scene switch via app, the app gateway sends scene store request to zigbee end device. On scene store callback, I am getting 2 times scene store callback for EP1 and EP1 (Same Endpoints). which is wrong, because it must have to receive EP1 & EP2 on callback.

My both end points are working fine individually for dimmer switch and can control through app. But scene switch  is dealing only with endpoint1 in case of dimmer switch.

Please help me regarding this issue.

mundevx, 2024-12-13:
Hi @xieqinan , 
Can you provide any link for creating zigbee sniffer?

xieqinan, 2024-12-16:
Please refer to: https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark

VIinhQuang, 2025-01-17:
Hi, I am working with a Tuya multi-gang Zigbee switch, but I am unable to control the individual gang switches independently. When I attempt to control endpoint 1 (EP1) or endpoint 2 (EP2), both gangs are toggled together (all on or all off). This issue persists even when I send commands specifically to one endpoint. Are you have the solution? many thanks

cheldonec, 2025-03-15:
@VIinhQuang Try my experiment, its work fine [https://github.com/cheldonec/esp_tuya_zigbee](url)

mundevx, 2025-03-17:
Hi @cheldonec ,

your link [https://github.com/cheldonec/esp_tuya_zigbee](https://github.com/espressif/esp-zigbee-sdk/issues/url) is not working and redirects me to https://github.com/espressif/esp-zigbee-sdk/issues/created_by/url.

cheldonec, 2025-03-17:
Hi! I corrected the link.