espressif/esp-zigbee-sdk #503: ESP as Zigbee range extender

cosmo84, 2024-12-12:
### Is your feature request related to a problem?

No

### Describe the solution you'd like.

Using an ESP as Zigbee range extender. I walked through the various examples and I can do lights, switches, thermostat and many more, but no range extender. Configurung it as light or switch would work too, but it's unneeded functionality.

### Describe alternatives you've considered.

Using Ikea Tradfri signal repeater, but I prefer hand-made solutions :-)

xieqinan, 2024-12-16:
Hi @cosmo84 ,

Are you looking for a simple Zigbee router that doesn't include any clusters and only acts as a repeater? If so, you can use the example provided at [esp_zigbee_all_device_types_app](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_all_device_types_app) to create one manually by inputting specific command-line instructions.

cosmo84, 2024-12-16:
Hi @xieqinan 

I was looking into this and it works in general. It is more my idea to have the 100% solution instead of 80/20 ;-)

In the all device example the range extender type is missing, so I can do it with a switch or a light instead:

[image]

What I am looking for is rather this:

[image]

xieqinan, 2024-12-16:
@cosmo84,

If so, I think you can refer to the below code to change the [dimmable light example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_color_dimmable_light) to get the range extender device;
```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZR_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = HA_COLOR_DIMMABLE_LIGHT_ENDPOINT,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_RANGE_EXTENDER_DEVICE_ID,
        .app_device_version = 0,
    };
    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(cluster_list, esp_zb_basic_cluster_create(NULL),
                                          ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster_create(NULL),
                                             ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    zcl_basic_manufacturer_info_t info = {
        .manufacturer_name = ESP_MANUFACTURER_NAME,
        .model_identifier = ESP_MODEL_IDENTIFIER,
    };

    esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config);
    esp_zcl_utility_add_ep_basic_manufacturer_info(ep_list, HA_COLOR_DIMMABLE_LIGHT_ENDPOINT, &info);
    esp_zb_device_register(ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
}
```

cosmo84, 2024-12-16:
Thanks, I will give it a try and let you know.

cosmo84, 2024-12-17:
@xieqinan 

It works perfectly, many thanks.

closed 2024-12-17 by cosmo84