espressif/esp-zigbee-sdk #504: Add void* user data to all callbacks

damian-kurek-wizzdev, 2024-12-16:
### Is your feature request related to a problem?

I want to pass an object instance that can process incoming calls instead of creating global objects

### Describe the solution you'd like.

static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void* message, **void* userData**)
user data shall be passed when registering a callback 
void esp_zb_core_action_handler_register(esp_zb_core_action_callback_t cb, **void* userData**);

xieqinan, 2024-12-24:
Hi @damian-kurek-wizzdev，

Could you provide a more detailed and concrete application case? I believe the feature doesn’t make much sense. The `esp_zb_core_action_handler_register()` serves as the interface between ZCL and the application. It is triggered by active behaviors (such as a read attribute request that triggers a read attribute response) and passive behaviors (such as a report attribute from a binding device).  Typically, it only needs to be registered once before the stack starts. Even if the `void *userData` parameter is added, the stack would only maintain the pointer and trigger it whenever any `callback_id` is activated, which is effectively the same as creating a global object in the application.

damian-kurek-wizzdev, 2024-12-30:
It is the same functionality if you create a global instance and call functions, but adding void* userData is the standard solution for libraries that use callbacks including esp idf.
That is a nice-to-have feature and won't break anything.
Examples from esp-idf
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/gpio.html#:~:text=esp_err_t%20gpio_isr_register(,%EF%83%81
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/mqtt.html#_CPPv430esp_mqtt_client_register_event24esp_mqtt_client_handle_t19esp_mqtt_event_id_t19esp_event_handler_tPv