espressif/esp-zigbee-sdk #509: zb_config_platform.h missing ifdef cplusplus extern C

damian-kurek-wizzdev, 2024-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.5

### esp-zigbee-lib version.

1.6.0

### esp-zboss-lib version.

1.6.0

### Espressif SoC revision.

ESP32C6

### What is the expected behavior?

All functions in header should be inside 

### What is the actual behavior?

#ifdef __cplusplus
extern "C" {}
#endif

### Steps to reproduce.

include file in cpp project

xieqinan, 2024-12-19:
Could you please add `extern "C"` for this header when using it in your project? Alternatively, use `esp_zigbee_core.h` instead of `zb_config_platform.h`.

damian-kurek-wizzdev, 2024-12-19:
Done that and reported the issue

chshu, 2024-12-20:
The `cplusplus extern C` is missing in most ZBOSS header files, and we don't expect our users to include ZBOSS header files in their applications, but should use the esp-zigbee APIs instead.

closed 2025-02-08 by chshu