espressif/esp-zigbee-sdk #510: esp_zb_zcl_scenes_table_store does only work sometimes

Haerteleric, 2024-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4 

### esp-zigbee-lib version.

1.6.1

### esp-zboss-lib version.

1.6.1

### Espressif SoC revision.

ESP-C6

### What is the expected behavior?

Trying to implement scene tables but only some entrys get a Extension Fields safed. As the function that fill the Extension Fields is always the same it should work for all groupID/sceneID combinations.

### What is the actual behavior?

+-------+----------+----------+-----------------+-------------------------------------+
| Index | Group ID | Scene ID | Transition Time |   [Cluster ID | Extension Field]   |
+-------+----------+----------+-----------------+-------------------------------------+
|   00  |  0x0003  |   0x01   |      0000       |
+-------+----------+----------+-----------------+-------------------------------------+
|   01  |  0x0003  |   0x02   |      0000       | [ 0x0300 | 0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x54 ] | [ 0x0008 | 0xfe ] | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   02  |  0x0003  |   0x03   |      0000       | [ 0x0300 | 0x28  0xbc  0xd6  0x43  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x54 ] | [ 0x0008 | 0xfe ] | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   03  |  0x0003  |   0x04   |      0000       |
+-------+----------+----------+-----------------+-------------------------------------+

### Steps to reproduce.

this is how i fill the table on ESP_ZB_CORE_SCENES_STORE_SCENE_CB_ID Action:

```c
//Structs used for storing Scenes
typedef union onOffExtensionField_u
{
   struct onOffExtensionField_s
   {
        uint8_t onOff;
   }fields;

   uint8_t raw[1];
}onOffExtensionField_t;

typedef union levelControlExtensionField_u
{
   struct levelControlExtensionField_s
   {
        uint8_t level;
   }fields;

   uint8_t raw[1]; 
}levelControlExtensionField_t;

typedef union colorControlExtensionField_u
{
   struct colorControlExtensionField_s
   {
        uint16_t x;
        uint16_t y;
        uint16_t hue;
        uint8_t saturation;
        uint8_t colorLoopActive;
        uint8_t colorLoopDirection;
        uint16_t colorLoopTime;
        uint16_t colorTemperature;
   }fields;

   uint8_t raw[13];
}colorControlExtensionField_t;

static void storeEndpointStateToScene(uint8_t epId, uint8_t sceneId, uint16_t groupId)
{
    printf("Store Scene %i for Group %i\n", sceneId, groupId);
    flsEndpoint_t * ep = NULL;
    endpointManager_getEndpointById(epId, &ep);

    bool colorEnabledEp = ((ep->capabilities.reg & ENDPOINT_CAPABILITIES_COLORTEMP) == ENDPOINT_CAPABILITIES_COLORTEMP)
        || ((ep->capabilities.reg & ENDPOINT_CAPABILITIES_RGB) == ENDPOINT_CAPABILITIES_RGB);

    onOffExtensionField_t onOffExtensionField = {0};
    levelControlExtensionField_t levelControlExtensionField = {0};
    colorControlExtensionField_t colorControlExtensionField = {0};

    esp_zb_zcl_scenes_extension_field_t onOffClusterData = {
        .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
        .length = 1,
        .extension_field_attribute_value_list = onOffExtensionField.raw,
        .next = NULL
    };

    esp_zb_zcl_scenes_extension_field_t levelControlClusterData = {
        .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_LEVEL_CONTROL,
        .length = 1,
        .extension_field_attribute_value_list = levelControlExtensionField.raw,
        .next = &onOffClusterData
    };

    esp_zb_zcl_scenes_extension_field_t colorControlClusterData = {
        .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
        .length = 13,
        .extension_field_attribute_value_list = colorControlExtensionField.raw,
        .next = &levelControlClusterData
    };

    if(esp_zb_lock_acquire(portMAX_DELAY))
    {
        if(colorEnabledEp)    
        {
            uint8_t mode = *(uint8_t *)(esp_zb_zcl_get_attribute(
                    epId, 
                    ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
                    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                    ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_MODE_ID
                )->data_p);
            
            switch (mode)
            {
            case 0x00:
                colorControlExtensionField.fields.hue = (*(uint8_t *)(esp_zb_zcl_get_attribute(
                    epId, 
                    ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
                    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                    ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_HUE_ID
                )->data_p)) << 8; //Hue is 8 bit, so shift it to the left by 8 bits to get the Enhanced Hue

                colorControlExtensionField.fields.saturation = *(uint8_t *)(esp_zb_zcl_get_attribute(
                    epId, 
                    ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
                    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                    ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_SATURATION_ID
                )->data_p);
                break;

            case 0x01:
                colorControlExtensionField.fields.x = *(uint16_t *)(esp_zb_zcl_get_attribute(
                    epId, 
                    ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
                    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                    ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_X_ID
                )->data_p);

                colorControlExtensionField.fields.y = *(uint16_t *)(esp_zb_zcl_get_attribute(
                    epId, 
                    ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
                    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                    ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_Y_ID
                )->data_p);
            
            case 0x02:
                colorControlExtensionField.fields.colorTemperature = *(uint16_t *)(esp_zb_zcl_get_attribute(
                    epId, 
                    ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
                    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                    ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_TEMPERATURE_ID
                )->data_p);
                break;
            
            default:
                break;
            }
        }

        levelControlExtensionField.fields.level = *(uint8_t *)(esp_zb_zcl_get_attribute(
            epId, 
            ESP_ZB_ZCL_CLUSTER_ID_LEVEL_CONTROL,
            ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
            ESP_ZB_ZCL_ATTR_LEVEL_CONTROL_CURRENT_LEVEL_ID
        )->data_p);

        onOffExtensionField.fields.onOff = *(bool *)(esp_zb_zcl_get_attribute(
            epId, 
            ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
            ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
            ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID
        )->data_p);
        
        printf("EP[%i]: Store Scene %i for Group %i\n", epId, sceneId, groupId);
        esp_err_t err = esp_zb_zcl_scenes_table_store(epId, sceneId, groupId, 0x0000, colorEnabledEp ? &colorControlClusterData : &levelControlClusterData);
        printf("Status: %s\n",esp_err_to_name(err));

        esp_zb_lock_release();
    }
}
```

Haerteleric, 2024-12-19:
```
+-------+----------+----------+-----------------+-------------------------------------+
| Index | Group ID | Scene ID | Transition Time |   [Cluster ID | Extension Field]   |
+-------+----------+----------+-----------------+-------------------------------------+
|   00  |  0x0003  |   0x01   |      0000       |
+-------+----------+----------+-----------------+-------------------------------------+
|   01  |  0x0003  |   0x02   |      0000       | [ 0x0300 | 0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x54 ] | [ 0x0008 | 0xfe ] | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   02  |  0x0003  |   0x03   |      0000       | [ 0x0300 | 0x28  0xbc  0xd6  0x43  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x54 ] | [ 0x0008 | 0xfe ] | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   03  |  0x0003  |   0x04   |      0000       |
+-------+----------+----------+-----------------+-------------------------------------+
Store Scene 2 for Group 3
EP[1]: Store Scene 2 for Group 3
Status: ESP_OK
+-------+----------+----------+-----------------+-------------------------------------+
| Index | Group ID | Scene ID | Transition Time |   [Cluster ID | Extension Field]   |
+-------+----------+----------+-----------------+-------------------------------------+
|   00  |  0x0003  |   0x01   |      0000       |
+-------+----------+----------+-----------------+-------------------------------------+
|   01  |  0x0003  |   0x02   |      0000       |
+-------+----------+----------+-----------------+-------------------------------------+
|   02  |  0x0003  |   0x03   |      0000       | [ 0x0300 | 0x28  0xbc  0xd6  0x43  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x0  0x54 ] | [ 0x0008 | 0xfe ] | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   03  |  0x0003  |   0x04   |      0000       |
+-------+----------+----------+-----------------+-------------------------------------+
```

Haerteleric, 2024-12-19:
some more observations:

  - The first Scene Table entry failes almost always.
  - The Entrys after succed but can't be updated (calling store again)
  
Open Question:
could that be that the corresponding funcs arn't in iram?

xieqinan, 2024-12-26:
@Haerteleric 

>  esp_err_t err = esp_zb_zcl_scenes_table_store(epId, sceneId, groupId, 0x0000, colorEnabledEp ? &colorControlClusterData : &levelControlClusterData);

Suggestion
```
esp_err_t err = esp_zb_zcl_scenes_table_store(epId, groupId, sceneId, 0x0000, colorEnabledEp ? &colorControlClusterData : &levelControlClusterData);
```

Haerteleric, 2025-01-06:
Ah merde. You are right i messed up the argument order...

Thanks !

closed 2025-01-06 by Haerteleric