espressif/esp-zigbee-sdk #511: Zigbee Custom Attribute & Custom Cluster

mundevx, 2024-12-19:
### Question

I am using these lines to define custom cluster with attribute:

esp_zb_attribute_list_t *esp_zb_custom_cluster = esp_zb_zcl_attr_list_create(0xFC00);
esp_zb_custom_cluster_add_custom_attr(esp_zb_custom_cluster, 0, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &dst_addr);
esp_zb_cluster_list_add_custom_cluster(cluster_list, esp_zb_custom_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

But home assistant not showing attribute (0).
[image]

This issue is also raised in https://github.com/espressif/esp-zigbee-sdk/issues/365

mundevx, 2024-12-21:
Hi,

Any update?

xieqinan, 2024-12-24:
Hi,

I believe this is an issue with Home Assistant. Since the cluster is customized, HA cannot determine its attributes. HA only queries the cluster information but does not inquire about its attributes. As a result, as you observed, there are no attribute options available for selection. I believe the same behavior would occur with other Zigbee stacks. Therefore, I recommend raising an issue with Home Assistant to seek a solution.

closed 2025-03-20 by chshu