espressif/esp-zigbee-sdk #513: Add an action to react on bind/unbind requests

theorlangur, 2024-12-22:
### Is your feature request related to a problem?

I didn't find a way of how to get and react to events of binding/unbinding to some other devices. I'd expect some dedicated `esp_zb_core_action_callback_id_t` for that. Right now in some of my projects I have to periodically check the current binds of myself by sending a `esp_zb_zdo_binding_table_req`.
Is there a better way? 
Could there be?

### Describe the solution you'd like.

I'd expect to have another 2 callback id enum values in `esp_zb_core_action_callback_id_t` like, say, `ESP_ZB_CORE_BIND_REQ_CB_ID` and `ESP_ZB_CORE_UNBIND_REQ_CB_ID` with dedicated structs containing all needed info. 

### Describe alternatives you've considered.

Atm I'm periodically executing requests to 'myself' to get the binding table.

xieqinan, 2024-12-23:
Hi @theorlangur,

I understand your requirement. By default, the stack manages and maintains binding/unbinding requests with callbacks in the Zigbee APS layer. We can add a binding/unbinding action callback if it is necessary and meaningful. Assuming the callback already exists, what actions do you intend to perform in the callback? Why do you want to receive binding requests from other devices in application layer?

theorlangur, 2024-12-23:
@xieqinan , I have this presence sensor based on `esp32h2` (`ld2412` + `PIR`). It's not used standalone but in a zigbee network managed by `zigbee2mqtt`. One of the features I've implemented for it is a support for a direct binding. Naturally the low-level support of the binding is done by your SDK, but I've created a client `on/off` cluster to send `on/off` commands to bound devices directly. In order for presence sensor to be 'smarter' it would be beneficial to have a logic that reacts to states of the device (devices) that it controls: if the light was manually turned off via other means (e.g. a button, Home Assistant, ...), you'd want the presence sensor to send a new `on` command at least on `PIR` detection (it typically reacts and clears much faster than mmWave `ld2412`).

For my presence sensor to receive reports from the light I'm controlling on the on/off attribute state, the lamp has to be bound to the presence sensor. I could mandate that for this to work you **have to bind explicitly** the lamp to the presence 
sensor in `zigbee2mqtt` (or via other external means) and it would work, but a **more convenient workflow** would be if: 
* the presence sensor would take care of that automatically as soon as it detects that it was bound to control some other device with an `on/off` cluster (the presence sensor would send a `bind` request to the controlled device, configure reporting for `on/off` attribute). 
* consequentually, when you unbind the lamp from the presence sensor it would also do the unbinding in the other direction (instruct the lamp to `unbind` from it not to send any more reports).

Another alternative could be: 
I could send the `on` command always on `PIR` motion detection. The downside would be: it's yet another radio communication that's not needed most of the time, but would unnecessarily 'polute' the medium. 

I hope it makes sense.
Thanks!

theorlangur, 2024-12-23:
@xieqinan , to additionally clarify: my intention is not to be able to alter the existing (un)binding low-level logic, but to merely react to events, be notified that `bind` or `unbind` has happened (with the details of course, like an address, a cluster, an endpoint)

xieqinan, 2024-12-23:
@theorlangur ，

Thank you for your explaination.

> the presence sensor would take care of that automatically as soon as it detects that it was bound to control some other device with an on/off cluster (the presence sensor would send a bind request to the controlled device, configure reporting for on/off attribute).

I’m a bit confused about this step. As far as I understand, `zigbee2mqtt` acts as the network manager and should handle the binding and unbinding configuration. It is used to bind the `presence sensor` to the `lamp` so that the presence sensor can send commands to the `lamp` directly. 

However, for the `lamp` to report its state, it is necessary to send a `bind_req` and `config_report` command to the `lamp`. Why are these commands sent by the presence sensor instead of `zigbee2mqtt`? I believe these commands should be sent by the network manager. 

If that’s the case, when `zigbee2mqtt` unbinds the `lamp` from the `presence sensor`, the `unbind_req` and `config_report` commands can also be sent by it. This would make the network configuration clearer and more unified.

How do you think about it?

theorlangur, 2024-12-23:
@xieqinan, thanks for taking time and reviewing this

You're right that `zigbee2mqtt` should be able to send a `bind_req` and `config_report` commands to the `lamp` and it is able to do it. In other words, as I also mentioned in my original post, you could explicitly issue those commands from `zigbee2mqtt`, however this:

1.  is an explicit additional action you as a user have to take
2.  presence sensor would somehow have to know that the `lamp` (or several lamps, I wouldn't want to limit this only to 1 device) is bound to it and is supposed to send reports about its state to the presence sensor in order for sensor to account for it in its logic.

I wanted to limit the required 'configuration' of the presence sensor only to binding the presence sensor to 1 or more `lamps` and the sensor would do the rest (attempt to bind those lamps to itself, configure reporting, etc.) and this would be an optional thing (if, say, binding or report configuring would not be succesfull), but the key point here would be: the presence sensor would know about it and could act accordingly.

Do you see this kind of functionality as superfluous?

theorlangur, 2024-12-26:
@xieqinan , thanks for the response and for the pointers
indeed I was able to get the notifications I wanted with the help of `esp_zb_aps_data_indication_handler_register`, although I must say it's a bit cumbersome to go that route.
Or am I doing something wrong?
What worked for me is:
on the callback of data indication, I checked:
1. `esp_zb_apsde_data_ind_t`.`dst_short_addr` is me (`esp_zb_get_short_address()`)
2. apparently, `esp_zb_apsde_data_ind_t`.`cluster_id` contains the APS command id: `0x21` for `bind` and `0x22` for `unbind` (btw, where are descriptions of such APS command identifiers? I cannot find them anywhere...)
3. first byte in `esp_zb_apsde_data_ind_t`.`asdu` I'm skipping as it seems to be some sort of a counter
4. `esp_zb_apsde_data_ind_t`.`asdu + 1` is basically as described in `APSME-BIND.request` of a ZigBee spec.

Is that what you meant?

theorlangur, 2024-12-26:
I guess I'm closing the issue. Apparently I'm not getting a convenient way to get notified about bind-unbind operations, but at least I was pointed towards a "good enough" lower-level approach to get what I want. 
@xieqinan , thank you for your time!

closed 2024-12-26 by theorlangur