espressif/esp-zigbee-sdk #514: Update On Off attribute value doesn't work

grebenyuk51, 2024-12-25:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### esp-zigbee-lib version.

1.6.1

### esp-zboss-lib version.

1.6.1

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

esp_zb_zcl_set_attribute_val updates value and sends report (if configured) about changed status

### What is the actual behavior?

After call esp_zb_zcl_set_attribute_val neither value changed nor report genereted

### Steps to reproduce.

1. Create simple device with ESP_ZB_ZCL_CLUSTER_ID_ON_OFF and ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID attribute
2. Connect device to coordinator. Create binding and configure reporting of the on_off attr to coordinator
3. Set value of on_off attr via coordinator (works fine)
4. Set value of on_off attr via device (no report about attr value change)
5. Read value of on_off attr via coordinator (reads not actual value)

Code for change value on device:
```
	gpio_set_level(RELAY_IO, state);
	esp_zb_lock_acquire(portMAX_DELAY);
	esp_zb_zcl_set_attribute_val(SW_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_ON_OFF, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID, &state, false);
	esp_zb_lock_release();
```

Same issue described on forum: http://esp32.io/viewtopic.php?t=41047

xieqinan, 2024-12-26:
Hi @grebenyuk51 ,

Could you please refer to the [temperature sensor example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor/main/esp_zb_temperature_sensor.c#L63) which is similar with your project?

grebenyuk51, 2024-12-26:
in that example there is no ESP_ZB_ZCL_CLUSTER_ID_ON_OFF. Thus, no ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID. Or you want to tell me, that I have to manage value of ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID and do reporting like in example?

But why I can only change ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID via esp_zb_zcl_set_attribute_val, and zigbee will do all magic stuff under hood?

xieqinan, 2024-12-27:
> Or you want to tell me, that I have to manage value of ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID and do reporting like in example?

What I mean is that the SDK can only provide typical examples to demonstrate API usage rather than covering every possible case. The temperature sensor and thermostat example share the similar application logic with your use case, differing only in the cluster. You can review it to understand the runtime logic of attribute setting and reporting, and then use it as a foundation to design or modify your application.

grebenyuk51, 2025-01-05:
@xieqinan, wait a minute. In temperature measurement cluster I don't need to manage reporting. All I need is to change value
`esp_zb_zcl_set_attribute_val(MAIN_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &temperature, false);`
SDK will manage all reporting stuff under the hood. Furthermore, SDK will report for every reporting endpoint (if I set binding and reporting to several devices)

But! OnOff cluster doesn't have such magic. If this is expected behavior, I will close this issue

xieqinan, 2025-01-09:
@grebenyuk51 ,

> SDK will manage all reporting stuff under the hood. Furthermore, SDK will report for every reporting endpoint (if I set binding and reporting to several devices)

The esp-zigbee-sdk aims to provide flexible APIs, allowing users to develop applications based on their unique requirements. As a result, the SDK does not manage all reporting functionalities internally. However, if you are interested, you can integrate the provided APIs to implement this functionality.

asward, 2025-01-19:
It would be great if it was knowable what aspects the SDK does/does not manage. I have also had trouble getting reporting to work for anything other than the specific examples provided.

xieqinan, 2025-01-20:
@asward ,

The APIs of the esp-zigbee-sdk primarily adhere to the Zigbee specification (R22), aiming to enable developers to design their applications seamlessly. If any behaviors or implementations of the SDK deviate from the Zigbee specification, please feel free to raise an issue. We will address it promptly by providing a fix or offering a reasonable explanation.

closed 2025-03-20 by chshu