espressif/esp-zigbee-sdk #515: HA_temperature_sensor example not updating HomeAssistant

ear9mrn, 2024-12-29:
### Question

I have installed esp-idf and compiled the HA_Temperature_sensor example and in general is works fine connecting to home assistant using zigbee2mqtt. The only issue is that the temperature is not automatically updating. The temperature is exposed but I have to hit the "refresh" button to get the latest temperature so show up. Not much use if you want to track temperature over time.
 
[image]

As I am using this to get proof of concept and to try and understand a little better how zigbee works using esp32-h2 and esp32-c6 I do not have a temperature sensor so I have it generate a random number inside the temp_sensor_driver_value_update function with the following.

`tsens_value = rand()%10+20;` 
 I do not think this would have any impact on the main code to update the home assistant.

 with the esp_app_temp_sensor_handler function I place a log output as follows:

 `ESP_EARLY_LOGI(TAG, "Send 'Temp change (%d)...' command", measured_value/100);`

Which show the value "sent" every 10 seconds. 

On the zigbee2mqtt I am not seeing anything in the logs showing any messages arriving. I do see log output when I hit the refresh button.

`
[2024-12-29 19:21:00] debug: 	zh:ember: ~~~> [ZCL to=0x4831b7fffec03a92:33424 apsFrame={"profileId":260,"clusterId":1026,"sourceEndpoint":1,"destinationEndpoint":10,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":61,"commandIdentifier":0}]
[2024-12-29 19:21:00] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=DIRECT], [indexOrDestination=33424], [apsFrame={"profileId":260,"clusterId":1026,"sourceEndpoint":1,"destinationEndpoint":10,"options":4416,"groupId":0,"sequence":250}], [messageTag=59]
[2024-12-29 19:21:00] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":1026,"sourceEndpoint":10,"destinationEndpoint":1,"options":3392,"groupId":0,"sequence":205}], [packetInfo:{"senderShortId":33424,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":172,"lastHopRssi":-68,"lastHopTimestamp":0}], [messageContents=183d01000000298c0a]
[2024-12-29 19:21:00] debug: 	zh:controller: Received payload: clusterID=1026, address=33424, groupID=0, endpoint=10, destinationEndpoint=1, wasBroadcast=false, linkQuality=172, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":61,"commandIdentifier":1},"payload":[{"attrId":0,"status":0,"dataType":41,"attrData":2700}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}

` 

So to me it looks like the "message" that should be sent from this function is not being sent for some reason. 

`
static void esp_app_temp_sensor_handler(float temperature)
{
    int16_t measured_value = zb_temperature_to_s16(temperature);
    /* Update temperature sensor measured value */
    esp_zb_lock_acquire(portMAX_DELAY);
    esp_zb_zcl_set_attribute_val(HA_ESP_SENSOR_ENDPOINT,
        ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &measured_value, false);
    esp_zb_lock_release();
    ESP_EARLY_LOGI(TAG, "Send 'Temp change (%d)...' command", measured_value/100);
}

`
Apologies as I am relatively new to this API and looking for some help to try and figure out what is happening. It does feel like it is an issue on the esp32-h2 side and not on the zigbee2mqtt side.

Thanks.

ear9mrn, 2024-12-29:
small update, add the following:

`

esp_zb_zcl_status_t status;
status = esp_zb_zcl_set_attribute_val.....
ESP_EARLY_LOGI(TAG, "Status: %d, Status code (hex): 0x%02X", status, status);
` 

`
static void esp_app_temp_sensor_handler(float temperature)
{
    int16_t measured_value = zb_temperature_to_s16(temperature);
    esp_zb_zcl_status_t status;
    /* Update temperature sensor measured value */
    esp_zb_lock_acquire(portMAX_DELAY);
    status = esp_zb_zcl_set_attribute_val(HA_ESP_SENSOR_ENDPOINT,
        ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &measured_value, true);
    esp_zb_lock_release();
    
    ESP_EARLY_LOGI(TAG, "Send 'Temp change (%d)...' command", measured_value/100);
    ESP_EARLY_LOGI(TAG, "Status: %d, Status code (hex): 0x%02X", status, status);
}
`

with the following response.
`
I (152807) ESP_ZB_TEMP_SENSOR: Send 'Temp change (23)...' command
I (152807) ESP_ZB_TEMP_SENSOR: Status: 136, Status code (hex): 0x88
`
scanning through the header files looks like the following error (read only).....

`
ESP_ZB_ZCL_STATUS_READ_ONLY             = 0x88U,     /*!< Read only */

`

xieqinan, 2024-12-30:
@ear9mrn ,
> status = esp_zb_zcl_set_attribute_val(HA_ESP_SENSOR_ENDPOINT,
ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &measured_value, true);

Suggestion
```
status = esp_zb_zcl_set_attribute_val(HA_ESP_SENSOR_ENDPOINT,
ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &measured_value, false);
```

Could you please share the actual log for the message `"Send 'Temp change (%d)...' command"`? Please note that in the example, the delta value for the temperature sensor's `report info` is set to 100. This means a report will only be generated when the change in temperature exceeds the delta value. Please double-check it.

ear9mrn, 2024-12-30:
made the changes you suggest above including setting the delta value to 0. Still no success, I am not seeing anything show up in the zigbee2mqtt logs when the message appears to be sent from the esp32-h2 

When I hit the refresh button (get command) within zigbee2mqtt it does respond (correctly) and the temperature is updated.

`
[2024-12-30 12:52:48] debug: 	z2m: Received Zigbee message from '0x4831b7fffec03a92', type 'readResponse', cluster 'msTemperatureMeasurement', data '{"measuredValue":2100}' from endpoint 15 with groupID 0
`

xieqinan, 2024-12-31:
Hi,

Could you please refer to the [example README](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_thermostat/README.md) without making any changes in application to confirm that the reporting mechanism works? This will help in addressing the issue.  

By the way, if you have made application changes or are experiencing failures when joining other platforms, I suggest providing a `.pcap` file captured by a sniffer, as it would be more helpful for troubleshooting.

ear9mrn, 2024-12-31:
> Hi,
>
> Could you please refer to the [example README](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_thermostat/README.md) without making any changes in [...]

I can confirm using the setup described in the README.md that the temperature is reported the thermostat device correctly via zigbee.

ear9mrn, 2024-12-31:
now using wireshark to sniff the packets. The first image below is from an Aqara Temp/Press/Humidity sensor.

[image]

and here is a packet from ESP32-H2 with the temperature example installed. 

It does not appear to be recognized as a Zigbee packet by wireshark. Maybe this is why it is not showing up in the zigbee2mqtt logs.

[image]

ear9mrn, 2025-01-01:
> now using wireshark to sniff the packets. The first image below is from an Aqara Temp/Press/Humidity sensor.
> 
> ![image](https://private-user-images.githubusercontent.com/471260/399514597-c35ea3ad-1ab8-4be0-bc4d-130730d64514.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3NjI3ODQsIm5iZiI6MTczNTc2MjQ4NCwicGF0aCI6Ii80NzEyNjAvMzk5NTE0NTk3LWMzNWVhM2FkLTFhYjgtNGJlMC1iYzRkLTEzMDczMGQ2NDUxNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwMVQyMDE0NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lODQ5Nzc3MjBlNzE1ZjNlOGIxYzg5OWQzMjYzZGNkZjM3N2NkNTRmMDI3YTE3NmIyZDZmZGRhNjIzYzA4N2EzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.KXoKBZmkNfFRWJdt1Id2amrLsfKAFk5tso3sl-Xu3po)
> 
> and here is a packet from ESP32-H2 with the temperature example installed. Although it is encrypted. I am wondering, maybe it is using the wrong key to encrypt the message that is sent to zigbee2mqtt?
> 
> ![image](https://private-user-images.githubusercontent.com/471260/399556375-e7e8d445-2dca-4852-8a28-3ebd9773bb35.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3NjI3ODQsIm5iZiI6MTczNTc2MjQ4NCwicGF0aCI6Ii80NzEyNjAvMzk5NTU2Mzc1LWU3ZThkNDQ1LTJkY2EtNDg1Mi04YTI4LTNlYmQ5NzczYmIzNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwMVQyMDE0NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZGMzMjEyNzA3ZjNlODY3MTNmZDQ4ZmQ5NDU5NzdjNWY3Njk4MjBkODg3OTBkZTAzOGY1NTE5ZTU0Y2U0OGRiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.yriUyoiB1XMyyqyHhM3XmGAtAd1dQVayi4L2b3w-MSc)
> 
> Where can I find the network key?

found this key in the docs but does not seem enable wireshark to decrypt the messages.

[image]

xieqinan, 2025-01-02:
Has the `temperature_sensor` joined the HA network? If it has, please ensure the sniffer is activated before the device commissions with the HA. This approach should allow the frame to be parsed successfully.  

Let's first confirm whether the temperature value is being sent by the device.

ear9mrn, 2025-01-02:
> Has the `temperature_sensor` joined the HA network? If it has, please ensure the sniffer is activated before the device commissions with the HA. This approach should allow the frame to be parsed successfully.
> 
> Let's first confirm whether the temperature value is being sent by the device.

Yes. my ESP-h2 (temperature) and ESP-C6 are talking to each other, no problems.

[image]

Still cannot decrypt the message though using wireshark. I also noticed changing the network mask (for to a certain channel eg. 16) appears to have no effect, they always appear to connect on channel 14.

ear9mrn, 2025-01-03:
> I believe you only need to modify the following code in the `HA_temperature_sensor` example. This will enable the temperature value to be reported to HA automatically at a 30-second interval (as set by HA). The root cause of the issue lies in the unit of the temperature.
> 
> ```
> static int16_t zb_temperature_to_s16(float temp)
> {
>    /* MeasuredValue = 100 x temperature in degrees Celsius, Where -273.15°C <= temperature <= 327.67 ºC, corresponding 
>       to a MeasuredValue in the range 0x954d to 0x7fff. The maximum resolution this format allows is 0.01 ºC.
>    */
>     int16_t ret = (temp * 100) + rand() % 10 * 100;
>     return ret;
> }
> ```

Many thanks, the temperature is now updating in home assistant as expected.  One slight issue when swapping from connecting to my esp32-c6 (as coordinator on channel 14) to zigbee2mqtt (in home assistant on channel 11) my ESP32-H2 would not join, error below. 

```
W (26646) ESP_ZB_TEMP_SENSOR: BDB Device Reboot failed with status: ESP_FAIL, retrying
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 54:32:04:ff:fe:33:9e:90 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
```

I had to erase the flash and re-flash. I assume it was trying to rejoin the old network and could not find it. Is there a more elegant solution in order to swap from one network to another other than "erase_flash". 

```
idf.py -p /dev/ttyACM0 erase_flash

```

Thanks for all your help.

Pete.

xieqinan, 2025-01-15:
> Is there a more elegant solution in order to swap from one network to another other than "erase_flash".

You may need to delete the original device information from Home Assistant first, and then try joining it again.