espressif/esp-zigbee-sdk #516: Attribute handler for window covering lift percentage data size and data value incorrect

radovanobal, 2024-12-29:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.2

### esp-zigbee-lib version.

1.6.1

### esp-zboss-lib version.

1.6.1

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

When the attribute change is processed the message should contain:

A data size of 1 byte
The data should contain the number between 0 (0x00) and 100 (0x64)

### What is the actual behavior?

The size is reported as 16896 bytes instead of 1 and the value is 0 when cast to uint8_t.
[image]

### Steps to reproduce.

1. Join the coordinator
2. Send to the exposed percentage position command
3. Read the received data

### More Information.

Processing the same response trough esp_zb_raw_command_handler_register the correct values get returned 
[image]

xieqinan, 2024-12-30:
@radovanobal ,

Could you please share how you determine the `attribute_size` of the window covering lift percentage? If the `ESP_ZB_CORE_WINDOW_COVERING_MOVEMENT_CB_ID` callback is triggered, the stack reports an `esp_zb_zcl_window_covering_movement_message_t` message, which does not include the `attribute_size` value.

radovanobal, 2024-12-30:
oh I was casting to `esp_zb_zcl_set_attr_value_message_t`, not to `esp_zb_zcl_window_covering_movement_message_t`, getting the size and value from the wrong cast resulted in the weird values. 

Sorry for the false report, I'm a bit new to all this.

closed 2024-12-30 by radovanobal