espressif/esp-zigbee-sdk #518: Adding carbon dioxide cluster makes the device crash

RobinFrcd, 2025-01-01:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-847-gcb3ac7429c

### esp-zigbee-lib version.

1.6.1

### esp-zboss-lib version.

1.6.1

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

Hi,
I currently have 2 clusters on my device (temperature and humidity) and I'm trying to add a new Carbon Dioxide one. But as soon as I add it, the device crashes with:

```
I (8304) ESP_ZB_TEMP_SENSOR: Humidity: 40.29 %
I (8304) ESP_ZB_TEMP_SENSOR: PM2.5: 9.1 µg/m³
I (8304) ESP_ZB_TEMP_SENSOR: CO2: 10000 ppm
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4202fd78  RA      : 0x4202fd74  SP      : 0x4084ee60  GP      : 0x4080e5a4  
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4202fd78: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4202fd74: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
TP      : 0x4084efb0  T0      : 0x400184be  T1      : 0x0000000f  T2      : 0x62eb8841  
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/.espressif/tools/esp-rom-elfs/20241011/esp32h2_rev0_rom.elf 0x400184be: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
S0/FP   : 0x4081c754  S1      : 0x00000003  A0      : 0x00000000  A1      : 0x00000000  
A2      : 0x00000000  A3      : 0x00000005  A4      : 0x4081c312  A5      : 0x00000055  
A6      : 0x00000005  A7      : 0x00000005  S2      : 0x00000000  S3      : 0x4081c6ac  
S4      : 0x00000000  S5      : 0x00000006  S6      : 0x00000000  S7      : 0x00000052  
S8      : 0x00000016  S9      : 0xfffffffe  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x000000fd  T4      : 0x000000fe  T5      : 0xffffffff  T6      : 0x000000ff  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000003  
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x40800001: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
MHARTID : 0x00000000  

Stack memory:
4084ee60: 0x00000008 0x00000017 0x0000004f 0x4081c6ac 0x4081c6ac 0x4081c47e 0x40818784 0x4202c9c4
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4202c9c4: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
4084ee80: 0x00000000 0x00000042 0x0000004d 0x4084f014 0x00000008 0x4084eec8 0x00000005 0x00000002
4084eea0: 0x7fffffff 0x40814000 0x4084f7f2 0x000186a0 0x00000000 0x00000000 0x00000006 0x00000000
4084eec0: 0x00000002 0x00000052 0x4081c6ac 0x4202f958 0x7fffffff 0x00000000 0x4202f868 0x009a24b3
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4202f958: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4202f868: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
4084eee0: 0x7fffffff 0x00000000 0x40810708 0x4201d538 0x00000000 0x00000000 0x00000000 0x00000000
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4201d538: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
4084ef00: 0x00000000 0x00000000 0x420320a2 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x420320a2: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
4084ef20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4200f990
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4200f990: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
4084ef40: 0x00000000 0x00000000 0x00000000 0x4200f9a6 0x00000000 0x00000000 0x00000000 0x4200cfd0
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4200f9a6: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4200cfd0: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
4084ef60: 0x00000002 0x00000000 0x00000006 0x00000bb8 0x00000000 0x00000000 0x00000000 0x408086b2
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x408086b2: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
4084ef80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4084efa0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x4084ed90
[... 60 lines cut ...]

Report value with

```c
void reportValue(uint16_t cluserID, uint16_t valueID, void *value)
{
    esp_zb_zcl_set_attribute_val(
        HA_ESP_SENSOR_ENDPOINT, 
        cluserID, 
        ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
        valueID, 
        value, 
        false
    );
}

uint16_t co2_value = (co2 > 10000) ? 10000 : co2;
reportValue(ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT, ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MEASURED_VALUE_ID, &co2_value);
```

### More Information.

Am I missing anything here ?

Thanks !

xieqinan, 2025-01-02:
The code you provided appears correct upon review. However, the crash log lacks some critical information, you could also try increasing the Zigbee task stack size to see if it resolves the issue. BTW, does the crash occur when adding the CO2 cluster or during the CO2 reporting process?  

Could you please share more detailed code to help us identify the root cause of the crash? Ideally, providing a reproducible code snippet would be very helpful.

RobinFrcd, 2025-01-02:
Hi,
Thank you for your answer. So I already tried to increase the stack (`16384`) an removing other clusters didn't help neither, so I don't think it's full.

Here's a full code with no sensor and hardcoded values:

```c
#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"
#include "esp_zb_sen66.h"
#include "string.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile sensor (End Device) source code.
#endif

#define DEFINE_PSTRING(var, str)   \
    const struct                   \
    {                              \
        unsigned char len;         \
        char content[sizeof(str)]; \
    }(var) = {sizeof(str) - 1, (str)}

static const char *TAG = "ESP_ZB_SEN66_SENSOR";

#define ESP_ZB_ZCL_CLUSTER_ID_PM10_MEASUREMENT 0x042dU
#define ESP_ZB_ZCL_ATTR_PM10_MEASUREMENT_MEASURED_VALUE_ID 0x0000

#define ESP_ZB_ZCL_CLUSTER_ID_VOC_MEASUREMENT 26113
#define ESP_ZB_ZCL_ATTR_VOC_MEASUREMENT_MEASURED_VALUE_ID 0x0000

void reportAttribute(uint16_t clusterID, uint16_t attributeID, void *value, uint8_t value_length)
{
    esp_zb_zcl_report_attr_cmd_t cmd = {
        .zcl_basic_cmd = {
            .dst_addr_u.addr_short = 0x0000,
            .dst_endpoint = HA_ESP_SENSOR_ENDPOINT,
            .src_endpoint = HA_ESP_SENSOR_ENDPOINT,
        },
        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
        .clusterID = clusterID,
        .attributeID = attributeID,
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
    };
    esp_zb_zcl_attr_t *value_r = esp_zb_zcl_get_attribute(HA_ESP_SENSOR_ENDPOINT, clusterID, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, attributeID);
    memcpy(value_r->data_p, value, value_length);
    esp_zb_zcl_report_attr_cmd_req(&cmd);
}

void reportValue(uint16_t cluserID, uint16_t valueID, void *value)
{
    //esp_zb_lock_acquire(portMAX_DELAY);
    esp_zb_zcl_set_attribute_val(
        HA_ESP_SENSOR_ENDPOINT, 
        cluserID, 
        ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
        valueID, 
        value, 
        false
    );
    // esp_zb_lock_release();
}

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_level_commissioning(mode_mask) == ESP_OK, ,
                        TAG, "Failed to start Zigbee bdb commissioning");
[... 205 lines cut ...]
        esp_zb_lock_release();
        vTaskDelay(pdMS_TO_TICKS(5000));
    }
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));

    init_zigbee();

    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
    
    xTaskCreate(read_temperature_task, "temp_reader", 16384, NULL, 4, NULL);
}
```

This one works, but as soon as I remove the `/* Code that makes it fail */` part, it starts crashing with the `Load access fault`.

Thank you for your precious help !

xieqinan, 2025-01-03:
Hi,

The hardcoded implementation provides better insights into your project, but it works fine on my end. I copied it into the `HA_temperature_sensor` example, removed the comments, and ran it alongside the `HA_thermostat` example. You can check the log below—there is no crashing observed.
```
I (288) app_init: ESP-IDF:          v5.5-dev-847-gcb3ac7429c7
I (295) efuse_init: Min chip rev:     v0.0
I (299) efuse_init: Max chip rev:     v0.99 
I (304) efuse_init: Chip rev:         v0.1
I (309) heap_init: Initializing. RAM available for dynamic allocation:
I (316) heap_init: At 40814940 len 00038A40 (226 KiB): RAM
I (322) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (329) heap_init: At 50000000 len 00000FE8 (3 KiB): RTCRAM
I (336) spi_flash: detected chip: generic
I (339) spi_flash: flash io: dio
W (343) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (357) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (364) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (372) main_task: Started on CPU0
I (382) main_task: Calling app_main()
I (402) phy: phy_version: 310,2, 463a420, Aug 15 2024, 18:52:11
I (402) phy: libbtbb version: a024da6, Aug 15 2024, 18:52:23
I (402) ESP_ZB_SEN66_SENSOR: Temperature measurement cluster added
I (412) ESP_ZB_SEN66_SENSOR: Humidity measurement cluster added
I (412) ESP_ZB_SEN66_SENSOR: PM2.5 measurement cluster added
I (422) ESP_ZB_SEN66_SENSOR: Carbon dioxide measurement cluster added
I (522) ESP_ZB_SEN66_SENSOR: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (522) ESP_ZB_SEN66_SENSOR: Initialize Zigbee stack
I (522) ESP_ZB_SEN66_SENSOR: Device started up in factory-reset mode
I (532) ESP_ZB_SEN66_SENSOR: Start network steering
I (542) ESP_ZB_SEN66_SENSOR: Temperature: 4000 °C
I (542) ESP_ZB_SEN66_SENSOR: Humidity: 40.00 %
I (542) ESP_ZB_SEN66_SENSOR: PM2.5: 100.0 µg/m³
I (552) ESP_ZB_SEN66_SENSOR: CO2: 200.0 ppm
I (562) main_task: Returned from app_main()
I (3522) ESP_ZB_SEN66_SENSOR: Joined network successfully (Extended PAN ID: xx:xx:xx:xx:xx:xx:xx:xx, PAN ID: 0x67e1, Channel:13, Short Address: 0x89d7)
I (5562) ESP_ZB_SEN66_SENSOR: Temperature: 4000 °C
I (5562) ESP_ZB_SEN66_SENSOR: Humidity: 40.00 %
I (5562) ESP_ZB_SEN66_SENSOR: PM2.5: 100.0 µg/m³
I (5562) ESP_ZB_SEN66_SENSOR: CO2: 200.0 ppm
I (10572) ESP_ZB_SEN66_SENSOR: Temperature: 4000 °C
I (10572) ESP_ZB_SEN66_SENSOR: Humidity: 40.00 %
I (10572) ESP_ZB_SEN66_SENSOR: PM2.5: 100.0 µg/m³
I (10572) ESP_ZB_SEN66_SENSOR: CO2: 200.0 ppm
I (15582) ESP_ZB_SEN66_SENSOR: Temperature: 4000 °C
I (15582) ESP_ZB_SEN66_SENSOR: Humidity: 40.00 %
I (15582) ESP_ZB_SEN66_SENSOR: PM2.5: 100.0 µg/m³
I (15582) ESP_ZB_SEN66_SENSOR: CO2: 200.0 ppm
I (20592) ESP_ZB_SEN66_SENSOR: Temperature: 4000 °C
I (20592) ESP_ZB_SEN66_SENSOR: Humidity: 40.00 %
I (20592) ESP_ZB_SEN66_SENSOR: PM2.5: 100.0 µg/m³
I (20592) ESP_ZB_SEN66_SENSOR: CO2: 200.0 ppm
I (25602) ESP_ZB_SEN66_SENSOR: Temperature: 4000 °C
I (25602) ESP_ZB_SEN66_SENSOR: Humidity: 40.00 %
I (25602) ESP_ZB_SEN66_SENSOR: PM2.5: 100.0 µg/m³
I (25602) ESP_ZB_SEN66_SENSOR: CO2: 200.0 ppm
I (30612) ESP_ZB_SEN66_SENSOR: Temperature: 4000 °C
I (30612) ESP_ZB_SEN66_SENSOR: Humidity: 40.00 %
I (30612) ESP_ZB_SEN66_SENSOR: PM2.5: 100.0 µg/m³
I (30612) ESP_ZB_SEN66_SENSOR: CO2: 200.0 ppm
I (35622) ESP_ZB_SEN66_SENSOR: Temperature: 4000 °C
I (35622) ESP_ZB_SEN66_SENSOR: Humidity: 40.00 %
I (35622) ESP_ZB_SEN66_SENSOR: PM2.5: 100.0 µg/m³
I (35622) ESP_ZB_SEN66_SENSOR: CO2: 200.0 ppm
I (40632) ESP_ZB_SEN66_SENSOR: Temperature: 4000 °C
I (40632) ESP_ZB_SEN66_SENSOR: Humidity: 40.00 %
I (40632) ESP_ZB_SEN66_SENSOR: PM2.5: 100.0 µg/m³
I (40632) ESP_ZB_SEN66_SENSOR: CO2: 200.0 ppm

```
I reviewed the code you provided, and its logic appears clear. I believe it should work well.  

Could you also try enlarging the `Zigbee_main` task? Since the esp-zigbee-lib is compiled based on ESP-IDF v5.3.2, could you check it out and give it a try? If the issue persists and the system still crashes, please share the `.elf` file and crashing log with us for further analysis.

RobinFrcd, 2025-01-03:
Hi,
So I indeed took the same base project (so we have the shame header file here). I increased the stack sizes up to `32768`, didn't have any impact.

Here's the full crashing log, with the .elf (inside a zip because of github constraints): [Senesp98.zip](https://github.com/user-attachments/files/18299157/Senesp98.zip)

```
$ '/home/robz/.espressif/python_env/idf5.5_py3.10_env/bin/python' '/home/robz/dev/HA/esp/master/esp-idf/tools/idf_monitor.py' -p /dev/ttyUSB0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32h2 '/home/robz/dev/HA/Senesp98/build/
Senesp98.elf'
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460f0,len:0x16f8
load:0x4083cad0
,len:0xf04
load:0x4083efd0,len:0x2f30
entry 0x4083cada
I (23) boot: ESP-IDF v5.5-dev-1050-gb5ac4fbdf9-dirty 2nd stage bootloader
I (24) boot: compile time Jan  3 2025 12:13:02
I (25) boot: chip revision: v0.1
I (26) boot: efuse block revision: v0.3
I (29) boot.esp32h2: SPI Speed      : 64MHz
I (33) boot.esp32h2: SPI Mode       : DIO
I (37) boot.esp32h2: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (61) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (67) boot:  2 factory          factory app      00 00 00010000 000e1000
I (74) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (80) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (87) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=42058020 size=0d9d0h ( 55760) map
I (113) esp_image: segment 1: paddr=0001d9f8 vaddr=40800000 size=02620h (  9760) load
I (117) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=56d18h (355608) map
I (218) esp_image: segment 3: paddr=00076d40 vaddr=40802620 size=0a05ch ( 41052) load
I (232) esp_image: segment 4: paddr=00080da4 vaddr=4080c680 size=01ba0h (  7072) load
I (239) boot: Loaded app from partition at offset 0x10000
I (240) boot: Disabling RNG early entropy source...
I (250) cpu_start: Unicore app
I (259) cpu_start: Pro cpu start user code
I (259) cpu_start: cpu freq: 96000000 Hz
I (259) app_init: Application information:
I (260) app_init: Project name:     Senesp98
I (263) app_init: App version:      cda3d9f-dirty
I (267) app_init: Compile time:     Jan  3 2025 12:12:59
I (272) app_init: ELF file SHA256:  c593e5dfc...
I (277) app_init: ESP-IDF:          v5.5-dev-1050-gb5ac4fbdf9-dirty
I (283) efuse_init: Min chip rev:     v0.0
I (286) efuse_init: Max chip rev:     v0.99 
I (290) efuse_init: Chip rev:         v0.1
I (294) heap_init: Initializing. RAM available for dynamic allocation:
I (300) heap_init: At 40812960 len 0003AA20 (234 KiB): RAM
I (306) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (311) heap_init: At 50000000 len 00000FE8 (3 KiB): RTCRAM
I (317) spi_flash: detected chip: generic
I (320) spi_flash: flash io: dio
W (323) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (335) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (341) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (348) main_task: Started on CPU0
I (358) main_task: Calling app_main()
I (388) phy: phy_version: 310,2, 463a420, Aug 15 2024, 18:52:11
I (388) phy: libbtbb version: a024da6, Aug 15 2024, 18:52:23
I (388) ESP_ZB_SEN66_SENSOR: Temperature measurement cluster added
I (388) ESP_ZB_SEN66_SENSOR: Humidity measurement cluster added
[... 68 lines cut ...]
40822070: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40822090: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
408220b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
408220d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
408220f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40822110: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40822130: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: c593e5dfc

Rebooting...
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400031b6
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/.espressif/tools/esp-rom-elfs/20241011/esp32h2_rev0_rom.elf 0x400031b6: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460f0,len:0x16f8
load:0x4083cad0,len:0xf04
load:0x4083efd0,len:0x2f30
entry 0x4083cada
```

PS: I will give a try to v5.3.2 a bit later but compilation failed on first tries for some reasons. But as it works fine for you with v5.5 it would be great if we manage to debug this for the latest version !

grebenyuk51, 2025-01-06:
I had the same issue once. Helped erase flash and reprogram
`esptool.py -p COM8 erase_flash` (change COM8 to yours specific port)

closed 2025-01-06 by RobinFrcd

grebenyuk51, 2025-01-06:
Every time you add new cluster to working (and connected to network) device, don't forget to erase flash

In `esp_zigbee_core.h` I found `esp_zb_zcl_reset_all_endpoints_to_factory_default()` and `esp_zb_zcl_reset_endpoint_to_factory_default()` and `esp_zb_zcl_reset_nvram_to_factory_default()` maybe they can help either, I haven't tested they yet
Maybe @xieqinan can help with this question

srett, 2025-08-02:
Holy shit I was just debugging my device for an hour and `erase-flash` solved it.

Is this documented somewhere that we need to erase the flash when changing the endpoint/cluster config? If so, it should be made more prominent... ;)

srett, 2025-08-04:
@xieqinan 

> Generally speaking, the data mode of an endpoint/cluster is configured at runtime and is not stored in flash.

That's what I'd assume, so this phenomenon seemed very strange to me. Coincidentally I'm also working on a CO2 monitor, but I can't imagine this is actually related to the problem.

What I've been doing at the time I ran into this problem was figuring out how to add a custom cluster with custom attributes, so I changed the EP and cluster configuration quite a lot. Then I ran into the crashing problem. At first I assumed I'm doing something wrong in how I allocate all the structs and register them, as the problem specifically occurred around attribute reporting. Either when setting up attribute reporting, or when calling `esp_zb_zcl_set_attribute_val()`.  When I encountered the problem, I tried moving the `esp_zb_zcl_update_reporting_info()` calls around. First I had them before `esp_zb_start(false)`, so I tried moving them after `esp_zb_start(false)`, but this a) didn't solve the problem, and b) now every call to `esp_zb_zcl_update_reporting_info()` _except the first one_ returned `ESP_ERR_NO_MEM`. So I moved it back to before `esp_zb_start(false)` and at least all the calls now returned `ESP_OK`.  
I guess out of desperation I tried increasing stack sizes and a bunch of other silly things to no avail.  
As soon as I did `erase-flash`, everything started working again.

Do you have any theory what might be causing this? Seems at least three people have now encountered this.

This was the crash I got:

```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

--- Stack dump detected
Core  0 register dump:
MEPC    : 0x4203989e  RA      : 0x4203989a  SP      : 0x4081c0f0  GP      : 0x40810ca4  
--- 0x4203989e: zb_zcl_get_next_reporting_info at ??:?
--- 0x4203989a: zb_zcl_get_next_reporting_info at ??:?

TP      : 0x4081c2d0  T0      : 0x40030dca  T1      : 0x0000000f  T2      : 0xaa2a1a09  
--- 0x40030dca: memset in ROM

S0/FP   : 0x40822b20  S1      : 0x00000001  A0      : 0x00000000  A1      : 0x00000001  
A2      : 0x00000000  A3      : 0x00000004  A4      : 0x408229c7  A5      : 0x00000055  
A6      : 0x00000004  A7      : 0x00000004  S2      : 0x00000004  S3      : 0x40822ae8  
S4      : 0x00000000  S5      : 0x00000006  S6      : 0x00000000  S7      : 0x00000052  
S8      : 0xffffffd0  S9      : 0x00000000  S10     : 0x0000001b  S11     : 0x00000000  
T3      : 0x000000fd  T4      : 0x000000fe  T5      : 0xffffffff  T6      : 0x000000ff  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000003  
--- 0x40800001: _vector_table at /home/sr/dev/esp/esp-idf/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000  

--- Backtrace:

0x4203989e in zb_zcl_get_next_reporting_info ()
#0  0x4203989e in zb_zcl_get_next_reporting_info ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

lpy4105, 2025-08-05:
@srett 
The report information is stored in NVRAM. That may cause problem when you switch project without erasing the flash.

srett, 2025-08-05:
@lpy4105 Ah that makes sense, thanks. In that case, if it is too much work to add verification code to detect incompatible changes, I would suggest adding a heads-up to the documentation around reporting functions such as `esp_zb_zcl_update_reporting_info`. I think especially during development of a new device it is quite likely that you change the device configuration a lot and might trigger this kind of problem.