espressif/esp-zigbee-sdk #519: ZCL OnOff: Off with effect has no effect in HA_color_dimmable_light

wejn, 2025-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.5 (latest docker image)

### esp-zigbee-lib version.

1.6.1 (a943a9118e9ad110e2641e1187fd4c5d533f8a06)

### esp-zboss-lib version.

1.6.1 (354f742fab22cdb12b4ae61a103a7b438c7c975f)

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Light turns off.

### What is the actual behavior?

Light stays on.

### Steps to reproduce.

1. Build and flash `HA_color_dimmable_light` example (see more in this comment on [#358](https://github.com/espressif/esp-zigbee-sdk/issues/358#issuecomment-2567111040))
2. Connect the light to Philips Hue hub, [using the Hue app] turn the light on
3. [using the Hue app] turn the light off

### More Information.

The app receives Off with effect message:

[image]

And that message never makes it to the user handler.

I diffed processing of an `On` message and compared it to `Off with effect` message and it seems that the `zb_zcl_process_on_off_off_with_effect_handler` does pretty much nothing:

[on-vs-off.diff.txt](https://github.com/user-attachments/files/18289860/on-vs-off.diff.txt)

Is the [HA_color_dimmable_light](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_color_dimmable_light/) example perhaps incomplete? Or what could be the reason for the `Off with effect` being ignored?

xieqinan, 2025-01-03:
@wejn ,

Based on the sniffer log, the `Off with effect` command is sent via a group. Could you please double-check if the destination endpoint has been correctly assigned to the `0x97A2` group?  

You can use the `esp_zb_zcl_groups_view_group_cmd_req()` API to send a command to the device itself to verify the group membership.

wejn, 2025-01-03:
@xieqinan,

No need to test that (even though the answer would've been yes), because:

[image]

When sent directly (the Hue app always sends the `Delayed All Off` when turning off a light), the same happens. The command is ignored by the bulb; although this time there's error reported back:

[image]

So I'm guessing the `Off with effect` isn't by default implemented? In the standard it's marked as optional, not mandatory:

[image]

Isn't this, given the response, somehow similar to #457 ?

Also, I searched around in the sdk docs and the issues, and in [this comment on #75](https://github.com/espressif/esp-zigbee-sdk/issues/75#issuecomment-1687387855) you're talking about the "privilege command". Are we supposed to implement these for all the optional messages? Or is there some default implementation in the SDK that just needs to be activated somehow?

wejn, 2025-01-03:
Hm, I had a bit of time to play with this, and indeed, adding:

``` c
// https://github.com/espressif/esp-zigbee-sdk/issues/457#issuecomment-2426128314
uint16_t on_off_on_time = 0;
bool on_off_global_scene_control = 0;
esp_zb_cluster_list_t *cluster_list = esp_zb_ep_list_get_ep(esp_zb_color_dimmable_light_ep, HA_COLOR_DIMMABLE_LIGHT_ENDPOINT);
esp_zb_attribute_list_t *attr_list =
    esp_zb_cluster_list_get_cluster(cluster_list, ESP_ZB_ZCL_CLUSTER_ID_ON_OFF, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    
esp_zb_on_off_cluster_add_attr(attr_list, ESP_ZB_ZCL_ATTR_ON_OFF_ON_TIME, &on_off_on_time);
esp_zb_on_off_cluster_add_attr(attr_list, ESP_ZB_ZCL_ATTR_ON_OFF_GLOBAL_SCENE_CONTROL,
        &on_off_global_scene_control);
```

straight before the `esp_zb_device_register(esp_zb_color_dimmable_light_ep);` (which is the [code you posted on #457](https://github.com/espressif/esp-zigbee-sdk/issues/457#issuecomment-2426128314) modded for the color light example) makes the `Off with effect` work:

```
I (50554) ESP_ZB_COLOR_DIMM_LIGHT: Light sets to On
I (50674) ESP_ZB_COLOR_DIMM_LIGHT: Received message: endpoint(10), cluster(0x8), attribute(0x0), data size(1)
I (50674) ESP_ZB_COLOR_DIMM_LIGHT: Light level changes to 255
I (54434) ESP_ZIGBEE_ZCL_SCENES: Update existed scene(0x0) for group(0x0)
W (54434) ESP_ZB_COLOR_DIMM_LIGHT: Receive Zigbee action(0x1) callback
I (54444) ESP_ZB_COLOR_DIMM_LIGHT: Received message: endpoint(10), cluster(0x6), attribute(0x0), data size(1)
I (54454) ESP_ZB_COLOR_DIMM_LIGHT: Light sets to Off
I (58104) ESP_ZB_COLOR_DIMM_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (64644) ESP_ZB_COLOR_DIMM_LIGHT: Received message: endpoint(10), cluster(0x8), attribute(0x0), data size(1)
I (64644) ESP_ZB_COLOR_DIMM_LIGHT: Light level changes to 0
I (64644) ESP_ZB_COLOR_DIMM_LIGHT: Received message: endpoint(10), cluster(0x6), attribute(0x0), data size(1)
I (64654) ESP_ZB_COLOR_DIMM_LIGHT: Light sets to On
I (64764) ESP_ZB_COLOR_DIMM_LIGHT: Received message: endpoint(10), cluster(0x8), attribute(0x0), data size(1)
I (64764) ESP_ZB_COLOR_DIMM_LIGHT: Light level changes to 255
I (66304) ESP_ZIGBEE_ZCL_SCENES: Update existed scene(0x0) for group(0x0)
W (66304) ESP_ZB_COLOR_DIMM_LIGHT: Receive Zigbee action(0x1) callback
I (66304) ESP_ZB_COLOR_DIMM_LIGHT: Received message: endpoint(10), cluster(0x6), attribute(0x0), data size(1)
I (66314) ESP_ZB_COLOR_DIMM_LIGHT: Light sets to Off
```

So, @xieqinan, maybe a different question:

What other things should be added to the `HA_color_dimmable_light` example to avoid other weird behavior (or nasty surprises like #358).

Because for example the scenes on the example are still emitting a warning like: `W (370704) ESP_ZIGBEE_ZCL_SCENES: Scene(0x1) not found for group(0x9731), unable to recall it.` (on in-hue-app scene activation and even on in-hue-app scene creation) so I'm guessing there's additional magic that needs to be somehow plugged in in order to have scenes work properly. And maybe other things[^1]?

Maybe additional question: isn't there some more complete color light example that would have most of the bugs ironed out? I know I'm kinda wishing for pretty much "give me a clone of philips hue lightbulb" here (heh), but the brokeness and having to scour bug tracker for various well hidden answers is... strange[^3]?

Or maybe just having some overview checklist of what specific things should be handled by the users of the SDK in order for most of the things to work well would be nice[^2].

[^1]: I'm betting the reset (network leave) behavior needs modding for sure.
[^2]:  (Shouldn't we, for example, somehow save the scenes somewhere? in flash? IDK...)
[^3]: I hope not intentional :wink:

xieqinan, 2025-01-09:
> Also, I searched around in the sdk docs and the issues, and in https://github.com/espressif/esp-zigbee-sdk/issues/75#issuecomment-1687387855 you're talking about the "privilege command". Are we supposed to implement these for all the optional messages? Or is there some default implementation in the SDK that just needs to be activated somehow?

The `privilege command` is a supplemental implementation for Zigbee provided in the esp-zigbee-sdk and is not part of the standard Zigbee specification. We do not recommend using it for general development. It will only be activated if explicitly registered.
> What other things should be added to the HA_color_dimmable_light example to avoid other weird behavior (or nasty surprises like https://github.com/espressif/esp-zigbee-sdk/issues/358).

The esp-zigbee-sdk adheres to the Zigbee specification in providing its APIs. However, compatibility issues with other manufacturers' implementations may arise. To address such cases, the SDK offers configuration APIs to help adapt to these differences. If you encounter compatibility issues, please feel free to open a discussion to address them.

Additionally, we recommend separating different issues into individual tickets for clarity and independence. If the issue described in this ticket has been resolved, please consider closing it. Any remaining issues can be discussed in separate tickets.

wejn, 2025-01-10:
Yes, original issue resolved, closing for now. Might open follow-up issues for the other problems (scenes, etc).