espressif/esp-zigbee-sdk #520: Cluster revision is RW should be R only

damian-kurek-wizzdev, 2025-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.5

### esp-zigbee-lib version.

1.6

### esp-zboss-lib version.

1.6

### Espressif SoC revision.

ESP32C6

### What is the expected behavior?

[image]
[image]

### What is the actual behavior?

Cluster revision should be read only

### Steps to reproduce.

Run any zigbee cluster and check parameter

### More Information.

I used conbee2 deconz software to get this

xieqinan, 2025-01-03:
Hi，

I have double-checked the `ClusterRevision` setting in the esp-zigbee-sdk, which is set to `ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY` by default. Do you know how HUE obtains `rw` access from other devices?

damian-kurek-wizzdev, 2025-01-03:
I double-checked and it doesn't look like I changed something, I can not see anything that would change it in the code I have access to. It looks like it is done by something in precompiled libraries. The code I use to create cluster.
[create_thermostat_cluster_code.txt](https://github.com/user-attachments/files/18298315/create_thermostat_cluster_code.txt)

Additionally, I noticed that I need to provide         .app_device_version = 1 in esp_zb_endpoint_config_t, shouldn't that number be used in basic cluster Application version attribute? 
[image]

xieqinan, 2025-01-09:
@damian-kurek-wizzdev ,

> I double-checked and it doesn't look like I changed something, I can not see anything that would change it in the code I have access to. It looks like it is done by something in precompiled libraries. The code I use to create cluster.

Could you please use the `esp_zb_zcl_get_attribute()` function to retrieve the related attribute and print its access type? I suspect that the `deconz` is only displaying the default value set within itself.

> Additionally, I noticed that I need to provide .app_device_version = 1 in esp_zb_endpoint_config_t, shouldn't that number be used in basic cluster Application version attribute?

No, I think they are not the same value.

closed 2025-01-14 by damian-kurek-wizzdev