espressif/esp-zigbee-sdk #521: Thermostat gets ACHTUNG: Sending OOM status

damian-kurek-wizzdev, 2025-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.5

### esp-zigbee-lib version.

1.6

### esp-zboss-lib version.

1.6

### Espressif SoC revision.

esp32c6

### What is the expected behavior?

It won't get oom 

### What is the actual behavior?

After sending raise/lower heat only 10 or -10 multiple times even when you get error that it is not possible to change temperature by 10 keep sending the message
ZBOSS: nwk/nwk_nlme.c:808   OOM state is detected. ACHTUNG: Out of memory!
ZBOSS: common/zb_bufpool_mult.c:1650   OOM! bufs allocated {21 i-o 38} max 80
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
And zigbee stops responding. It is not possible to send anything to the device

### Steps to reproduce.

1. Configure thermosta cluter
2. Send raise/lower command only one way e.g. raise by 10
3. When you get to the limit of what can be changed keep sending the command
4. You should get the error at some point
...

damian-kurek-wizzdev, 2025-01-02:
It seems that I don't get callback when raise lower is too low but callbacks get allocated.

xieqinan, 2025-01-03:
Hi,

Some questions need to be clarified for this issue:  

- How many `raise/lower` commands are being sent?  
- What is the frequency of sending these commands?  
- If the commands stop being sent, does the issue still occur?  

If the commands are sent at a high frequency, you can call `esp_zb_io_buffer_size_set()` to improve the stack's throughput.

damian-kurek-wizzdev, 2025-01-03:
[image]
**What is the frequency of sending these commands?**
Frequency does not matter as long as you send it when the result value is outside of the max value ( or min value)
**How many raise/lower commands are being sent?** 32 I think happened 2 times after 32 out of range raises
**If the commands stop being sent, does the issue still occur?** No, when I send a command that raises/lower is in range

xieqinan, 2025-01-09:
@damian-kurek-wizzdev,

I tried to reproduce your issue on my side, and everything seems to work fine. I might have missed something, so I’ve attached an example for you. Could you please modify it to create a reproducible example and share it with me?
[test-thermostat.zip](https://github.com/user-attachments/files/18355705/test-thermostat.zip)

damian-kurek-wizzdev, 2025-01-09:
https://github.com/damian-kurek-wizzdev/zigbee

xieqinan, 2025-08-12:
Hi @damian-kurek-wizzdev ,

Has this issue been resolved on your side? Please let me know the status so I can provide better assistance.

damian-kurek-wizzdev, 2025-08-13:
We will check if that still happens

closed 2025-08-13 by damian-kurek-wizzdev