espressif/esp-zigbee-sdk #522: esp_zb_zdo_device_leave_req()   It did't work.

badjane, 2025-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### esp-zigbee-lib version.

1.6.0

### esp-zboss-lib version.

1.6.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

When the network is open, the endpoint device will be able to re-enter after leaving the current network

### What is the actual behavior?

This is because the end device that invokes the interface cannot connect to the previous network after the restart

### Steps to reproduce.

1. A coordinator and endpoint device
2. Ask the coordinator to turn on network 180s
3. The endpoint device successfully joined the coordinator network
4. Ask the coordinator to turn off the network 180s
5. Shortly after joining, the endpoint device calls "esp_zb_zdo_device_leave_req()", leaves the network, and the software restarts
6. Ask the coordinator to open the network 180s again
7. At this point, the endpoint device can easily repeat the error of not continuing the network

### More Information.

not

badjane, 2025-01-03:
esp_zb_factory_reset() --> I tested it, but it didn't work

badjane, 2025-01-03:
[image]

badjane, 2025-01-03:
[image]

badjane, 2025-01-03:
Look at the above two pictures, the device calls "esp_zb_zdo_device_leave_req()" and restarts after 5 seconds, the device still continues to send requests to the network

badjane, 2025-01-04:
esp-zigbee-lib version.
1.5.1

esp-zboss-lib version.
1.5.1
It works correctly

xieqinan, 2025-01-09:
@badjane ,

Could you share the log of the end device? I suspect that the re-steering logic might have been triggered in the application.

badjane, 2025-01-11:
Sorry, it's project privacy. But you can reproduce it as follows
Use an HA switch to open the network by pressing the button, and let the HA lamp connect to the switch. After the lamp is connected, press the button to close the HA switch and connect to the network. At this time, the switch informs the lamp to leave the network, and then start all steps again after the lamp leaves the network。

chshu, 2025-03-20:
@badjane Any update on the issue?

xieqinan, 2025-08-12:
Hi @badjane ,

Since you have not responded to this issue for a long time, and we have tested and verified that `esp_zb_zdo_device_leave_req()` works, I will close this issue. If you have any questions, please feel free to reopen it.

closed 2025-08-12 by xieqinan