espressif/esp-zigbee-sdk #526: Report comfig action handler

grebenyuk51, 2025-01-05:
### Is your feature request related to a problem?

I have Thermostat-ESP32-C6 and Thermometr-ESP32-C6. They are simple devices with thermometer+relay and thermometer respectively. I can configure binding and reporting from one device to another. So I bind Thermometer to Thermostat and configure reporting of temperature value Minimum Reporting Interval for 120 seconds. Therefore there is no need to measure temperature more often than once in 120 seconds

### Describe the solution you'd like.

receive in `zb_action_handler()` callback with parameters of configured report

xieqinan, 2025-01-06:
I cannot fully grasp your requirement. Could you please provide a detailed application scenario to explain why you expect to obtain the configured report from the `zb_action_handler()`?

grebenyuk51, 2025-01-06:
In my network there are 3 devices: Coordinator, Thermostat, Thermometer
I do configure throw Coordinator
So, Coordinator sends Binding request and bind Thermometer to Coordinator and configures Reporting of MeasuredValue {minimum reporting interval: 120 sec, maximum reporting interval: 3600 sec, minimum change: 0.5 degree}

I would like to change temperature calculation interval on Thermometer. There is no need to read temperature every 10 seconds, if minimum reporting interval configured to 120 seconds

So i thought the best way to get reporting params is `zb_action_handler()`

On the other hand, I can see some stuff in `zb_raw_command_handler()` during reporting configuration. But I have now idea how to parse raw command handler

grebenyuk51, 2025-01-06:
I think I've done what I wanted
```
#include "zboss_api.h"
...
esp_zb_raw_command_handler_register(zb_raw_command_handler);
...
bool zb_raw_command_handler(uint8_t bufid)
{
	uint8_t buf[zb_buf_len(bufid)];
	zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, zb_zcl_parsed_hdr_t);
	printf("cluster id: 0x%x, command id: %d\n", cmd_info->cluster_id, cmd_info->cmd_id);
	memcpy(buf, zb_buf_begin(bufid), sizeof(buf));
	ESP_LOGE("RAW", "bufid: %d size: %zu", bufid, sizeof(buf));
	for (int i = 0; i < sizeof(buf); ++i) {
		printf("0x%02X ", buf[i]);
	}
	printf("\n");
	
	if (cmd_info->cmd_id == ZB_ZCL_CMD_CONFIG_REPORT)
	{
		zb_zcl_configure_reporting_req_t *rep_req = (zb_zcl_configure_reporting_req_t *)buf;
		ESP_LOGI("REPORTING", "Endpoint: 0x%d, ClusterID: 0x%04x, AttrID: 0x%02x, Min interval: %d, Max interval: %d, RepChange: %d", 
			cmd_info->addr_data.common_data.dst_endpoint, 
			cmd_info->cluster_id,
			rep_req->attr_id,
			rep_req->u.clnt.min_interval,
			rep_req->u.clnt.max_interval,
			rep_req->u.clnt.delta[0]);
			
	}
	
	return false;
}
```

Output is:
```
cluster id: 0x402, command id: 6
E (9749) RAW: bufid: 15 size: 10
0x00 0x00 0x00 0x29 0x78 0x00 0x10 0x0E 0x32 0x00 
I (9749) REPORTING: Endpoint: 0x4, ClusterID: 0x0402, AttrID: 0x00, Min interval: 120, Max interval: 3600, RepChange: 50
```

I would greatly appreciate it if you could provide a callback through the SDK

xieqinan, 2025-01-15:
@grebenyuk51,

The `esp_zb_raw_command_handler_register()` indeed allows users to intercept ZCL commands. However, tampering with valid commands from other devices may not be a good practice. If you believe the `Configure Reporting` command from the coordinator is not as expected, you can use `esp_zb_zcl_update_reporting_info()` to update the reporting information locally.  

Therefore, a more reasonable and justified request is needed to ensure that introducing a new callback for the `Configure Reporting` command is meaningful and beneficial.

grebenyuk51, 2025-01-18:
Zigbee2MQTT works in such way. So to be fully compatible with Z2M is needed to intercept Z2M Configuration commends (ZCL commands) such as Configure Reporting
If this argument is not enough, [here ](https://github.com/espressif/esp-zigbee-sdk/issues/526#issuecomment-2572588627) I gave an example just in case

closed 2025-01-23 by grebenyuk51