espressif/esp-zigbee-sdk #527: PM2.5 Cluster implementation

RobinFrcd, 2025-01-06:
### Question

Hi,
I saw in https://github.com/espressif/esp-zigbee-sdk/issues/490 it's possible to get access to some parts of the code. I'm not a professional (doing this as a hobby side project) so my request by mail got denied.

I want to implement a PM10 sensor (`0x042dU`) following the same pattern as the native PM2.5 implementation. Would it be possible to get the official codes of `esp_zb_cluster_list_add_pm2_5_measurement_cluster` and `esp_zb_pm2_5_measurement_cluster_create` ?

If not possible, would it be then ok to get the modified versions for **pm10** ? As they do not exist, I guess they are not copyrighted, right ?

Thanks !

xieqinan, 2025-01-09:
@RobinFrcd ,

> I want to implement a PM10 sensor (0x042dU) following the same pattern as the native PM2.5 implementation. Would it be possible to get the official codes of esp_zb_cluster_list_add_pm2_5_measurement_cluster and esp_zb_pm2_5_measurement_cluster_create ?

I am afraid the standard ZCL cluster, such as `PM10 sensor (0x042dU)`, cannot be implemented using the public APIs due to limitations in the SDK. However, you can implement a custom cluster with an ID greater than `0x8000` using the public APIs.

RobinFrcd, 2025-01-09:
Alright, understood.
Is that what the assertion error is all about in https://github.com/espressif/esp-zigbee-sdk/issues/529 then ? All custom clusters must have a greater ID ?

RobinFrcd, 2025-01-09:
Asked for official implementation here: https://github.com/espressif/esp-zigbee-sdk/issues/531

closed 2025-01-09 by RobinFrcd