espressif/esp-zigbee-sdk #528: ColorControl Cluster Group Commands are only forwarded to one en

Haerteleric, 2025-01-07:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### esp-zigbee-lib version.

1.6.1

### esp-zboss-lib version.

1.6.1

### Espressif SoC revision.

ESP-C6 Rev0

### What is the expected behavior?

I have a multi EP Application running on the ESP-C6. These are multiple independent Extended Color Light EPs. 

If i put multiple of those EPs in the same Group and issued a Move-To-Color Command. But only one EP in the Group Updates it's Attributes. 

This is only happens for Color Control Cluster Commands. ON/Off and Level Commands work fine in that scenario.

### What is the actual behavior?

if multiple EPs are in the Same Multicast-Group. The Color Attributes for all EPs in that group should get updated.

### Steps to reproduce.

1. Create App with >1 Extended Color Light EPs.
2. Put both of these EPs into the same Group
3. Issue a Move-to-Color Command to that Group
4. Only one EP gets the Attribute changed Callbacks

xieqinan, 2025-01-09:
Thanks for your report.

I have confirmed this issue, and it is caused by the incomplete implementation of the ZCL Color Control cluster. This limitation means that only one endpoint with color control can be processed correctly, while other endpoints will fail in your case.

We plan to address this issue, but it will take some time. In the meantime, if it is possible, you could divide the multiple endpoints across multiple chips to meet your application's requirements.

Haerteleric, 2025-02-03:
Could you give me an estimate when you will address this issue ?

xieqinan, 2025-02-13:
Hi,

Sorry for the delay. This feature will be supported, but not in the recent version. If you have specific requirements, please contact our [Technical Support](https://www.espressif.com/en/contact-us/technical-inquiries).

Haerteleric, 2025-02-26:
can you give me a ETA when this addressed ?

chshu, 2025-02-26:
@Haerteleric It's a limitation in the ZBOSS stack. We are rewriting the lower-layer stack with optimized design, and plan to release the brand new SDK by middle this year. The issue will also be fixed then.

xieqinan, 2025-09-02:
@Haerteleric,

> are there any updates ?

Sorry, as mentioned in [this comment](https://github.com/espressif/esp-zigbee-sdk/issues/528#issuecomment-2684425864), this feature is not supported temporarily. If it blocks your development, please contact [our business team](https://www.espressif.com/en/contact-us/sales-questions). If I receive your contact details from them, I can provide a temporary library to fix this issue as soon as possible.

Alternatively, you can use the ZCL privilege command mechanism to intercept the color control command and implement the command handler yourself. If you’d like to proceed with this approach, let me know and I can provide documentation to guide you through its usage.