espressif/esp-zigbee-sdk #529: How to set custom attribute value on custom cluster

RobinFrcd, 2025-01-07:
### Question

Hi,
I'm trying to create a simple custom cluster and then set a custom attribute value.

The creation of the cluster works fine with:
```c
#define ESP_ZB_ZCL_CLUSTER_ID_VOC_MEASUREMENT 0x042EU
#define ESP_ZB_ZCL_ATTR_VOC_MEASUREMENT_MEASURED_VALUE_ID 0x0000

float_t zero_value = 0.0;
esp_zb_attribute_list_t *voc_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_VOC_MEASUREMENT);
esp_zb_custom_cluster_add_custom_attr(
    voc_cluster, 
    ESP_ZB_ZCL_ATTR_VOC_MEASUREMENT_MEASURED_VALUE_ID, 
    ESP_ZB_ZCL_ATTR_TYPE_SINGLE, 
    ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, 
    &zero_value
);
esp_zb_cluster_list_add_custom_cluster(cluster_list, voc_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
```

But as soon as I try to update a value with

```c
void reportValue(uint16_t cluserID, uint16_t valueID, void *value)
{
    esp_zb_lock_acquire(portMAX_DELAY);
    esp_zb_zcl_set_attribute_val(
        HA_ESP_SENSOR_ENDPOINT, 
        cluserID, 
        ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
        valueID, 
        value, 
        false
    );
    esp_zb_lock_release();
}

float_t voc_value = 0;
reportValue(ESP_ZB_ZCL_CLUSTER_ID_VOC_MEASUREMENT, ESP_ZB_ZCL_ATTR_VOC_MEASUREMENT_MEASURED_VALUE_ID, &voc_value);
```

It crashes with this stack:
```
Zigbee stack assertion failed zcl/zcl_general_commands.c:612

abort() was called at PC 0x420196d1 on core 0
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x420196d1: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
Core  0 register dump:
MEPC    : 0x408009b6  RA      : 0x4080776c  SP      : 0x408235e0  GP      : 0x4080e634  
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x408009b6: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4080776c: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
TP      : 0x40823780  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x40823644  A0      : 0x4082360c  A1      : 0x40823642  
A2      : 0x00000000  A3      : 0x40823639  A4      : 0x00000001  A5      : 0x40814000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x40823728  S3      : 0x0000000a  
S4      : 0x0000042e  S5      : 0x00000001  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x40800001: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
MHARTID : 0x00000000  

Stack memory:
408235e0: 0x00000000 0x00000000 0x40823640 0x4080cc34 0x00000000 0x00000001 0x0000042e 0x4080eea8
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4080cc34: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40823600: 0x40823644 0x4080eec4 0x40823640 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
40823620: 0x43502074 0x34783020 0x39313032 0x20316436 0x63206e6f 0x2065726f 0x00000030 0x42010000
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x42010000: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40823640: 0x00000030 0x31303234 0x31643639 0x00000000 0x0000002f 0x0000ffff 0x42065549 0x420196d4
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x420196d4: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40823660: 0x408236ec 0x42063870 0x42063100 0x00000264 0xffffffff 0x42063870 0x00000000 0x4202c1b2
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4202c1b2: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40823680: 0x00000000 0x00000000 0x00000001 0x0000042e 0x00000000 0x40823728 0x0000042e 0x4202a8fe
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4202a8fe: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
408236a0: 0x408236ec 0x42063870 0x00000003 0x00000000 0x45cccc00 0x00000cbf 0x42063000 0x00000000
408236c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x40823728 0x00000000 0x0000042e 0x4200e39a
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4200e39a: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
408236e0: 0x00000000 0x00000000 0x0000040d 0x4200d13a 0x45cccc00 0x00000cbf 0x42063000 0x4200d410
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4200d13a: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4200d410: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40823700: 0x00000897 0x00000000 0x4200d14a 0x00000000 0x00000000 0x00000000 0x00000000 0x08970001
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x4200d14a: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40823720: 0x45cccc00 0x3c23d70a 0x00000000 0xffff0000 0x00007fff 0x0c47112e 0xffffffff 0xffffffff
40823740: 0x00000000 0x00000000 0x00000000 0x408086ae 0x00000000 0x00000000 0x00000000 0x00000000
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/dev/HA/Senesp98/build/Senesp98.elf 0x408086ae: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40823760: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40823780: 0xa5a5a5a5 0xa5a5a5a5 0x00029bf1 0x408141c4 0x408141c4 0x4082373c 0x00000054 0x00000000
408237a0: 0x4082379c 0x00000000 0x00000000 0x00000000 0x408237b4 0xffffffff 0x408237b4 0x408237b4
408237c0: 0x00000000 0x408237c8 0xffffffff 0x408237c8 0x408237c8 0x00000001 0x00000001 0x00000000
408237e0: 0x0000ffff 0x00000000 0x40823790 0x40823794 0x00029b89 0x408141c4 0x408141c4 0x00000000
40823800: 0x00000000 0x00000000 0x4082380c 0xffffffff 0x4082380c 0x4082380c 0x00000000 0x40823820
40823820: 0xffffffff 0x40823820 0x40823820 0x00000001 0x00000001 0x00000000 0x3300ffff 0x00000000
40823840: 0x0de24339 0x408237ec 0x00029b33 0x408141c4 0x408141c4 0xadac0167 0xc11f7e45 0xa13e85f4
40823860: 0xe2a50cf0 0xeb6657df 0x907facf0 0x648475d8 0xe5d46550 0x5b9c8b46 0xa5675bd3 0xa73d5340
40823880: 0x5730a13b 0x2fd07be1 0x675539d5 0xed1f2428 0x3af738a8 0xe4569fa1 0xa3e1a921 0x5b85cfc8
408238a0: 0xfb3969ad 0x3ae77a00 0x84d1ad65 0xa6c59996 0x71571e65 0x3ef2df72 0x0c6a07e1 0x5f129983
408238c0: 0xbf7ec2d2 0x74489d22 0x40d0e4d8 0xaffca69a 0xfa9d634c 0xe5bdd4d9 0x01d4f9a0 0x3cf0aeaf
408238e0: 0xda2c1a72 0xe15c0ed6 0x70f2cffb 0x98cf3a8d 0x30780696 0x84ae2b9d 0xa8bd17df 0x2c4e5b30
40823900: 0xe4f32e55 0x1f499185 0x132e7189 0x540dc2a4 0x8961b31b 0xb28308cb 0x2732331a 0x3fa85495
40823920: 0xd7247176 0x49340f53 0x07218a9e 0xe037629b 0x2a359e5f 0xfe89fe09 0x830437ad 0x3746e042
40823940: 0x7dfcf617 0xf5b7394b 0x755d77b8 0x2268b148 0xd19bf80b 0x8409cc4d 0x14458f09 0x96972a86
40823960: 0xb72e768b 0x8c90c6d5 0x394e676a 0x70480020 0x2cbddea1 0x2c6348ad 0x7b063c73 0xd0c2553b
40823980: 0xb4371a29 0x1db4691f 0x8dde79e5 0x8b2eb610 0xffe59ac1 0x21567ce1 0x6d5f33b7 0x3214f261
408239a0: 0xb5b84a0b 0x8734befb 0xbf911099 0xe116c25a 0xbe51de93 0x54351543 0x9848aee5 0x815dc495
408239c0: 0xf8c3da23 0x12b226b5 0xb4b47978 0x782ad07d 0xb17f534c 0x1feb5263 0x4fa7b38c 0xcae0233f

ELF file SHA256: 2aaa2367c

Rebooting...
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400031b6
riscv32-esp-elf-addr2line -pfiaC -e /home/robz/.espressif/tools/esp-rom-elfs/20241011/esp32h2_rev0_rom.elf 0x400031b6: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460f0,len:0x16f8
load:0x4083cad0,len:0xf04
load:0x4083efd0,len:0x2f30
entry 0x4083cada
```

So I'm not sure if it's a bug or if I'm not using these functions correctly. The stack trace is not helpful as we don't know which assertion is failing. I couldn't find any example/documentation on this topic (only cluster creation is available).

I'm using `ESP-IDF v5.5-dev-1050-gb5ac4fbdf9-dirty`.

Already tried clearing the ESP32 with `esptool.py -p /dev/ttyACM0 erase_flash`.

Thanks

grebenyuk51, 2025-01-07:
In esp_zigbee_attribute.h
`esp_zb_zcl_status_t esp_zb_zcl_set_manufacturer_attribute_val(uint8_t endpoint, uint16_t cluster_id, uint8_t cluster_role, uint16_t manuf_code, uint16_t attr_id, void *value_p, bool check);`

RobinFrcd, 2025-01-07:
Thank you again for your answer ! So `esp_zb_zcl_set_manufacturer_attribute_val` should be used instead of `esp_zb_zcl_set_attribute_val` ?
There's even less doc available for this function, I've tried a few Manufacturer Codes (as the list seems not to be public) I could find but still getting the same assertion error.

grebenyuk51, 2025-01-08:
don't forget about `esp_err_t esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list_t *cluster_list, esp_zb_attribute_list_t *attr_list, uint8_t role_mask);` and `esp_err_t esp_zb_cluster_add_manufacturer_attr(esp_zb_attribute_list_t *attr_list, uint16_t cluster_id, uint16_t attr_id, uint16_t manuf_code, uint8_t attr_type, uint8_t attr_access, void *value_p);`

RobinFrcd, 2025-01-08:
Thank you.

Yes, this is what I tried when suggesting switching to manufacturer:
```c
#define ESP_ZB_ZCL_CLUSTER_ID_VOC_MEASUREMENT 0x042EU
#define ESP_ZB_ZCL_ATTR_VOC_MEASUREMENT_MEASURED_VALUE_ID 0x0000
#define ESP_ZB_ZCL_MANUFACTURER_CODE_SENSIRION 0x1234

float_t zero_value = 0.0;
esp_zb_attribute_list_t *voc_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_VOC_MEASUREMENT);
esp_zb_cluster_add_manufacturer_attr(
    voc_cluster, 
    ESP_ZB_ZCL_CLUSTER_ID_VOC_MEASUREMENT, 
    ESP_ZB_ZCL_ATTR_VOC_MEASUREMENT_MEASURED_VALUE_ID, 
    ESP_ZB_ZCL_MANUFACTURER_CODE_SENSIRION, 
    ESP_ZB_ZCL_ATTR_TYPE_SINGLE, 
    ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE | ESP_ZB_ZCL_ATTR_MANUF_SPEC, 
    &zero_value
);
esp_zb_cluster_list_add_custom_cluster(cluster_list, voc_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

float_t voc_value = 0;
esp_zb_zcl_set_manufacturer_attribute_val(
    HA_ESP_SENSOR_ENDPOINT, 
    ESP_ZB_ZCL_CLUSTER_ID_VOC_MEASUREMENT, 
    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
    ESP_ZB_ZCL_MANUFACTURER_CODE_SENSIRION,
    ESP_ZB_ZCL_ATTR_VOC_MEASUREMENT_MEASURED_VALUE_ID, 
    &voc_value, 
    false
);
```

But still same error.
(And I'm also erasing the flash on every build, just to be sure)

xieqinan, 2025-01-09:
Hi @RobinFrcd ,

> #define ESP_ZB_ZCL_CLUSTER_ID_VOC_MEASUREMENT 0x042EU

The standard cluster ID cannot be created as a custom cluster. Could you please share your purpose or specific use case? This will help us explore alternative ways to address your issue effectively.

RobinFrcd, 2025-01-09:
I'm reading all data coming from https://sensirion.com/products/catalog/SEN66. I then want to expose this data to Zigbee2MQTT.
CO2, temperature, PM2.5 and humidity clusters work fine. I now want to expose new clusters for VOC, NO, PM10, ...

Thanks !

xieqinan, 2025-01-09:
Understood. However, I am concerned that your requirement may not be implementable with the current SDK, as standard clusters can only be added using the standard APIs provided by the SDK. Therefore, I recommend issuing a ticket, similar to (https://github.com/espressif/esp-zigbee-sdk/issues/478), to request the related feature.

RobinFrcd, 2025-01-09:
And it's not doable either with custom clusters ? I need this fairly quickly so I'm not able to wait for all these standard clusters to be implemented right now :/

closed 2025-01-09 by RobinFrcd