espressif/esp-zigbee-sdk #530: Can you stop/deinit zigbee stack

damian-kurek-wizzdev, 2025-01-08:
### Question

Can I stop/deinit Zigbee stack or do I need to restart the device?

### Additional context.

I have existing firmware that uses WiFi to connect and I'm creating a version of firmware that will support Zigbee and Wi-Fi. I assume that ZigBee and wifi won't work together. 
If the user runs ZigBee commissioning and fails, I want to deinit ZigBee so the user can use WiFi or Zigbee again. Is this possible or I need to reboot the esp

chshu, 2025-02-05:
@damian-kurek-wizzdev Please note that there is only one RF path on C6, meaning Wi-Fi, BLE, Zigbee cannot access the radio simultaneously. The coexistence mechanism prioritizes access based on the modem with higher priority.

> If the user runs ZigBee commissioning and fails, I want to deinit ZigBee so the user can use BLE+WIFI or Zigbee again is this possible?

This is possible and can be implemented through application layer logic.

> Can I run BLE+WiFi commissioning when ZigBee is running?

It depends on the Zigbee role:
- If it's a Zigbee router, it's not recommended. Zigbee router needs to continuously receive signals, increased traffic on the Wi-Fi/BLE side may result in a higher packet loss rate for Zigbee communication.
- If it's a Zigbee end device, it's possible. The Wi-Fi, BLE and Zigbee could access the Radio according to their priorities.

chshu, 2025-03-20:
The supported coexistence scenarios on C6: https://docs.espressif.com/projects/esp-idf/en/v5.2.5/esp32c6/api-guides/coexist.html#supported-coexistence-scenario-for-esp32-c6

closed 2025-03-20 by chshu