espressif/esp-zigbee-sdk #535: Sets INSTALLCODE_POLICY_ENABLE to true, but is not available

Mushan-bin, 2025-01-15:
### Question

On the coordinator
[image]
[image]
[image]
On the terminal, I use mfg_tool to set up the configuration
[image]
[image]
Coordinator result
[image]
End result
[image]
The installation code is consistent, but the terminal simply cannot successfully join the coordinator.

xieqinan, 2025-01-15:
Could you please refer to https://github.com/espressif/esp-zigbee-sdk/issues/398#issuecomment-2268591073 comment first? It may be helpful for you.

Mushan-bin, 2025-01-16:
Thank you very much