espressif/esp-zigbee-sdk #538: ESP Zigbee Gateway example failes (ESP32S3 + ESP32H2)

czagaadam, 2025-01-18:
Hi mates!

I've seen that similiar issues are already raised in github but I haven't found a proper solution.
So I'm using an ESP32 S3 (WROOM 1) with the esp_zigbee_gateway code and an ESP32 H2 (MINI 1) with esp_zigbee_rcp code.
H2's RX and TX pins are connected with S3 PIN 4 and 5, but I don't think this would be the issue.

When I launch the monitor for S3 in VSCode I'm getting this error message:
host mode uart not supported

[image]

Please let me know what is the issue and how can I resolve this.

xieqinan, 2025-01-20:
@czagaadam ,

Could you please follow the setup [README](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_gateway/README.md) of gateway to test again?

czagaadam, 2025-01-20:
> [@czagaadam](https://github.com/czagaadam) ,
> 
> Could you please follow the setup [README](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_gateway/README.md) of gateway to test again?

Okay I tried, it seems my original Gateway example on S3 was outdated because this new one is much longer:
https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway
Although it is still not working, I'm getting this in Monitor:

[image]

[image]

I updated H2 with ot_rcp example, and here is what I'm seeing in Monitor:

[image]
 
S3 pin config:

[image]

H2 pin config:

[image]

For me the RESET pins seems working because H2 is resetting again and again but serial connection seems wrong.
In H2 logs: rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0)) means that H2 waiting the update in UART0?
As it can be seen in pin config I redefined H2's UART to pin 4 and 5, is it problem ?

xieqinan, 2025-01-21:
Hi,

Could you try using the default TX/RX pins of the RCP for testing? or disable the `ZIGBEE_GW_AUTO_UPDATE_RCP` in menuconfig.

czagaadam, 2025-01-22:
I tried with H2's default TX RX pins (GPIO 24 & 23) with both ZIGBEE_GW_AUTO_UPDATE_RCP disable and enable.
That's what I've got:

[image]

[image]

Should I try shorter jumper wire  or lower the baud rate ?
I'll also check the connection with uart analyzer.

chshu, 2025-02-26:
@czagaadam Noticed that you are using the S3 GPIO17 and GPIO18, as noted in [ESP32-S3 TRM, Chapter 6.12](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf), the GPIO17 and GPIO18 on ESP32-S3 have different driver current, it may cause some issue with longer jumper wires. Please try with other GPIOs.

Also it would be easier if you use the [ESP Zigbee Gateway board](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway#esp-zigbee-gateway-board), the gateway example runs on the board as is.

closed 2025-03-20 by chshu