espressif/esp-zigbee-sdk #539: zigbee stack assertion failed nwk/nwk_join.c

Mushan-bin, 2025-01-20:
### Question

I would like to know, only access coordinator, terminal will have this problem, which is why.
[image]

xieqinan, 2025-01-20:
It seems that an unauthorized device is attempting to join the network. Could you please provide more context regarding this issue for further investigation?

If possible, I would like to receive the reproduction steps, the logs from the affected device, and the Wireshark `.pcap` file captured during the reproduction process.

chshu, 2025-03-20:
@Mushan-bin could you share more details regarding this issue?

xieqinan, 2025-06-13:
Hi @Mushan-bin ,

It is impossible to locate the root reason  by only a assert(). Since there has been no feedback for some time, I’ll go ahead and close this issue. Feel free to reopen it with the more detailed background for your project if you're still experiencing confusion.

closed 2025-06-13 by xieqinan