espressif/esp-zigbee-sdk #541: Failed to initialize zigbee stack  after resuming from deep sleep

ulyssesdotcodes, 2025-01-22:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### esp-zigbee-lib version.

1.6.2

### esp-zboss-lib version.

1.6.2

### Espressif SoC revision.

ESP32-c6

### What is the expected behavior?

Continue resuming from sleep correctly

### What is the actual behavior?

Fails to initialize zigbee stack after a few hours.

### Steps to reproduce.

1. Flash my project based on the sleepy end device project
2. Wait for a fer hours

### More Information.

My project (https://github.com/[ulyssesdotcodes/esp32-zigbee-sensor](https://github.com/ulyssesdotcodes/esp32-zigbee-sensor) based on the sleepy end device project stops reporting zigbee information to my Home Assistant setup after a few hours. This happens on both an espressif devkit and a firebeetle. Last night I left the devkit attached to my computer and took logs showing this:

```
--- Error: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
--- Waiting for the device to reconnect............................................................
I (3344) ESP_ZB_TEMP_SENSOR: INITTED sensor
I (3344) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (3354) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (3364) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (3344) ESP_ZB_TEMP_SENSOR: Deferred driver initialization successful
I (3374) ESP_ZB_TEMP_SENSOR: Device started up in non factory-reset mode
I (3384) ESP_ZB_TEMP_SENSOR: Wake up from timer. Time spent in deep sleep and boot: 33391ms
I (3394) ESP_ZB_TEMP_SENSOR: Start one-shot timer for 10s to enter the deep sleep
I (3404) ESP_ZB_TEMP_SENSOR: Device rebooted
I (4374) ESP_ZB_TEMP_SENSOR: start sensor_read
I (4374) ESP_ZB_TEMP_SENSOR: setting gpio pin  18 to level 1
I (4374) ESP_ZB_TEMP_SENSOR: reading...
I (4374) ESP_ZB_TEMP_SENSOR: ADC1 Channel[2] Raw Data: 1585
I (4384) ESP_ZB_TEMP_SENSOR: ADC1 Channel[2] Voltage: 1558
I (4384) ESP_ZB_TEMP_SENSOR: ADC1 Channel[0] Raw Data: 1072
I (4394) ESP_ZB_TEMP_SENSOR: ADC1 Channel[0] Voltage: 1054
I (4404) ESP_ZB_TEMP_SENSOR: Callback from handler
I (4404) ESP_ZB_TEMP_SENSOR: Reporting 5125 battery 34
I (4414) ESP_ZB_TEMP_SENSOR: lock acquired - setting attribute
I (4414) ESP_ZB_TEMP_SENSOR: lock releasing
I (4424) ESP_ZB_TEMP_SENSOR: Send 'report attributes' command
I (13404) ESP_ZB_TEMP_SENSOR: Enter deep sleep
--- Error: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
--- Waiting for the device to reconnect............................................................
W (2826) ESP_ZB_TEMP_SENSOR: Failed to initialize Zigbee stack (status: ESP_FAIL)
```

It looks like for some reason the zigbee stack fails to initialize which means the deep_sleep_start function never gets called.

None of the other threads regarding this issue seem relevant - I think they all have fixes in previous versions of the zigbee sdk. What else could cause this issue?

xieqinan, 2025-01-22:
Hi,

I think it is not a bug, you only need to retry it. Please refer to the [deep sleep implementation](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/deep_sleep/main/esp_zb_sleepy_end_device.c#L154) of esp-zigbee-sdk v1.6.2.

ulyssesdotcodes, 2025-01-22:
I did try that this morning but it was still failing on the devkit. After removing the battery reporting it started working again - I'll report back in a few hours to see if it is still working. 

It would be ideal if I could have battery reporting and deep sleep though. Is this possible with the current SDK?

ulyssesdotcodes, 2025-01-25:
It looks like this is working now. Sorry for creating the issue!

closed 2025-01-25 by ulyssesdotcodes