espressif/esp-zigbee-sdk #543: gateway example report "OPENTHREAD:[W] P-RadioSpinel-: radio tx timeout"

wensenl, 2025-01-22:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.3.2

### esp-zigbee-lib version.

1.6.1

### esp-zboss-lib version.

1.6.1

### Espressif SoC revision.

ESP32S3/ESP32H2

### What is the expected behavior?

do not report "radio tx timeout" and do not reset esp32h2

### What is the actual behavior?

gateway rand report  "radio tx timeout" and reset esp32h2

### Steps to reproduce.

1. esp32S3 and esp32h2 Zigbee gateway
2. 35 zigbee devices connect to gateway
3. gateway send data to 35 zigbee device every second
4. gateway can rand report  "radio tx timeout" and reset esp32h2
W(19078783) OPENTHREAD:[W] P-RadioSpinel-: radio tx timeout
W(19078783) OPENTHREAD:[W] P-RadioSpinel-: RCP failure detected
W(19078783) OPENTHREAD:[W] P-RadioSpinel-: Trying to recover (1/3)
I(19078833) OPENTHREAD:[N] P-RadioSpinel-: RCP recovery is done

I have set esp32s3 and esp32h2 gpio :
if (config.radio_config.radio_uart_config.rx_pin != -1) {
        gpio_set_drive_capability(config.radio_config.radio_uart_config.rx_pin, GPIO_DRIVE_CAP_3);
    }
    if (config.radio_config.radio_uart_config.tx_pin != -1) {
        gpio_set_drive_capability(config.radio_config.radio_uart_config.tx_pin, GPIO_DRIVE_CAP_3);
    }

xieqinan, 2025-01-23:
Hi,

In complex scenarios like the one mentioned above, it is impossible to ensure 100% accuracy in data transmission between the gateway and RCP. The recovery mechanism is helpful in addressing failures caused by asynchronous states between the host and RCP. While it may slightly impact performance, it is a beneficial measure to enhance the system's stability.

wensenl, 2025-01-23:
esp32s3 esp32h2 uart baud rate is 460800. Will writing log files to flash storage affect the serial communication between ESP32S3 and ESP32H2?

xieqinan, 2025-06-03:
Hi @wensenl,

> Will writing log files to flash storage affect the serial communication between ESP32S3 and ESP32H2?

Sorry for the delay. I don't think it will. I believe the problem is triggered by network traffic—specifically, the coordinator is receiving data from 35 devices every second, which may cause the UART (FIFO size is 128 bytes) to be overwhelmed by the transmission load.

Could you please share what kind of data is being sent by the 35 devices? We will try to reproduce the issue on our side.

In any case, the `recovery mechanism` of the radio spinel can handle such failures and does not affect data transmission.

xieqinan, 2025-09-30:
@wensenl ,

Based on a similar test case, the `radio tx timeout` is indeed triggered, but I believe this behavior is reasonable. Could you please share what impact it had in your test?

closed 2026-01-20 by xieqinan