espressif/esp-zigbee-sdk #544: How do I get the current connection state

jonasbu01, 2025-01-22:
### Question

How do I get the current zigbee connection status? (I want to use an LED as a connection-indicator)
Thank you

xieqinan, 2025-01-23:
The `esp_zb_bdb_dev_joined()` indicates whether the device is currently on the network, while the `esp_zb_bdb_is_factory_new()` indicates that the device may not currently be on the network but has previously joined one.

jonasbu01, 2025-01-25:
When I unplug the Zigbee dongle, the `esp_zb_bdb_dev_joined()` function still returns `true`. However, there are no other Zigbee devices in the network, meaning the ESP32-H2 has no connection in this scenario.

Is this normal and is there another way to check if the esp32 is connected to the network?

xieqinan, 2025-02-06:
As we know, the Zigbee network will not disappear just because a single device goes offline. Any Zigbee router can maintain the network independently. Please clarify which scenario you consider a connection and which one you consider a disconnection.

wejn, 2025-03-08:
I guess I would add to this question:

How can I tell that the coordinator is reachable?

While

> Any Zigbee router can maintain the network independently

is all well and good, it'd be great to have an easy way to test whether the coordinator is reachable.

xieqinan, 2025-03-10:
@wejn ,

> How can I tell that the coordinator is reachable?

You can try sending a message (such as `esp_zb_zdo_ieee_addr_req()`) to the coordinator. If the message is unreachable, `ZB_ZDO_DEVICE_UNAVAILABLE` will be triggered.

wejn, 2025-03-15:
Thank you, in the end I opted for a different approach -- using `esp_zb_nwk_get_next_neighbor` to get the coord.

xieqinan, 2025-03-17:
@jonasbu01 @wejn ，

Do you have any other questions regarding this issue? If not, please consider closing it.

wejn, 2025-03-17:
Closeable from my side. Not sure about @jonasbu01.

closed 2025-03-20 by chshu