espressif/esp-zigbee-sdk #545: You cannot reach μA in light_sleep

Mushan-bin, 2025-01-23:
### Question

When using the ESP32-H2, perform the configuration as described in the example.

[image]

[image]
I cannot reach the μA level shown in the picture above, only 1mA, why is this, is there any configuration not set.

xieqinan, 2025-01-23:
I think the above configuration is sufficient. Could you please share the test steps and results with me?

Mushan-bin, 2025-01-24:
I (23) boot: ESP-IDF v5.1.4 2nd stage bootloader
I (24) boot: compile time Jan 23 2025 10:04:30
I (25) boot: chip revision: v0.1
I (27) boot.esp32h2: SPI Speed      : 64MHz
I (31) boot.esp32h2: SPI Mode       : DIO
I (36) boot.esp32h2: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 000e1000
I (79) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (87) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=42060020 size=112f8h ( 70392) map
I (129) esp_image: segment 1: paddr=00021320 vaddr=40800000 size=0e6fch ( 59132) load
I (150) esp_image: segment 2: paddr=0002fa24 vaddr=4080e700 size=005f4h (  1524) load
I (151) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=5ede0h (388576) map
I (274) esp_image: segment 4: paddr=0008ee08 vaddr=4080ecf4 size=0119ch (  4508) load
I (276) esp_image: segment 5: paddr=0008ffac vaddr=50000000 size=0005ch (    92) load
I (284) boot: Loaded app from partition at offset 0x10000
I (286) boot: Disabling RNG early entropy source...
I (302) cpu_start: Unicore app
I (303) cpu_start: Pro cpu up.
W (312) clk: esp_perip_clk_init() has not been implemented yet
I (318) cpu_start: Pro cpu start user code
I (319) cpu_start: cpu freq: 96000000 Hz
I (319) cpu_start: Application information:
I (321) cpu_start: Project name:     hello_world
I (327) cpu_start: App version:      1
I (331) cpu_start: Compile time:     Jan 23 2025 10:04:03
I (337) cpu_start: ELF file SHA256:  28a77e5c35bcd34d...
I (343) cpu_start: ESP-IDF:          v5.1.4
I (348) cpu_start: Min chip rev:     v0.0
I (353) cpu_start: Max chip rev:     v0.99
I (358) cpu_start: Chip rev:         v0.1
I (362) heap_init: Initializing. RAM available for dynamic allocation:
I (370) heap_init: At 40814750 len 00038C30 (227 KiB): D/IRAM
I (376) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
I (384) spi_flash: detected chip: generic
I (387) spi_flash: flash io: dio
I (392) sleep: Enable automatic switching of GPIO sleep configuration
I (399) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (407) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (415) sleep_sys_periph: Interrupt Matrix sleep retention initialization
I (423) sleep_sys_periph: HP System sleep retention initialization
I (430) sleep_sys_periph: TEE/APM sleep retention initialization
I (436) sleep_sys_periph: UART sleep retention initialization
I (443) sleep_sys_periph: Timer Group sleep retention initialization
I (450) sleep_sys_periph: IO Matrix sleep retention initialization
I (457) sleep_sys_periph: SPI Mem sleep retention initialization
I (463) sleep_sys_periph: SysTimer sleep retention initialization
I (481) app_start: Starting scheduler on CPU0
I (482) main_task: Started on CPU0
I (482) main_task: Calling app_main()
I (493) pm: Frequency switching config: CPU_MAX: 96, APB_MAX: 96, APB_MIN: 96, Light sleep: ENABLED
I (495) esp_zb_sleep: Enable ieee802154 zigbee light sleep, the wake up source is ESP timer
I (510) phy: phy_version: 230,2, 9aae6ea, Jan 15 2024, 11:17:12
I (512) phy: libbtbb version: 944f18e, Jan 15 2024, 11:17:25
I (516) btbb_init: btbb sleep retention initialization
I (521) ieee802154: ieee802154 mac sleep retention initialization
I (540) main_task: Returned from app_main()
I (542) ESP_ZB_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (543) ESP_ZB_SLEEP: Initialize Zigbee stack
I (550) ESP_ZB_SLEEP: Deferred driver initialization successful
I (553) ESP_ZB_SLEEP: Device started up in factory-reset mode
I (560) ESP_ZB_SLEEP: Start network steering
I (2827) ESP_ZB_SLEEP: Network steering was not successful (status: -1)
I (2828) ESP_ZB_SLEEP: Zigbee can sleep
I (6101) ESP_ZB_SLEEP: Network steering was not successful (status: -1)
I (6152) ESP_ZB_SLEEP: Zigbee can sleep
I (9420) ESP_ZB_SLEEP: Network steering was not successful (status: -1)
I (9471) ESP_ZB_SLEEP: Zigbee can sleep
I (9549) ESP_ZB_SLEEP: Zigbee can sleep

Mushan-bin, 2025-01-24:
> 我认为上述配置就足够了。您能否与我分享测试步骤和结果？

I (23) boot: ESP-IDF v5.1.4 2nd stage bootloader
I (24) boot: compile time Jan 23 2025 10:04:30
I (25) boot: chip revision: v0.1
I (27) boot.esp32h2: SPI Speed      : 64MHz
I (31) boot.esp32h2: SPI Mode       : DIO
I (36) boot.esp32h2: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 000e1000
I (79) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (87) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=42060020 size=112f8h ( 70392) map
I (129) esp_image: segment 1: paddr=00021320 vaddr=40800000 size=0e6fch ( 59132) load
I (150) esp_image: segment 2: paddr=0002fa24 vaddr=4080e700 size=005f4h (  1524) load
I (151) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=5ede0h (388576) map
I (274) esp_image: segment 4: paddr=0008ee08 vaddr=4080ecf4 size=0119ch (  4508) load
I (276) esp_image: segment 5: paddr=0008ffac vaddr=50000000 size=0005ch (    92) load
I (284) boot: Loaded app from partition at offset 0x10000
I (286) boot: Disabling RNG early entropy source...
I (302) cpu_start: Unicore app
I (303) cpu_start: Pro cpu up.
W (312) clk: esp_perip_clk_init() has not been implemented yet
I (318) cpu_start: Pro cpu start user code
I (319) cpu_start: cpu freq: 96000000 Hz
I (319) cpu_start: Application information:
I (321) cpu_start: Project name:     hello_world
I (327) cpu_start: App version:      1
I (331) cpu_start: Compile time:     Jan 23 2025 10:04:03
I (337) cpu_start: ELF file SHA256:  28a77e5c35bcd34d...
I (343) cpu_start: ESP-IDF:          v5.1.4
I (348) cpu_start: Min chip rev:     v0.0
I (353) cpu_start: Max chip rev:     v0.99
I (358) cpu_start: Chip rev:         v0.1
I (362) heap_init: Initializing. RAM available for dynamic allocation:
I (370) heap_init: At 40814750 len 00038C30 (227 KiB): D/IRAM
I (376) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
I (384) spi_flash: detected chip: generic
I (387) spi_flash: flash io: dio
I (392) sleep: Enable automatic switching of GPIO sleep configuration
I (399) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (407) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (415) sleep_sys_periph: Interrupt Matrix sleep retention initialization
I (423) sleep_sys_periph: HP System sleep retention initialization
I (430) sleep_sys_periph: TEE/APM sleep retention initialization
I (436) sleep_sys_periph: UART sleep retention initialization
I (443) sleep_sys_periph: Timer Group sleep retention initialization
I (450) sleep_sys_periph: IO Matrix sleep retention initialization
I (457) sleep_sys_periph: SPI Mem sleep retention initialization
I (463) sleep_sys_periph: SysTimer sleep retention initialization
I (481) app_start: Starting scheduler on CPU0
I (482) main_task: Started on CPU0
I (482) main_task: Calling app_main()
I (493) pm: Frequency switching config: CPU_MAX: 96, APB_MAX: 96, APB_MIN: 96, Light sleep: ENABLED
I (495) esp_zb_sleep: Enable ieee802154 zigbee light sleep, the wake up source is ESP timer
I (510) phy: phy_version: 230,2, 9aae6ea, Jan 15 2024, 11:17:12
I (512) phy: libbtbb version: 944f18e, Jan 15 2024, 11:17:25
I (516) btbb_init: btbb sleep retention initialization
I (521) ieee802154: ieee802154 mac sleep retention initialization
I (540) main_task: Returned from app_main()
I (542) ESP_ZB_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (543) ESP_ZB_SLEEP: Initialize Zigbee stack
I (550) ESP_ZB_SLEEP: Deferred driver initialization successful
I (553) ESP_ZB_SLEEP: Device started up in factory-reset mode
I (560) ESP_ZB_SLEEP: Start network steering
I (2827) ESP_ZB_SLEEP: Network steering was not successful (status: -1)
I (2828) ESP_ZB_SLEEP: Zigbee can sleep
I (6101) ESP_ZB_SLEEP: Network steering was not successful (status: -1)
I (6152) ESP_ZB_SLEEP: Zigbee can sleep
I (9420) ESP_ZB_SLEEP: Network steering was not successful (status: -1)
I (9471) ESP_ZB_SLEEP: Zigbee can sleep
I (9549) ESP_ZB_SLEEP: Zigbee can sleep @xieqinan

chshu, 2025-02-08:
@Mushan-bin did you make any change to the [light_sleep](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep) example? Could you test it again with a later IDF version like [v5.4](https://github.com/espressif/esp-idf/releases/tag/v5.4).

xieqinan, 2025-02-17:
@Mushan-bin ，

If you use the default [light_sleep](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep) example, it will attempt to wake up every second to discover a network. Could you try commenting out [this line](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/light_sleep/main/esp_zb_sleepy_end_device.c#L137) or allowing the device to join a network to test again and avoid periodic wake-ups?

closed 2025-03-20 by chshu