espressif/esp-zigbee-sdk #546: NCP Sample. esp_ncp_zb_action_handler never triggered

cheldonec, 2025-01-23:
### Question

The esp_ncp_zb_action_handler handler is not triggered. The binding is successful! Light control from boot btn is successful!
The registration of the handler is provided in the component, but I can't get the data in any way. (esp_ncp_zb.c line 492 - handler, line 538 registration)

I use the following chips:
NCP   - ESP32-H2
HOST - ESP32-C6

Help me please!

### Additional context.

idf_component.yml
`## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zboss-lib: "~1.6.0"
  espressif/esp-zigbee-lib: "~1.6.0"
  espressif/esp-zigbee-ncp:
    version: "1.*"
    override_path: "../../../components/esp-zigbee-ncp"
  ## Required IDF version
  idf:
    version: ">=5.0.0"`
*****************************************************************************************************
**HOST LOG**
[image]
****************************************************************************************************
**NCP LOG**

[image]
****************************************************************************************************

lhespress, 2025-02-05:
@cheldonec I check the examples on main branch, it works fine. the LOG is [Host.log](https://github.com/user-attachments/files/18670046/Host.log), [Ncp.log](https://github.com/user-attachments/files/18670045/Ncp.log), [HA_on_off_light.log](https://github.com/user-attachments/files/18670057/HA_on_off_light.log). What do i missing something? you can also check it again use the firmware [issues_546_0205.zip](https://github.com/user-attachments/files/18670188/issues_546_0205.zip) which use pin as follows:
**HOST**
```
CONFIG_HOST_BUS_UART_RX_PIN=4
CONFIG_HOST_BUS_UART_TX_PIN=5
```
**NCP**
```
CONFIG_NCP_BUS_UART_RX_PIN=5
CONFIG_NCP_BUS_UART_TX_PIN=4
```

cheldonec, 2025-02-05:
The real device sends a report when the status changes. I expect the function to be called in the NCP modules.

***********************************************************************************************************
[image]
***********************************************************************************************************
I wait (ESP_LOGI(TAG, "Reveived report from address(0x%x) src endpoint(%d) to dst endpoint(%d) cluster(0x%x)", message->src_address.u.short_addr,
             message->src_endpoint, message->dst_endpoint, message->cluster);)

lhespress, 2025-02-06:
@cheldonec It need call `esp_zb_zcl_report_attr_cmd_req` API in your host project which doesn't provide in current host example.

cheldonec, 2025-02-06:
I assumed that this call should happen in the ActionHandler. ApsIndication is work!!!!! ActionHandler not(((

cheldonec, 2025-02-06:
@lhespress I've tried with real devices that send reports. My question is not about the end device example, but about NCP. My project got stuck because of this issue. If you enable ApsIndication, reports arrive, but problems start with adding devices. Maybe I'm doing something wrong? But the impression remains that NCP is not working correctly.

lhespress, 2025-02-07:
@cheldonec I mean you should develop `ESP_NCP_ZCL_ATTR_REPORT` frame ID for send reports in your host.

cheldonec, 2025-02-07:
@lhespress 
I think we're talking about different things.
ESP_NCP_ZCL_ATTR_REPORT frame ID is already implemented in the NCP library.

https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-ncp/src/esp_ncp_zb.c

`static esp_err_t esp_ncp_zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message)
{
    esp_err_t ret = ESP_OK;
    esp_ncp_header_t ncp_header = { 
        .sn = esp_random() % 0xFF,
    };
    uint8_t *output = NULL;
    uint16_t outlen = 0;

    switch (callback_id) {
        case ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID:
            ncp_header.id = ESP_NCP_ZCL_ATTR_READ;
            ret = esp_ncp_zb_read_attr_resp_handler((esp_zb_zcl_cmd_read_attr_resp_message_t *)message, &output, &outlen);
            break;
        case ESP_ZB_CORE_CMD_WRITE_ATTR_RESP_CB_ID:
            ncp_header.id = ESP_NCP_ZCL_ATTR_WRITE;
            ret = esp_ncp_zb_write_attr_resp_handler((esp_zb_zcl_cmd_write_attr_resp_message_t *)message, &output, &outlen);
            break;
        case ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID:
            ncp_header.id = ESP_NCP_ZCL_REPORT_CONFIG;
            ret = esp_ncp_zb_report_configure_resp_handler((esp_zb_zcl_cmd_config_report_resp_message_t *)message, &output, &outlen);
            break;
        case ESP_ZB_CORE_CMD_DISC_ATTR_RESP_CB_ID:
            ncp_header.id = ESP_NCP_ZCL_ATTR_DISC;
            ret = esp_ncp_zb_disc_attr_resp_handler((esp_zb_zcl_cmd_discover_attributes_resp_message_t *)message, &output, &outlen);
            break;
        case ESP_ZB_CORE_REPORT_ATTR_CB_ID:
            ncp_header.id = ESP_NCP_ZCL_ATTR_REPORT;
            ret = esp_ncp_zb_report_attr_handler((esp_zb_zcl_report_attr_message_t *)message, &output, &outlen);
            break;
        default:
            ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
            break;
    }

    if (output) {
        esp_ncp_noti_input(&ncp_header, output, outlen);
        free(output);
        output = NULL;
    }

    return ret;
}`

This function is not called!!!

lhespress, 2025-02-07:
@cheldonec Please git apply the patch [report_0207.zip](https://github.com/user-attachments/files/18706060/report_0207.zip) and check it use `esp_zigbee_host, esp_zigbee_ncp` and `HA_on_off_light` examples.

When click `HA_on_off_light` boot button, the NCP output LOG as follows :
```
I (44164) ESP_NCP_ZB: Reveived report from address(0xd12a) src endpoint(10) to dst endpoint(1) cluster(0x6)
I (44164) ESP_NCP_ZB: Received report information: attribute(0x0), type(0x10), value(0)

I (45334) ESP_NCP_ZB: Reveived report from address(0xd12a) src endpoint(10) to dst endpoint(1) cluster(0x6)
I (45334) ESP_NCP_ZB: Received report information: attribute(0x0), type(0x10), value(0)

I (46324) ESP_NCP_ZB: Reveived report from address(0xd12a) src endpoint(10) to dst endpoint(1) cluster(0x6)
I (46324) ESP_NCP_ZB: Received report information: attribute(0x0), type(0x10), value(0)
```

cheldonec, 2025-02-07:
@lhespress 
I'm sorry, I do not know how?) How to use [report_0207.zip](https://github.com/user-attachments/files/18706060/report_0207.zip)

cheldonec, 2025-02-07:
Sorry with my back vocal))) I understand some problemes. 
When receiving non-standard responses (commands), actionhandler does not work, this is already an axiom!
My (tuya, not standart) button sends the FD command to cluster 0x0006 and at the same time the espressif__esp-zigbee-lib library is silent and does not report errors, but if we turn to the espressif__esp-zboss-lib library, then everything falls into place.

lhespress, 2025-02-08:
@cheldonec PLS do it step by step as follows:

1. download the patch
2. copy it to the `esp-zigbee-sdk` path
3. cd esp-zigbee-sdk
4. unzip report_0207.zip
5. git apply report.patch
6. compile [esp_zigbee_host](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_host) and download it to ESP32C6
7. compile [esp_zigbee_ncp](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_ncp) and download it to one ESP32H2 which connect with ESP32C6
8. compile [HA_on_off_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light) and download it to the other ESP32H2 which work on as End Device
9. wait for the End device join the network, then click its boot button, NCP will output LOG:

```
I (44164) ESP_NCP_ZB: Reveived report from address(0xd12a) src endpoint(10) to dst endpoint(1) cluster(0x6)
I (44164) ESP_NCP_ZB: Received report information: attribute(0x0), type(0x10), value(0)
```

cheldonec, 2025-02-09:
@lhespress Thank you very much.

cheldonec, 2025-02-09:
@lhespress Thanks for the help.

closed 2025-02-09 by cheldonec