espressif/esp-zigbee-sdk #547: Thermostat cluster issue, not able to connect with SmartLife app.

luisyague, 2025-01-24:
### Question

**Conditions:** 
**ESP-IDF Version:** 5.3.1
**esp-zigbee-lib:** 1.6.2
**esp-zboss-lib:** 1.6.2
**Board:** custom (ESP32C6)

Hello, I am trying to develop a Zigbee thermostat which can work with smart life app. I have tested that I can connect correctly to the Gateway (a generic one from AliexPress) and the device is recognized as a Thermostat. However, as soon as I add the thermostat cluster the app stops recognizing the device. Unfortunately the internal implementation of the function `esp_zb_cluster_list_add_thermostat_cluster` is no public so I can not see which setting is failing. I have two threads runnning with a priority of 1 doing the pertinent delays, i have tested that it still fails without creating those threads. 

Is there any other setting I have to do? Thanks in advance. 

This is the code I am using: 
`#include "Zigbee/Zigbee.h"
#include "esp_zigbee_core.h"
#include "esp_zigbee_trace.h"
#include "esp_zigbee_cluster.h"
#include "nvs_flash.h"
#include "string.h"
#include "esp_check.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"

#define TAG "ZIGBEE"

// Device and cluster definitions
#define DEFAULT_KEPP_ALIVE_INTERVAL 3000

// Data types
#define MANUFACTURER_NAME  "\x09""ESPRESSIF"
#define MODEL_IDENTIFIER   "\x07"CONFIG_IDF_TARGET
#define ENDPOINT_ID        (0x01)

// Private variables
static esp_zb_attribute_list_t *basicCluster;
static esp_zb_attribute_list_t *identifyServerCluster;
static esp_zb_attribute_list_t *identifyClientCluster;
static esp_zb_attribute_list_t *groupsCluster;
static esp_zb_attribute_list_t *thermostatCluster;

// Private function declaration
static void espZbTask(void *pvParameters);
static void InitializeZigbeeStack(void);
static void CreateThermostatEndpoint(void);
static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask);
static esp_err_t ZigbeeActionHandler(esp_zb_core_action_callback_id_t callback_id, const void *message);
static void CreateBasicCluster(void);
static void CreateIdentityCluster(void);
static void CreateGroupsCluster(void);
static void CreateThermostatCluster(void);
static void CreateEndPoint(void);

// Public functions
void ZigbeeInit() {
  ESP_LOGI(TAG, "Initializing Zigbee stack");

  esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_INFO, ESP_ZB_TRACE_SUBSYSTEM_MAC | ESP_ZB_TRACE_SUBSYSTEM_APP);

  esp_zb_platform_config_t config = {
      .radio_config = {
        .radio_mode = ZB_RADIO_MODE_NATIVE,
      },
      .host_config = {
        .host_connection_mode = ZB_HOST_CONNECTION_MODE_NONE,
      },
  };

  ESP_ERROR_CHECK(nvs_flash_init());
  ESP_ERROR_CHECK(esp_zb_platform_config(&config));
[... 209 lines cut ...]

void bdb_start_top_level_commissioning_cb(uint8_t mode_mask){
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_level_commissioning(mode_mask) == ESP_OK, ,
                        TAG, "Failed to start Zigbee bdb commissioning");
}

esp_err_t ZigbeeActionHandler(esp_zb_core_action_callback_id_t callback_id, const void *message) {
  esp_err_t ret = ESP_OK;
  switch (callback_id) {
    case ESP_ZB_CORE_REPORT_ATTR_CB_ID:
      ESP_LOGI(TAG, "ESP_ZB_CORE_REPORT_ATTR_CB_ID");
      break;
    case ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID:
      ESP_LOGI(TAG, "ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID");
      break;
    case ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID:
      ESP_LOGI(TAG, "ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID");
      break;
    default:
      ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
      break;
  }
  return ret;
}
`

xieqinan, 2025-02-06:
I think the thermostat cluster is created by you correctly. The `esp_zb_thermostat_ep_create()` function will generate a standard thermostat endpoint list. Could you try registering it directly to the device? I suspect that the Smart Life app will recognize these attributes in the thermostat cluster.

luisyague, 2025-02-10:
Hi @xieqinan , thanks for the response. What do you mean by "installing it directly to the device"? I am trying to install it directly from the SmartLife app.

dzungpv, 2025-03-11:
@luisyague , I see that you set it up as an end device; you could also try a router type. You might want to pair it first with Home Assistant or Zigbee2mqtt to confirm that it is working. 
In my experience, sometimes devices like the Tuya hub/gateway only work with a specific model name or manufacturer name, as they store a dictionary in their firmware.

Additionally, in Zigbee development you can use a Zigbee sniffer to inspect the packets in detail. The hardware can be a USB Zigbee dongle—I used TI because it is widely available. For software, you can use the open-source Wireshark or the trial app “Ubiqua Protocol Analyzer.” You can refer to this guide to learn how to use a sniffer: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

closed 2025-03-20 by chshu