espressif/esp-zigbee-sdk #549: How is esp_zb_identify_notify_handler_register intended to be used?

asward, 2025-01-25:
### Question

Finding `esp_zb_identify_notify_handler_register` in the API I figured registering a callback would automatically route identify commands for the endpoint to the callback. But this doesn't seem to be the case. 

Further, when I register a callback with `esp_zb_raw_command_handler_register`, the identify command is seen on every request. If I return `false` from this callback (unhandled) the command appears to be then routed to the callback registered in `esp_zb_core_action_handler_register` - but only on the first identify request for that boot. If I call `zb_zcl_send_default_handler` and return `true`, the core handler is not called.

The affect is the same if I do not register a raw command handler at all (single identify command received at core callback per boot).

xieqinan, 2025-02-17:
Hi,

To verify your issue, I have added the following patch to the esp-zigbee-sdk, and the light example works fine with Home Assistant. Could you please point out the differences in your application?

```c
diff --git a/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c b/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c
index a402971..47b5c58 100644
--- a/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c
+++ b/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c
@@ -124,6 +124,11 @@ static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id,
     return ret;
 }
 
+static void zb_identify_notify_handler(uint8_t identify_on)
+{
+    printf("identify: %d\n", identify_on);
+}
+
 static void esp_zb_task(void *pvParameters)
 {
     /* initialize Zigbee stack */
@@ -139,6 +144,7 @@ static void esp_zb_task(void *pvParameters)
     esp_zcl_utility_add_ep_basic_manufacturer_info(esp_zb_on_off_light_ep, HA_ESP_LIGHT_ENDPOINT, &info);
     esp_zb_device_register(esp_zb_on_off_light_ep);
     esp_zb_core_action_handler_register(zb_action_handler);
+    esp_zb_identify_notify_handler_register(HA_ESP_LIGHT_ENDPOINT, zb_identify_notify_handler);
     esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
     ESP_ERROR_CHECK(esp_zb_start(false));
     esp_zb_stack_main_loop();
``` 
The log would be like:
```
I (7732) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
identify: 1
I (8742) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (9752) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (10762) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (11772) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (12782) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
identify: 0
I (13792) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (16992) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
identify: 1
I (18002) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (19012) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (20022) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (21032) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (22042) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
identify: 0
I (23052) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (24332) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
identify: 1
I (25342) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (26352) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (27362) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (28372) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
I (29382) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
identify: 0
I (30392) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x3), attribute(0x0), data size(2)
```

chshu, 2025-03-20:
@asward please reopen if any follow up questions.

closed 2025-03-20 by chshu