espressif/esp-zigbee-sdk #550: GPIO Wake has no affect on example

asward, 2025-01-26:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### esp-zigbee-lib version.

1.6.2

### esp-zboss-lib version.

1.6.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I expected to be able to modify the sleep example to use `gpio_wakeup_enable` and `esp_sleep_enable_gpio_wakeup`.

### What is the actual behavior?

It doesn't wake up on GPIO.

### Steps to reproduce.

1. Clone example code (https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep)
2. Modify

```diff
80,81c80,81
<         ESP_ERROR_CHECK(esp_sleep_enable_ext1_wakeup(1ULL << CONFIG_GPIO_INPUT_IO_WAKEUP, ESP_EXT1_WAKEUP_ANY_LOW));
<  
---  
>         ESP_ERROR_CHECK(gpio_wakeup_enable(CONFIG_GPIO_INPUT_IO_WAKEUP, GPIO_INTR_LOW_LEVEL)) ;
>         ESP_ERROR_CHECK(esp_sleep_enable_gpio_wakeup()) ;
```

3. `idf.py flash monitor`
4. Toggle `CONFIG_GPIO_INPUT_IO_WAKEUP` (defaults to 7 for C6)
5. No wake.

### More Information.

Also observed:

1. With original config the wake reason is always `(0) other`, and randomly `(4) timer`
2. With original config the system wakes constantly (~50-100ms). It's difficult to determine if the ext pin is waking the chip or something else (ref 1 above).
3. With GPIO config it only wakes with `(0) other` but is always at the pre-set ED_KEEP_ALIVE time, which makes me think it's just the timer.
4. Changed to GPIO9 (C6 BOOT button): ext breaks (I think GPIO9 is not an RTC gpio on C6), but GPIO wake also does not work.

xieqinan, 2025-02-06:
Hi @asward ,

This is a system issue and not related to Zigbee. I believe it can be resolved more quickly if you submit an issue [here](https://github.com/espressif/esp-idf/issues).

asward, 2025-02-06:
I will open a ticket in the idf project.

For what it's worth, an older issue on this thread seems to have been resolved in April. https://github.com/espressif/esp-zigbee-sdk/issues/99#issuecomment-2054070134

The `esp_sleep_enable_gpio_wakeup()` functionality works flawlessly without the zigbee sdk.

xieqinan, 2025-02-20:
Hi,

This issue has been discussed [here](https://github.com/espressif/esp-idf/issues/15346). Would you mind closing this one?

closed 2025-03-03 by chshu