espressif/esp-zigbee-sdk #551: Thermostat weekly schedule

kiw-github, 2025-01-28:
### Question

The SDK has a function esp_zb_zcl_thermostat_weekly_schedule_start(), what schedule algorithm does it perform?
⁃ What cluster attributes does it change and use in its work?
⁃ Where does it get the time from?

When changing a cluster attribute according to the schedule, how can I get an event about its change?

If the device has several endpoints from thermostat clusters, how can I specify which endpoint to work with when starting esp_zb_zcl_thermostat_weekly_schedule_start()?

According to the SDK, there is a schedule reading function esp_zb_zcl_thermostat_weekly_schedule_get_next_record(), but there is no function that sets this schedule from a program and not via ZigBee.

When the ESP32 is overloaded, the specified schedule is not saved in NVS. How can I implement saving the schedule?

ANARHIST1984, 2025-02-04:
Same, how I can read schedule from Zigbee SDK?

xieqinan, 2025-02-06:
@kiw-github ,

> The SDK has a function esp_zb_zcl_thermostat_weekly_schedule_start(), what schedule algorithm does it perform?

The `esp_zb_zcl_thermostat_weekly_schedule_start()` function enables the thermostat weekly scheduler to set the next temperature from the schedule table by comparing the transition times.

> ⁃ What cluster attributes does it change and use in its work?

It is only related to the `ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_COOLING_SETPOINT_ID` and `ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID`.

> ⁃ Where does it get the time from?

The time is from the UTC time of system.

> When changing a cluster attribute according to the schedule, how can I get an event about its change?

The callback registered with `esp_zb_core_action_handler_register()` will be triggered by the `ESP_ZB_CORE_THERMOSTAT_VALUE_CB_ID` event.

> If the device has several endpoints from thermostat clusters, how can I specify which endpoint to work with when starting esp_zb_zcl_thermostat_weekly_schedule_start()?

The current SDK (v1.6.2) only supports singleton mode for thermostat cluster.

> According to the SDK, there is a schedule reading function esp_zb_zcl_thermostat_weekly_schedule_get_next_record(), but there is no function that sets this schedule from a program and not via ZigBee.

The `esp_zb_zcl_thermostat_set_weekly_schedule_cmd_req()` function can be used to configure the schedule table.

> When the ESP32 is overloaded, the specified schedule is not saved in NVS. How can I implement saving the schedule?

The thermostat schedule table is required to be saved in the application. If the NVS space is insufficient, you will need to increase its size.

kiw-github, 2025-02-11:
> > When changing a cluster attribute according to the schedule, how can I get an event about its change?
> 
> The callback registered with `esp_zb_core_action_handler_register()` will be triggered by the `ESP_ZB_CORE_THERMOSTAT_VALUE_CB_ID` event.

I don't receive the ESP_ZB_CORE_THERMOSTAT_VALUE_CB_ID event when the temperature changes according to a schedule.
What could be the reason?
> 
> > If the device has several endpoints from thermostat clusters, how can I specify which endpoint to work with when starting esp_zb_zcl_thermostat_weekly_schedule_start()?
> 
> The current SDK (v1.6.2) only supports singleton mode for thermostat cluster.

When can you expect the implementation of a schedule for multiple thermostat enpoints?

xieqinan, 2025-02-13:
@ANARHIST1984 @kiw-github ,

There is a thermostat example that may help you.

[thermostat.zip](https://github.com/user-attachments/files/18776489/thermostat.zip)

> When can you expect the implementation of a schedule for multiple thermostat enpoints?

This feature will not be supported in the recent version. If you have specific requirements, please contact our [Technical Support](https://www.espressif.com/en/contact-us/technical-inquiries).

kiw-github, 2025-02-20:
> [@ANARHIST1984](https://github.com/ANARHIST1984) [@kiw-github](https://github.com/kiw-github) ,
>
> There is a thermostat example that may help you. [...]

I tested your example.
If you set mode_for_seq equal to ESP_ZB_ZCL_THERMOSTAT_WEEKLY_SCHEDULE_MODE_FOR_SEQ_HEAT then the ESP_ZB_CORE_THERMOSTAT_VALUE_CB_ID event does not arrive. If you set mode_for_seq equal to ESP_ZB_ZCL_THERMOSTAT_WEEKLY_SCHEDULE_MODE_FOR_SEQ_COOL or ESP_ZB_ZCL_THERMOSTAT_WEEKLY_SCHEDULE_MODE_FOR_SEQ_BOTH then the event arrive.

The command to clear the thermostat schedule also does not work. When sending this command, records with a day of week value containing the ESP_ZB_ZCL_THERMOSTAT_DAY_OF_WEEK_VACATION bit are not deleted

xieqinan, 2025-02-21:
> The command to clear the thermostat schedule also does not work. When sending this command, records with a day of week value containing the ESP_ZB_ZCL_THERMOSTAT_DAY_OF_WEEK_VACATION bit are not deleted

Yes, since `ESP_ZB_ZCL_THERMOSTAT_DAY_OF_WEEK_VACATION` is not clearly defined in the ZCL specification, it is currently an unsupported bit in the SDK. The example above provides a way to intercept thermostat commands. If you're interested, you can try implementing the thermostat cluster based on your own approach.

ANARHIST1984, 2025-02-22:
> Yes, since `ESP_ZB_ZCL_THERMOSTAT_DAY_OF_WEEK_VACATION` is not clearly defined in the ZCL specification, it is currently an unsupported bit in the SDK. The example above provides a way to intercept [...]

Now we have
Two channels cannot be implemented with the current SDK
The schedule does not work and if you need it to work, make a project similar to Tuya.
Then why should we invest in developing a product using your products if you not only have a limited SDK but also do not try to listen to the wishes of people who actually use it to create commercial products?

xieqinan, 2025-02-24:
Hi @ANARHIST1984 ，

> Then why should we invest in developing a product using your products if you not only have a limited SDK but also do not try to listen to the wishes of people who actually use it to create commercial products?

Sorry for the inconvenience. We are making every effort to optimize the SDK based on user suggestions.

> Now we have
Two channels cannot be implemented with the current SDK

The feature is not supported in the current version (v1.6.4), please review this example: https://github.com/espressif/esp-zigbee-sdk/issues/551#issuecomment-2655298554. It provides a way to intercept the thermostat command if you enable `ENABLE_THERMOSTAT_PRIVILEGE_CMD`.

Additionally, we will include this feature in our release plans. To better understand your project schedule and priorities, please contact our [Technical Support](https://www.espressif.com/en/contact-us/technical-inquiries).

kiw-github, 2025-02-24:
1) Your example using ENABLE_THERMOSTAT_PRIVILEGE_CMD does not show how to generate a response to the ESP_ZB_ZCL_CMD_THERMOSTAT_GET_WEEKLY_SCHEDULE command (or any other command).
**How to send a response to a command?**

2) If I use ESP_ZB_ZCL_THERMOSTAT_WEEKLY_SCHEDULE_MODE_FOR_SEQ_HEAT in your SDK, the ESP_ZB_CORE_THERMOSTAT_VALUE_CB_ID event does not arrive at the specified time.
**Please fix the bugs!**

xieqinan, 2025-02-24:
> If I use ESP_ZB_ZCL_THERMOSTAT_WEEKLY_SCHEDULE_MODE_FOR_SEQ_HEAT in your SDK, the ESP_ZB_CORE_THERMOSTAT_VALUE_CB_ID event does not arrive at the specified time.
Please fix the bugs!

Please provide the context of this issue, including which function is called to trigger `ESP_ZB_CORE_THERMOSTAT_VALUE_CB_ID` and how you configure its parameters. It may be not a bug.

xieqinan, 2025-02-24:
> Your example using ENABLE_THERMOSTAT_PRIVILEGE_CMD does not show how to generate a response to the ESP_ZB_ZCL_CMD_THERMOSTAT_GET_WEEKLY_SCHEDULE command (or any other command).
How to send a response to a command?

Please refer to the code below. Note that this code focuses on constructing a ZCL Thermostat Get Weekly Schedule Response command. Different ZCL commands SHALL have different ZCL payloads.

``` c
static esp_err_t zb_privilege_cmd_handler(esp_zb_zcl_privilege_command_message_t *message)
{
    esp_err_t ret = ESP_FAIL;
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    if (message->info.dst_endpoint == HA_ESP_THERMOSTAT_ENDPOINT &&
        message->info.cluster == ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT) {
        ESP_LOGI(TAG, "Payload Size: %d", message->size);
        ESP_LOG_BUFFER_HEXDUMP(TAG, message->data, message->size, ESP_LOG_INFO);
        switch (message->info.command.id) {
        case ESP_ZB_ZCL_CMD_THERMOSTAT_GET_WEEKLY_SCHEDULE:
            ret = zb_thermostat_get_weekly_schedule_cmd_handler(message);
            break;
        default:
            ESP_LOGW(TAG, "No implementation");
            break;
        }
    }
    return ret;
}

static esp_err_t zb_thermostat_get_weekly_schedule_cmd_handler(esp_zb_zcl_privilege_command_message_t *message)
{
    esp_err_t ret = ESP_OK;
    typedef struct zb_thermostat_weekly_schedule_cmd_header_s {
        uint8_t num_of_trans;
        uint8_t day_of_week;
        uint8_t mode;
    } ESP_ZB_PACKED_STRUCT zb_thermostat_weekly_schedule_cmd_header_t;

    typedef struct zb_thermostat_weekly_schedule_cmd_payload_s {
        uint16_t trans_time;
        int16_t heat;
        int16_t cool;
    } ESP_ZB_PACKED_STRUCT zb_thermostat_weekly_schedule_cmd_payload_t;

    typedef struct zb_thermostat_get_weekly_schedule_cmd_header_s {
        uint8_t days_to_return;
        uint8_t mode_to_return;
    } ESP_ZB_PACKED_STRUCT zb_thermostat_get_weekly_schedule_cmd_header_t;

    typedef struct zb_zcl_cmd_header_s {
        uint8_t frame_type : 2;
        uint8_t manuf_spec : 1;
        uint8_t dir : 1;
        uint8_t disable_def_resp : 1;
        uint8_t reserved : 3;
        uint8_t tsn;
        uint8_t cmd;
    } ESP_ZB_PACKED_STRUCT zb_zcl_cmd_header_t;

    zb_thermostat_get_weekly_schedule_cmd_header_t *req_header;

    req_header = (zb_thermostat_get_weekly_schedule_cmd_header_t *)message->data;

    ESP_LOGI(TAG, "get weekly day: %d", req_header->days_to_return);
    ESP_LOGI(TAG, "get weekly mode: %d", req_header->mode_to_return);

    if (req_header->mode_to_return == 0) {
        return ESP_OK;
    }
    /* If req_header->mode_to_return = 1 or req_header->mode_to_return = 2, based on the spec the heat or cool should be
     * deleted from the structure */
    assert(req_header->mode_to_return == 0x03);
    uint8_t *payload =
        (uint8_t *)malloc(sizeof(zb_zcl_cmd_header_t) + sizeof(zb_thermostat_weekly_schedule_cmd_header_t) +
                          sizeof(zb_thermostat_weekly_schedule_cmd_payload_t) * 1);
    uint8_t *ptr = payload;
    zb_zcl_cmd_header_t *payload_hdr = (zb_zcl_cmd_header_t *)ptr;
    payload_hdr->frame_type = 1;
    payload_hdr->manuf_spec = 0;
    payload_hdr->dir = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI;
    payload_hdr->disable_def_resp = 1;
    payload_hdr->reserved = 0;
    payload_hdr->tsn = message->info.header.tsn;
    payload_hdr->cmd = ESP_ZB_ZCL_CMD_THERMOSTAT_GET_WEEKLY_SCHEDULE_RESP;
    ptr += sizeof(zb_zcl_cmd_header_t);
    zb_thermostat_weekly_schedule_cmd_header_t *payload_info = (zb_thermostat_weekly_schedule_cmd_header_t *)ptr;
    payload_info->day_of_week = req_header->days_to_return;
    payload_info->mode = req_header->mode_to_return;
    payload_info->num_of_trans = 1;
    ptr += sizeof(zb_thermostat_weekly_schedule_cmd_header_t);
    zb_thermostat_weekly_schedule_cmd_payload_t *payload_trans = (zb_thermostat_weekly_schedule_cmd_payload_t *)ptr;
    payload_trans->trans_time = 1;
    payload_trans->cool = 2100;
    payload_trans->heat = 2500;

    esp_zb_apsde_data_req_t req;
    req.dst_addr_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    req.dst_addr.addr_short = message->info.src_address.u.short_addr;
    req.dst_endpoint = message->info.src_endpoint;
    req.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
    req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT;
    req.src_endpoint = message->info.dst_endpoint;
    req.asdu_length = (sizeof(zb_zcl_cmd_header_t) + sizeof(zb_thermostat_weekly_schedule_cmd_header_t) +
                       sizeof(zb_thermostat_weekly_schedule_cmd_payload_t) * 1);
    req.asdu = payload;
    req.radius = 2;
    req.tx_options = (ESP_ZB_APSDE_TX_OPT_ACK_TX | ESP_ZB_APSDE_TX_OPT_FRAG_PERMITTED);
    req.use_alias = false;
    ESP_LOGI(TAG, "Get weekly schedule response");
    ret = esp_zb_aps_data_request(&req);
    free(payload);
    return ret;
}

```

closed 2025-03-20 by chshu