espressif/esp-zigbee-sdk #552: Adding action handler register for esp_zigbee_host example

czagaadam, 2025-01-30:
Hi All,

I'm developing a zigbee device (ESP32S3 host + ESP32H2 zcp) with I/O control functionality.
I started from **esp_zigbee_host** example on **S3**.
The **H2** is running the **zcp** example without midification.
I think this example represents how to handle a button:
When GPIO_INPUT_IO_TOGGLE_SWITCH pin is toggled a zb command is sent.

[image]

My question is: how to implement the backward situation
So when a light on/off command is received how to handle it with the S3 host program, I just want to toggle an output.
I've found this example: HA_on_off_light 
In this example the handle registered by the **esp_zb_core_action_handler_register**
but this function is not available in S3 host device example and the signature is missing from esp_zigbee_core

[image]

HA_on_off_light example:
[image]

Maybe the issue is this HA_on_off_light is for H2 but I don't know.

Thanks for supporting me!

chshu, 2025-02-08:
@czagaadam With ESP32-S3 + ESP32-H2 setup, we recommend the Gateway+RCP mode, you can check the [esp_zigbee_gateway](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway) example.

The Host + NCP mode is used for 3rd party host SoC, which introduces complex communication logic between Host SoC and 15.4 SoC.

czagaadam, 2025-02-18:
@chshu the issue with gateway example is that it is working when you are forming a coordinator device.
I want to utilize an end device because the coordinator will be a sonoff zigbee dongle interfaced with an rpi.
Actullay I moved forward to try esp_zb_light example but is also not working.
I'm getting continously I (545840) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL) error.

lhespress, 2025-08-27:
@czagaadam Please `git apply` the [host-ncp-light.zip](https://github.com/user-attachments/files/22006750/host-ncp-light.zip) patch to the `~/path/esp-zigbee-sdk` and check again which support mutual control by click BOOT button.