espressif/esp-zigbee-sdk #555: Getting lag in ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID

mundevx, 2025-02-04:
### Question

I connected Home Assistant to zigbee network. All my Espressif Zigbee nodes are getting lag in receiving data from home assistant. Sometimes i am getting irregular "zb_attribute_handler" callback data receptions.

I compare my Espressif Zigbee light node with GLEDOPTO Zigbee light. The GLEDOPTO Zigbee light is continuously running without any lag with home assistant. 

I then checked simple Zigbee Example  "examples/esp_zigbee_HA_sample/HA_on_off_light", my issue is same.

What will be the issue? Can anyone please let me know?

mundevx, 2025-02-04:
I am getting this lag and data skip in "zb_attribute_handler" when this message "ZDO signal: NLME Status Indication (0x32), status: ESP_OK" displayed on Terminal .

mundevx, 2025-02-04:
On printing printf("%s, status: 0x%x\n", esp_zb_zdo_signal_to_string(sig_type), *(uint8_t *)esp_zb_app_signal_get_params(p_sg_p));
I am getting signal status "ZB_NWK_COMMAND_STATUS_PARENT_LINK_FAILURE". What will be the cause? My zigbee device is very near to gateway Home Assistant.

mundevx, 2025-02-06:
Hi @xieqinan

Any update?

xieqinan, 2025-02-06:
The `ZB_NWK_COMMAND_STATUS_PARENT_LINK_FAILURE` indicates that the end device is expired and cannot maintain a connection with its parent. I have tested the `examples/esp_zigbee_HA_sample/HA_on_off_light` example with Home Assistant, and it works fine without encountering this issue. It would be helpful if you could provide an example to reproduce the issue along with the `.pcap` file for further investigation.

mundevx, 2025-02-06:
My example code is same as "examples/esp_zigbee_HA_sample/HA_on_off_light". I am testing it with Home Assistant and i am getting this message not at all time but getting when attribute callback executed. 
When home assistant send some command to turn on/off light, then Zigbee attribute callback executed. Then sometime i got this message and got stuck for some time for callback. Means there is lag between data send by home assistant and data received by zigbee callaback.

xieqinan, 2025-02-17:
Please try to capture the communication message by the sniffer, it can let's know the root of this issue rather than discussing the surface phenomenon.

mundevx, 2025-02-21:
[homeassistant_espressif.zip](https://github.com/user-attachments/files/18901729/homeassistant_espressif.zip)

@xieqinan , 

The devices with short address 0x8ad3,  0x5787, 0x74ef, 0xd18d are creating lag and displaying signal code 0x09.
I attached the .pcapng file.

mundevx, 2025-02-21:
[new_capture_0x0b79.zip](https://github.com/user-attachments/files/18902350/new_capture_0x0b79.zip)
@xieqinan ,

Above capture file is for a single device 0x0b79. This can be helpful for you.

mundevx, 2025-02-22:
Hi @xieqinan , Any update please.

mundevx, 2025-02-25:
Zigbee Network Key: A6 C3 89 89 0A 5D 69 47 EF 92 2A 55 CF 2F FA 5A

mundevx, 2025-02-25:
@xieqinan ,

Read by function "esp_zb_secur_primary_network_key_get"
Zigbee Network Key: A6 C3 89 89 0A 5D 69 47 EF 92 2A 55 CF 2F FA 5A

Also pre-configured 64-bit encryption keys entered in Wireshark are
Zigbee Alliance: 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39
Nordic Keys: ab:cd:ef:01:23:45:67:89:00:00:00:00:00:00:00:00

If your network key is not listed here, then let me know please.

xieqinan, 2025-02-25:
Hi,

> Means there is lag between data send by home assistant and data received by zigbee callaback.

Based on the Wireshark analysis, the issue may be caused by a large number of route requests generated by Home Assistant. It appears that HA's neighbor table size is limited to 19, and it is filled with the around devices, it cannot store additional network addresses of newly joined devices. As a result, HA has to send a route request each time it attempts to communicate with these devices. Please double-check this.

mundevx, 2025-02-25:
Hi @xieqinan ,

Thanks for the answer. But actually point to be noted that the other third parties zigbee lights are working fine with no lag that are  connected to same network that Espressif zigbee lights have. 

Now what i have to do to make it proper functional. Please help me for that.

xieqinan, 2025-02-25:
> But actually point to be noted that the other third parties zigbee lights are working fine with no lag that are connected to same network that Espressif zigbee lights have.

First, update the esp-zigbee-sdk to version 1.6.3 or later, as it includes a fix for the unnecessary route request issue. Then, use Wireshark to capture frames. Next, try re-forming the network and ensure that ESP devices join the network before third-party devices. Finally, send  commands from the coordinator to both the ESP device and the third-party device to check if the issue still exists. If it does, please provide the `.pcap` file.

By the way, how long is the lag？

mundevx, 2025-02-25:
> First, update the esp-zigbee-sdk to version 1.6.3 or later, as it includes a fix for the unnecessary route request issue. Then, use Wireshark to capture frames. Next, try re-forming the network and [...]

Thank you very much @xieqinan , Actually sometime i got a lag of maximum 8 seconds and sometimes i got no lag. When i got lag, my code stuck and i got attribute callback after that passed lag time.

mundevx, 2025-02-27:
> esp-zigbee-sdk to version 1.6.3

Hi @xieqinan ,

I updated the esp-zigbee-sdk to version 1.6.3 , update my devices firmware, and result is the same. i am still getting lag.

xieqinan, 2025-02-28:
Please follow the steps: https://github.com/espressif/esp-zigbee-sdk/issues/555#issuecomment-2681755982, if this issue still exists, please share the wireshark `.pcap` here.

mundevx, 2025-03-01:
> First, update the esp-zigbee-sdk to version 1.6.3 or later, as it includes a fix for the unnecessary route request issue. Then, use Wireshark to capture frames. Next, try re-forming the network and [...]

@xieqinan ,

I re-form the network and joined third-party device in the last to Home Assistant.
Then I send command from coordinator to ESP and third party devices. The issue still remains the same.

Espressif devices zb addresses:  0xf1dd, 0x81c2
Third-party device zb address: 0x9c76

I am sharing you the .pacp file. Please find the attachment.

Read by function "esp_zb_secur_primary_network_key_get"
Zigbee Network Key: 89 0A 5D 69 47 EF 92 2A 55 CF 2F FA 5A

Also pre-configured 64-bit encryption keys entered in Wireshark are
Zigbee Alliance: 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39
Nordic Keys: ab:cd:ef:01:23:45:67:89:00:00:00:00:00:00:00:00

[ha_esp_01032025.zip](https://github.com/user-attachments/files/19037174/ha_esp_01032025.zip)

xieqinan, 2025-03-04:
After reviewing the `.pcap` file, the current topology appears to be:  
**Coordinator (0x0000) <-> Third-party device (0x9C76) <-> ESP device (0xF1DD)**  

This means the third-party device is still acting as the parent of the ESP device, forwarding frames for it, and potentially affecting transmission performance.  

Additionally, I have checked that most **On/Off** commands from the coordinator are successfully responded to by the ESP device. However, further testing is needed to determine exactly when and where the lag occurs.  

Given the `ZB_NWK_COMMAND_STATUS_PARENT_LINK_FAILURE` signal, I suspect the issue might stem from the link between the third-party device and the ESP device. They might be using different **keep-alive** mechanisms, which could lead to communication inconsistencies.  Could you also share the **.pcap** file from the commissioning process (network formation and device joining) to further analyze this?  

By the way, to speed up troubleshooting, please try conducting additional tests and debugging on your end. Providing more detailed information will help us resolve the issue more efficiently.

chshu, 2025-03-20:
@mundevx any update on this issue?

mundevx, 2025-03-20:
> [@mundevx](https://github.com/mundevx) any update on this issue?

No, i am still having issue. I am getting ZB_NWK_COMMAND_STATUS_PARENT_LINK_FAILURE  on every gateway, like tuya, home assistant or sonoff. This issue occurs after more than 10 espressif zigbee light devices.