espressif/esp-zigbee-sdk #556: How to bind many EP in 1 device?

VIinhQuang, 2025-02-05:
### Question

I want to ask about how to bind to all endpoints in 1 device, I encountered a problem where I sent bind requests to 2 endpoints but only 1 EP responded.
[image]

[image]

[image]
Is there a way to bind each EP in 1 device?
Many thanks if anyone help me

xieqinan, 2025-02-06:
@VIinhQuang ,

> I want to ask about how to bind to all endpoints in 1 device, I encountered a problem where I sent bind requests to 2 endpoints but only 1 EP responded.

The issue is likely caused by `zdo_info_user_ctx_t test_info_ctx`. The context in `esp_zb_zdo_device_bind_req()` only records the pointer address instead of copying its content. To resolve this, consider using `zdo_info_user_ctx_t *test_info_ctx = (zdo_info_user_ctx_t *)malloc(sizeof(zdo_info_user_ctx_t));` instead.

> status is 0082 is The supplied endpoint was equal to 0x00 or between 0xf1 and 0xff.

It appears that the source endpoint is invalid. Please double-check it.

VIinhQuang, 2025-02-06:
> It appears that the source endpoint is invalid. Please double-check it.

[image]

[image]
I checked and made sure the EP is 1
do you have any other solution? 
thanks for reply

cheldonec, 2025-02-06:
@VIinhQuang  
all manipulation after binding. 

in actionhandler
*******************************************************************************************************************
case ESP_ZB_CORE_CMD_IAS_ZONE_ZONE_STATUS_CHANGE_NOT_ID:
        ESP_LOGW(TAG, "ESP_ZB_CORE_CMD_IAS_ZONE_ZONE_STATUS_CHANGE_NOT_ID(0x%x) callback", callback_id);
        ret = zb_attribute_reporting_ias_zone_change_handler((esp_zb_zcl_ias_zone_status_change_notification_message_t *)message);
        break;
*******************************************************************************************************************
**zb_attribute_reporting_ias_zone_change_handler**

static esp_err_t zb_attribute_reporting_ias_zone_change_handler(const esp_zb_zcl_ias_zone_status_change_notification_message_t *message)
{
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)",
                        message->info.status);

    ESP_LOGI(TAG, "Read attribute ZONE STATUS response: from address(0x%x) src endpoint(%d) to dst endpoint(%d) cluster(0x%x) ZONE STATUS (%d)", 
             message->info.src_address.u.short_addr, message->info.src_endpoint,
             message->info.dst_endpoint, message->info.cluster, message->zone_status);
    if (message->zone_status == 0){
        //if (is_wifi_enable()){
        bool status = false;
        xTaskCreate(https_request_task, "HTTPS_Client", 4096, &status, 5, NULL);
        //}
    } 
    switch ((esp_zb_zcl_ias_zone_zonestatus_t)message->zone_status){
        case ESP_ZB_ZCL_IAS_ZONE_ZONE_STATUS_ALARM1:
            ESP_LOGI(TAG, "PIR ТРЕВОГА ALARM1:");
            //if (is_wifi_enable()){
            bool status = true;
            //xTaskCreate(https_request_task, "HTTPS_Client", 4096, &status, 5, NULL);
            //}
            break;
        case ESP_ZB_ZCL_IAS_ZONE_ZONE_STATUS_ALARM2:
            ESP_LOGI(TAG, "PIR ТРЕВОГА ALARM2:");
            break;
    default:
        //if (is_wifi_enable()) https_post_request_for_switch(false);
        break;
    } 
******************************************
Can not insert correctly))) carefully

cheldonec, 2025-02-06:
@VIinhQuang You don't have to contact all the devices if it's an IAS, the main thing is that they pair with your point, with the 0x0500 cluster. At the same time, they get your ESP_ZB_ZCL_ATTR_IAS_ZONE_IAS_CIE_ADDRESS_ID, I may be mistaken))) but this is one of the conditions for 0x0500 to start sending reports.

VIinhQuang, 2025-02-07:
> [@VIinhQuang](https://github.com/VIinhQuang) You don't have to contact all the devices if it's an IAS, the main thing is that they pair with your point, with the 0x0500 cluster. At the same time, [...]
thanks for reply
do you mean send a Write Attribute command on the IAS Zone server’s CIE_IAS_Address attribute with its IEEE address

[image]
but i receive the callback for write attribute is
[image]
can you give more detailed instructions?
that would be great helpfull

cheldonec, 2025-02-07:
@VIinhQuang It's much easier for me.

When new IAS Zone dev ANNCE in app_signal_handler
1. I call **find_IAS_PIR_sensor(&cmd_req, user_find_cb, NULL);**
*******************************************************************************
esp_zb_zdo_match_desc_req_param_t  cmd_req;
        cmd_req.dst_nwk_addr = dev_annce_params->device_short_addr;
        cmd_req.addr_of_interest = dev_annce_params->device_short_addr;
********************************************************************************
`static void find_IAS_PIR_sensor(esp_zb_zdo_match_desc_req_param_t *param, esp_zb_zdo_match_desc_callback_t user_cb, void *user_ctx)
{
    ESP_LOGI(TAG, "Try to find PIR sensor");
    uint16_t cluster_list[] = {ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE};
    esp_zb_zdo_match_desc_req_param_t on_off_req = {
        .dst_nwk_addr = param->dst_nwk_addr,
        .addr_of_interest = param->addr_of_interest,
        .profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .num_in_clusters = 1,
        .num_out_clusters = 0,
        .cluster_list = cluster_list,
    };
    esp_zb_zdo_match_cluster(&on_off_req, user_cb, (void *)&pir_sensor);
}`

2. **user_find_cb**
************************************************************************************************
`typedef struct pir_sensor_device_params_s {
    esp_zb_ieee_addr_t ieee_addr;
    uint8_t  endpoint;
    uint16_t short_addr;
} pir_sensor_device_params_t;`
*************************************************************************************************
`static void user_find_cb(esp_zb_zdp_status_t zdo_status, uint16_t peer_addr, uint8_t peer_endpoint, void *user_ctx)
{
     if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Found PIR sensor");
        ESP_LOGI(TAG, "Match desc response: status(%d), address(0x%x), endpoint(%d)", zdo_status, peer_addr, peer_endpoint);
        
        /* Store the information of the remote device */
        pir_sensor_device_params_t *sensor = (pir_sensor_device_params_t *)user_ctx;
        sensor->endpoint = peer_endpoint;
        sensor->short_addr = peer_addr;
        esp_zb_ieee_address_by_short(sensor->short_addr, sensor->ieee_addr);

         /* find the active endpoint */
        esp_zb_zdo_active_ep_req_param_t *active_ep_req =(esp_zb_zdo_active_ep_req_param_t *)calloc(sizeof(esp_zb_zdo_active_ep_req_param_t), 1);
        active_ep_req->addr_of_interest = peer_addr;
        esp_zb_zdo_active_ep_req(active_ep_req, ep_cb, active_ep_req);

        

        /* 1. Send binding request to the sensor */
        esp_zb_zdo_bind_req_param_t *bind_req = (esp_zb_zdo_bind_req_param_t *)calloc(sizeof(esp_zb_zdo_bind_req_param_t), 1);
        bind_req->req_dst_addr = peer_addr;

        /* populate the src information of the binding */
        memcpy(bind_req->src_address, sensor->ieee_addr, sizeof(esp_zb_ieee_addr_t));
        bind_req->src_endp = peer_endpoint;
        bind_req->cluster_id = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE;

        /* populate the dst information of the binding */
        bind_req->dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        esp_zb_get_long_address(bind_req->dst_address_u.addr_long);
        bind_req->dst_endp = HA_ESP_LIGHT_ENDPOINT_1;

        ESP_LOGI(TAG, "Request PIR sensor to bind us");
        esp_zb_zdo_device_bind_req(bind_req, bind_cb, bind_req);

        /* 2. Send binding request to self */
        bind_req = (esp_zb_zdo_bind_req_param_t *)calloc(sizeof(esp_zb_zdo_bind_req_param_t), 1);
        bind_req->req_dst_addr = esp_zb_get_short_address();

        /* populate the src information of the binding */
        esp_zb_get_long_address(bind_req->src_address);
        bind_req->src_endp = HA_ESP_LIGHT_ENDPOINT_1;
        bind_req->cluster_id = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE;

        /* populate the dst information of the binding */
        bind_req->dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        memcpy(bind_req->dst_address_u.addr_long, sensor->ieee_addr, sizeof(esp_zb_ieee_addr_t));
        bind_req->dst_endp = peer_endpoint;

        ESP_LOGI(TAG, "Bind PIR sensor");
        esp_zb_zdo_device_bind_req(bind_req, bind_cb, bind_req);
    }
/////////////////////////////////////////////////////////
    
}`
3. **bind_cb**
`
static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
    esp_zb_zdo_bind_req_param_t *bind_req = (esp_zb_zdo_bind_req_param_t *)user_ctx;

    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        /* Local binding succeeds */
        if (bind_req->req_dst_addr == esp_zb_get_short_address()) {
            ESP_LOGI(TAG, "Successfully bind the PIR sensor from address(0x%x) on endpoint(%d)",
                     pir_sensor.short_addr, pir_sensor.endpoint);
        }
        if (bind_req->req_dst_addr == pir_sensor.short_addr) {
            ESP_LOGI(TAG, "The PIR sensor from address(0x%x) on endpoint(%d) successfully binds us",
                     pir_sensor.short_addr, pir_sensor.endpoint);
        }
        free(bind_req);
    }
    else {
        /* Bind failed, maybe retry the binding ? */

        // esp_zb_zdo_device_bind_req(bind_req, bind_cb, bind_req);
    }
}
`

AND IT'S ALL ))  After that, the messages ESP_ZB_CORE_CMD_IAS_ZONE_ZONE_STATUS_CHANGE_NOT_ID will be sent to the actionhandler.: And I wrote about this in previous answers.

VIinhQuang, 2025-02-07:
@cheldonec 
I tried and still can't get the report
i use sensor (https://www.ihorn-tech.com/Product/SensorRF/DoorSwitch/ca6cc01d-6883-4374-b610-433acda21b8e.html)

cheldonec, 2025-02-07:
@VIinhQuang  Try it. It works with a Smart Button with non-standard commands

#include "zboss_api.h"
.........
esp_zb_raw_command_handler_register(raw_command_handler);
........
`bool raw_command_handler(uint8_t bufid)
{
zb_bufid_t zb_buf = bufid;

typedef struct raw_data_ind_s {
    uint8_t status;             /*!< The status of the incoming frame processing, 0: on success */
    uint8_t dst_addr_mode;      /*!< The addressing mode for the destination address used in this primitive and of the APDU that has been received,
                                     refer to esp_zb_aps_address_mode_t */
    uint16_t dst_short_addr;    /*!< The individual device address or group address to which the ASDU is directed.*/
    uint8_t dst_endpoint;       /*!< The target endpoint on the local entity to which the ASDU is directed.*/
    uint8_t src_addr_mode;      /*!< Reserved, The addressing mode for the source address used in this primitive and of the APDU that has been received.*/
    uint16_t src_short_addr;    /*!< The individual device address of the entity from which the ASDU has been received.*/
    uint8_t src_endpoint;       /*!< The number of the individual endpoint of the entity from which the ASDU has been received.*/
    uint16_t profile_id;        /*!< The identifier of the profile from which this frame originated.*/
    uint16_t cluster_id;        /*!< The identifier of the received object.*/
    uint8_t  cmd_id;
    uint32_t data_length;       /*!< The number of octets comprising the ASDU being indicated by the APSDE.*/
    uint8_t *data;              /*!< The set of octets comprising the ASDU being indicated by the APSDE. */
    uint8_t security_status;    /*!< UNSECURED if the ASDU was received without any security. SECURED_NWK_KEY if the received ASDU was secured with the NWK key.*/
    int lqi;                    /*!< The link quality indication delivered by the NLDE.*/
    int rx_time;                /*!< Reserved, a time indication for the received packet based on the local clock */
} raw_data_ind_t;

uint32_t data_len = zb_buf_len(zb_buf);
    uint8_t data[data_len];
    zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(zb_buf, zb_zcl_parsed_hdr_t);
    memcpy(&data, zb_buf_begin(zb_buf), sizeof(data));
    raw_data_ind_t raw_data;
    raw_data.profile_id = (uint16_t)cmd_info->profile_id;
    raw_data.dst_short_addr = (uint16_t)cmd_info->addr_data.common_data.dst_addr;
    raw_data.dst_endpoint  = (uint8_t)cmd_info->addr_data.common_data.dst_endpoint;
    raw_data.src_short_addr = (uint16_t)cmd_info->addr_data.common_data.source.u.short_addr;
    raw_data.src_endpoint = (uint8_t)cmd_info->addr_data.common_data.src_endpoint;
    raw_data.cluster_id = (uint16_t)cmd_info->cluster_id;
    raw_data.cmd_id = (uint8_t)cmd_info->cmd_id;
    raw_data.data_length = data_len;
    memcpy(&raw_data.data, &data, data_len);

    ESP_LOGI(TAG,"RAW_DATA_INDICATION profileId: %04x clusterId: %04x cmdId: 0x%02x srcAddr: %04x srcEndpoint: %02x dstAddr: %04x  dstEndpoint: %02x dataLen: %d",
        raw_data.profile_id, raw_data.cluster_id,raw_data.cmd_id, raw_data.src_short_addr, raw_data.src_endpoint, 
        raw_data.dst_short_addr, raw_data.dst_endpoint, (int)raw_data.data_length);
    ESP_LOG_BUFFER_HEX_LEVEL(TAG, &raw_data.data, raw_data.data_length, ESP_LOG_INFO);
}`

closed 2025-02-17 by VIinhQuang