espressif/esp-zigbee-sdk #557: Add callback for resending the same command id value for cluster on_off

igoshev-media, 2025-02-06:
### Is your feature request related to a problem?

I have a two-key Green Power switch. I was able to connect it to my ESP32-C6 board, and I get reports ( esp_zb_core_action_handler_register ) when pressing the 1-key "ON", when pressing the 2-key "OFF". But I do not receive a report on the repeated key press. When using the esp_zb_raw_command_handler_register function, I see that the repeated data is received. How can I configure the code to receive repeated presses?

`
static esp_err_t zb_attribute_handler(const esp_zb_zcl_set_attr_value_message_t *message)
{
    esp_err_t ret = ESP_OK;
    bool light_state = 0;
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, "TAG", "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, "TAG", "Received message: error status(%d)",
                        message->info.status);
    LOG(ESP_LOG_INFO, "Received message: endpoint(%d), cluster(0x%x), attribute(0x%x), data size(%d)", message->info.dst_endpoint, message->info.cluster,
             message->attribute.id, message->attribute.data.size);
    if (message->info.dst_endpoint == HA_LIGHT_ENDPOINT) {
        if (message->info.cluster == ESP_ZB_ZCL_CLUSTER_ID_ON_OFF) {
            if (message->attribute.id == ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID && message->attribute.data.type == ESP_ZB_ZCL_ATTR_TYPE_BOOL) {
                light_state = message->attribute.data.value ? *(bool *)message->attribute.data.value : light_state;
                LOG(ESP_LOG_INFO, "Light sets to %s", light_state ? "On" : "Off");
            }
        }
    }
    return ret;
}

static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message)
{
    esp_err_t ret = ESP_OK;
    switch (callback_id) {
    case ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID:
        ret = zb_attribute_handler((esp_zb_zcl_set_attr_value_message_t *)message);
        break;
    default:
        LOG(ESP_LOG_WARN, "Receive Zigbee action(0x%x) callback", callback_id);
        break;
    }
    return ret;
}

static void esp_zb_task(void *pvParameters)
{
    esp_zb_cfg_t zb_nwk_cfg = {
        .esp_zb_role = ESP_ZB_DEVICE_TYPE_COORDINATOR,
        .install_code_policy = INSTALLCODE_POLICY_ENABLE,
        .nwk_cfg.zczr_cfg = {
            .max_children = MAX_CHILDREN,
        },
    };
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    esp_zb_zgp_set_skip_gpdf(!ZGP_COMBO_PROXY_ENABLED);
    esp_zb_zgps_set_mapping_table((const esp_zb_zgps_mapping_entry_t **)zgp_mapping_table, &zgp_mapping_table_length);
    esp_zb_zgps_set_communication_mode(ESP_ZB_ZGP_COMMUNICATION_MODE_GROUPCAST_DERIVED);
    esp_zb_zgps_set_commissioning_exit_mode(ESP_ZGP_COMMISSIONING_EXIT_MODE_ON_PAIRING_SUCCESS);
    esp_zb_zgps_set_commissioning_window(180);
    esp_zb_zgps_set_functionality(ESP_ZGP_GPSB_FUNCTIONALITY, ESP_ZGP_ACTIVE_FUNCTIONALITY_ID);
    esp_zb_zgps_set_security_level(ESP_ZB_ZGP_FILL_GPS_SECURITY_LEVEL(
        ESP_ZB_ZGP_SEC_LEVEL_FULL_NO_ENC, ESP_ZB_ZGP_SEC_LEVEL_PROTECTION_WITH_GP_LINK_KEY, ESP_ZB_ZGP_SEC_LEVEL_PROTECTION_DO_NOT_INVOLVE_TC));

    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_endpoint_config_t endpoint_config1 = {
        .endpoint = HA_LIGHT_ENDPOINT,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_REMOTE_CONTROL_DEVICE_ID,
        .app_device_version = 0,
    };

    esp_zb_attribute_list_t *basic_cluster = esp_zb_basic_cluster_create(NULL);
    esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, ESP_MANUFACTURER_NAME);
    esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, ESP_MODEL_IDENTIFIER);
    esp_zb_cluster_list_add_basic_cluster(cluster_list, basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_cluster_list_add_on_off_cluster          (cluster_list, esp_zb_on_off_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config1);
    esp_zb_device_register(ep_list);

    // https://github.com/espressif/esp-zigbee-sdk/issues/21
    esp_zb_secur_link_key_exchange_required_set(false);

    esp_zb_core_action_handler_register(zb_action_handler);
    //esp_zb_raw_command_handler_register(rawCmdHandlerCb);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
    vTaskDelete(NULL);
}`

RAW Data:
cluster id: 0x6, command id: 1, dst_ep: 1, src_ep: 0    <--- There is a report on the switch
I (15154) RAW: bufid: 15 size: 0

cluster id: 0x6, command id: 0, dst_ep: 1, src_ep: 0    <--- There is a report on the switch
I (16904) RAW: bufid: 17 size: 0

cluster id: 0x6, command id: 0, dst_ep: 1, src_ep: 0   <---  This package does not appear in the esp_zb_core_action_handler_register(zb_action_handler) report;

Is it possible to set it to display all received events so that I can use each key separately?

xieqinan, 2025-02-17:
Hi,

> cluster id: 0x6, command id: 0, dst_ep: 1, src_ep: 0 <--- This package does not appear in the esp_zb_core_action_handler_register(zb_action_handler) report;

Could you please provide more detailed information about this command? This will help identify why `zb_action_handler()` was not triggered in this scenario.

igoshev-media, 2025-02-17:
> Hi,
>
>
> Could you please provide more detailed information about this command? This will help identify why `zb_action_handler()` was not triggered in this scenario.

Hi,
It works, but only once, when the switch retransmits the same command_id value, zb_action_handler() does not work. I assume that there is a variable inside that remembers the last state of command_id and is ignored when the same value is received again.

My switch sends the same command_id value when the key is pressed, so it only fires once

xieqinan, 2025-02-18:
@igoshev-media ，

OK, I think I understand the reason. The On/Off cluster (0x0006) has only two states: On or Off. Commands 0 and 1 are used to switch the state to Off and On, respectively. If you send the Off command twice, the first one will be executed, but the second will be considered a redundant operation by the stack. Since the On/Off cluster is already in the Off state, sending another Off command has no effect. This behavior seems reasonable.  

If you expect repeated commands to be reported, please provide more justification. If appropriate, we will consider supporting it.

igoshev-media, 2025-02-18:
> [@igoshev-media](https://github.com/igoshev-media)，
> 
> Хорошо, я думаю, я понимаю причину. Кластер On/Off (0x0006) имеет только два состояния: On или Off. Команды 0 и 1 используются для переключения состояния в Off и On соответственно. Если вы отправите команду Off дважды, первая будет выполнена, но вторая будет считаться стеком избыточной операцией. Поскольку кластер On/Off уже находится в состоянии Off, отправка еще одной команды Off не даст никакого эффекта. Такое поведение кажется разумным.
> 
> Если вы ожидаете, что будут сообщаться о повторных командах, предоставьте больше обоснований. Если это уместно, мы рассмотрим возможность их поддержки.

[image]

I have a switch that has 2 keys, the first key sends only ON, the second only OFF. In this mode, I can not use this switch to control two lamps. It seems to me too redundant to use 2 keys to control only 1 load. I do not know the exact model of these switches, it works on Green Power

igoshev-media, 2025-02-18:
> > [@igoshev-media](https://github.com/igoshev-media)，
> > Хорошо, я думаю, я понимаю причину. Кластер On/Off (0x0006) имеет только два состояния: On или Off. Команды 0 и 1 используются для переключения состояния в Off и On соответственно. Если вы отправите команду Off дважды, первая будет выполнена, но вторая будет считаться стеком избыточной операцией. Поскольку кластер On/Off уже находится в состоянии Off, отправка еще одной команды Off не даст никакого эффекта. Такое поведение кажется разумным.
> > Если вы ожидаете, что будут сообщаться о повторных командах, предоставьте больше обоснований. Если это уместно, мы рассмотрим возможность их поддержки.
> 
> [image]
> 
> [image]
> 
> I have a switch that has 2 keys, the first key sends only ON, the second only OFF. In this mode, I can not use this switch to control two lamps. It seems to me too redundant to use 2 keys to control only 1 load. I do not know the exact model of these switches, it works on Green Power. The device does not use a battery, it generates energy when you press a key

xieqinan, 2025-02-19:
Have you tried switching between pressing the `only On` and `only Off` buttons? I believe `esp_zb_core_action_handler_register()` should trigger the correct callback for you.

>  I can not use this switch to control two lamps. It seems to me too redundant to use 2 keys to control only 1 load. 

I agree. Continuously pressing "Off"  also doesn't seem meaningful in the application.

igoshev-media, 2025-02-19:
> Have you tried switching between pressing the `only On` and `only Off` buttons? I believe `esp_zb_core_action_handler_register()` should trigger the correct callback for you.
>
>
> I agree. Continuously [...]

Hi, 
Yes, when you change the ON/OFF key, events come. But it seems that it is not very convenient to control the lighting this way. I would like to be able to switch the state with only 1 key

closed 2025-02-19 by igoshev-media