espressif/esp-zigbee-sdk #558: Is there a way/example to shut off the zigbee radio?

IgnacioHR, 2025-02-06:
### Question

Hi everyone,

I'm developing an end device based on the `seMetering` cluster  that will be running on batteries.

I'm trying to find strategies to save as much energy as possible. One of the strategies is to have the capacity to turn on/off the zigbee radio. 

The idea is that inside a task the code can decide to (1) turn on Zigbee (2) report some values (3) turn off zigbee. 

Point number (1) and (2) are already implemented but I don't see in the API any mechanism to turn off the zigbee radio. For example:

After `esp_zb_stack_main_loop();` is called. Is there a way to stop it cleanly?
After `esp_zb_start(...)` is called. Is there a corresponding `esp_zb_stop(...)` call? I don't see anything in the code api.

Thanks in advance
Ignacio

### Additional context.

The idea is that the device maintains an internal counter.
The internal counter value is stored on NVM.
The device has a magnetic sensor that will trigger once every 0,01 cubic meters of gas is consumed in my home. I would like to use an interrupt to wakeup the device once this happens. This is the most critical task of the device. Zigbee is not needed for this task at all.
The device has to report the counter value at least once every hour and this is when Zigbee will be started.
I know I can enter deep sleep to shut off radio device but are there other mechanism to do it cleanly by code?

xieqinan, 2025-02-07:
@IgnacioHR ,

> After esp_zb_start(...) is called. Is there a corresponding esp_zb_stop(...) call? I don't see anything in the code api.

The current SDK (v1.6.2) does not support this feature. However, based on your description, I recommend using the sleep feature to achieve energy savings.

IgnacioHR, 2025-02-07:
Thank you for the confirmation. 

Let's close this and create a feature request instead.

closed 2025-02-07 by IgnacioHR