espressif/esp-zigbee-sdk #559: Are interrupts disabled while the zigbee radio is startting? When?

IgnacioHR, 2025-02-06:
### Question

I've noticed that interrupts received when the zigbee device is starting (don't know exacly which one of the calls) GPIO interrupts are ignored. 

Can you confirm this is the case?

Can you tell us exactly between which calls interrupts are disabled?

Thanks in advance

### Additional context.

I've implemented a mechanism that delays Zigbee Initialisation in order to save energy. 

The device has also a "main button" the user can press to interact with the device. There are two possibilities:

Short press: less than 5 seconds, the device zigbee radio is started and values are reported.
Long press: longer than 5 seconds, the device leaves the network and prepares itself to join a network.

The detection of the Main Button press/release is using a GPIO interrupt.
And the 5 seconds delay to detect long press is detected starting a timer when the user press the button.
The timer is cancelled if the user releases the button (that is detected via interrupt)

The problem I'm having is that the user press the main button, the zigbee radio is started (and takes about 2 seconds) if the user release the button the interrupt is not detected so as soon as the 5 seconds timer expires the device consider to leave the network.

I may implement workarounds but I need to know when the interrupts are ignored.

xieqinan, 2025-02-07:
Hi,

Based on your description, it seems that the issue is easily reproducible on your side. Could you please share a minimal reproducible code sample?

IgnacioHR, 2025-02-07:
Hi,

I tried to develop a minimal code sample but it works as expected. Most likely there is a bug in my code so let's close this issue.

My conclusion is that interrupts are not **disabled** while the Zigbee radio is starting.

closed 2025-02-07 by IgnacioHR