espressif/esp-zigbee-sdk #560: Schedule feature in HA_thermostat end device

GangadharShinde-Copeland, 2025-02-07:
### Question

How is time parameter considered in the set schedule command? I understand that it is in minutes from midnight (e.g 06:00 AM: 480min). 
How does sdk know what time is it to change the attributes?
Does sdk automatically synchronize its time with UTC?
Do we need to declare/initialize anything for sdk to trigger attribute change events as per schedule?

lhespress, 2025-02-08:
@GangadharShinde-Copeland 

- `ESP_ZB_ZCL_CLUSTER_ID_TIME` is used for sync its time with coordinator.
- `esp_zb_get_utc_time_callback_register` is used for sync UTC time.
- No, you don't do anything.

GangadharShinde-Copeland, 2025-02-10:
@lhespress 
For set schedule command to work properly, do I need to declare/initialize the time cluster in my application program?

Does sdk itself manage the time and time-zone for the schedule and update the attribute as per schedule?

lhespress, 2025-02-13:
@GangadharShinde-Copeland 

- Yes, you need declare/initialize the time cluster in your application program.
- No, SDK doesn't care the time zone which get real time by `esp_zb_get_utc_time_callback_register` function.

lhespress, 2025-02-13:
@GangadharShinde-Copeland The example as follows:

```
static uint32_t zb_utc_time_handler(void)
{
    return real time;
}

static void esp_zb_task(void *pvParameters)
{
    ......

    esp_zb_get_utc_time_callback_register(zb_utc_time_handler);
    
    ......
}
```

closed 2025-03-20 by chshu