espressif/esp-zigbee-sdk #562: Bind event intercept

grebenyuk51, 2025-02-08:
### Question

In Zigbee2MQTT (Z2M) I can bind one device to another. Is there any possibility to intercept bind request/response?
For clarity add sniffer file. Bind 1-2 endpoint OnOff cluster 0xD498 (ESP32-C6) to 0x0520 (Tuya) 1 endpoint
Pay attention, that Bind Request generated from Coordinator, not  enddevice. In `esp_zb_device_cb_id_handler_register` and `esp_zb_raw_command_handler_register` there is no activity during  binding process. Key: 79:16:28:b9:8c:82:89:10:10:4d:87:9b:2c:0e:01:fc

[github_bind_req.zip](https://github.com/user-attachments/files/18718521/github_bind_req.zip)

lpy4105, 2025-02-11:
Bindings are mananged by the SDK, and developers are not expected to care about the very details. 

BTW, `esp_zb_device_cb_id_handler_register` and `esp_zb_raw_command_handler_register` are for ZCL messages, so you couldn't get the `binding request` message from them.

grebenyuk51, 2025-02-11:
is it possible just to generate some event and pass information about binding request? No additional control, just information

lpy4105, 2025-02-13:
On the requester side, the status could be get from the callback of `esp_zb_zdo_device_bind_req()`.
On the responser side, there is no notification of a successful binding. However, you could use `esp_zb_zdo_binding_table_req()` to query the binding table contents.

closed 2025-02-13 by grebenyuk51