espressif/esp-zigbee-sdk #565: In what file is secret_zll_trust_center_key[] defined?

jaccobezemer, 2025-02-11:
### Question

Hi, I successfully compiled the Zigbee_Dimmable_Light.ino and got it working with Home Assistant ZHA.
Now I want to join it to my Philips HUE Bridge and have to change the secret_zll_trust_center_key[].
Silly question but in what file is the trust center key defined.

xieqinan, 2025-02-13:
This https://github.com/espressif/esp-zigbee-sdk/issues/168#issuecomment-1952370582 may help you.

closed 2025-02-13 by jaccobezemer

wejn, 2025-03-08:
@jaccobezemer https://github.com/espressif/esp-zigbee-sdk/issues/358#issuecomment-2567111040 and https://github.com/wejn/esp32-huello-world/ might be useful. Not for arduino, though.

wejn, 2025-03-08:
Still working on it (was delayed by external circumstances, but hoping to finish in a week or two), so haven't tried running the esp32 for longer than a few hours.