espressif/esp-zigbee-sdk #566: Missing HA_DIMMABLE_LIGHT implementation

jaccobezemer, 2025-02-15:
### Is your feature request related to a problem?

Hi,

I'm missing the HA_DIMMABLE_LIGHT implementation. I see references like ZB_HA_DIMMABLE_LIGHT_DEVICE_ID = 0x0101 but no ESP_ZB_DEFAULT_DIMMABLE_LIGHT_CONFIG() and esp_zb_dimmable_light_clusters_create(esp_zb_dimmable_light_cfg_t *light_cfg);

Can this be implemented? Or is there an other option to create a dimmable light?

xieqinan, 2025-02-27:
Please refer to:
``` c
esp_zb_ep_list_t *esp_zb_dimmable_light_ep_create(zb_uint8_t endpoint_id, esp_zb_color_dimmable_light_cfg_t *light_cfg)
{
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = endpoint_id,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_DIMMABLE_LIGHT_DEVICE_ID,
        .app_device_version = 0,
    };
    esp_zb_basic_cluster_cfg_t *basic_cfg = light_cfg ? &(light_cfg->basic_cfg) : NULL;
    esp_zb_identify_cluster_cfg_t *identify_cfg = light_cfg ? &(light_cfg->identify_cfg) : NULL;
    esp_zb_groups_cluster_cfg_t *groups_cfg = light_cfg ? &(light_cfg->groups_cfg) : NULL;
    esp_zb_scenes_cluster_cfg_t *scenes_cfg = light_cfg ? &(light_cfg->scenes_cfg) : NULL;
    esp_zb_on_off_cluster_cfg_t *on_off_cfg = light_cfg ? &(light_cfg->on_off_cfg) : NULL;
    esp_zb_level_cluster_cfg_t *level_cfg = light_cfg ? &(light_cfg->level_cfg) : NULL;
    // esp_zb_color_cluster_cfg_t *color_cfg = light_cfg ? &(light_cfg->color_cfg) : NULL;
    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(cluster_list, esp_zb_basic_cluster_create(basic_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster_create(identify_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(cluster_list, esp_zb_groups_cluster_create(groups_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(cluster_list, esp_zb_scenes_cluster_create(scenes_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(cluster_list, esp_zb_on_off_cluster_create(on_off_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_level_cluster(cluster_list, esp_zb_level_cluster_create(level_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    // esp_zb_cluster_list_add_color_control_cluster(cluster_list, esp_zb_color_control_cluster_create(color_cfg), ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_add_ep(ep_list, esp_zb_color_dimmable_light_clusters_create(light_cfg), endpoint_config);
    return ep_list;
}

```

jaccobezemer, 2025-02-28:
Thanks xieqinan. Sorry but I have no idea how to fit this in the Color Dimmable Light example. Where and how do I place this code?

jaccobezemer, 2025-03-06:
Forget it, I’ve succeeded

jaccobezemer, 2025-03-09:
@xieqinan I put in your proposed changes but Homeassistant ZHA stil thinks it is a color light.

xieqinan, 2025-03-10:
Please share how you register the ZCL data model in your application.

jaccobezemer, 2025-03-10:
```
esp_zb_ep_list_t *esp_zb_dimmable_light_ep_create(uint8_t endpoint_id, esp_zb_color_dimmable_light_cfg_t *light_cfg)
{
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = endpoint_id,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_DIMMABLE_LIGHT_DEVICE_ID,
        .app_device_version = 1,
    };
    esp_zb_basic_cluster_cfg_t *basic_cfg = light_cfg ? &(light_cfg->basic_cfg) : NULL;
    esp_zb_identify_cluster_cfg_t *identify_cfg = light_cfg ? &(light_cfg->identify_cfg) : NULL;
    esp_zb_groups_cluster_cfg_t *groups_cfg = light_cfg ? &(light_cfg->groups_cfg) : NULL;
    esp_zb_scenes_cluster_cfg_t *scenes_cfg = light_cfg ? &(light_cfg->scenes_cfg) : NULL;
    esp_zb_on_off_cluster_cfg_t *on_off_cfg = light_cfg ? &(light_cfg->on_off_cfg) : NULL;
    esp_zb_level_cluster_cfg_t *level_cfg = light_cfg ? &(light_cfg->level_cfg) : NULL;
    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(cluster_list, esp_zb_basic_cluster_create(basic_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster_create(identify_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(cluster_list, esp_zb_groups_cluster_create(groups_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(cluster_list, esp_zb_scenes_cluster_create(scenes_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(cluster_list, esp_zb_on_off_cluster_create(on_off_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_level_cluster(cluster_list, esp_zb_level_cluster_create(level_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_add_ep(ep_list, esp_zb_color_dimmable_light_clusters_create(light_cfg), endpoint_config);
    return ep_list;
}

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZR_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    // allow joining the Philips Hue network(s) Has no effect in ZHA
    esp_zb_enable_joining_to_distributed(true);
    uint8_t secret_zll_trust_center_key[] = {
        0x81, 0x42, 0x86, 0x86, 0x5D, 0xC1, 0xC8, 0xB2, 0xC8, 0xCB, 0xC5, 0x2E, 0x5D, 0x65, 0xD1, 0xB8
    };
    esp_zb_secur_TC_standard_distributed_key_set(secret_zll_trust_center_key);

    esp_zb_color_dimmable_light_cfg_t light_cfg = ESP_ZB_DEFAULT_COLOR_DIMMABLE_LIGHT_CONFIG();  // in the SDK DIMMABLE LIGHT is not defined 
    esp_zb_ep_list_t *esp_zb_dimmable_light_ep = NULL;
    esp_zb_dimmable_light_ep = esp_zb_dimmable_light_ep_create(HA_DIMMABLE_LIGHT_ENDPOINT,&light_cfg);

    zcl_basic_manufacturer_info_t info = {
        .manufacturer_name = ESP_MANUFACTURER_NAME,
        .model_identifier = ESP_MODEL_IDENTIFIER,
    };
    esp_zcl_utility_add_ep_basic_manufacturer_info(esp_zb_dimmable_light_ep, HA_DIMMABLE_LIGHT_ENDPOINT, &info);

    // for Philips Hue. Has no effect in ZHA
    // https://github.com/espressif/esp-zigbee-sdk/issues/457#issuecomment-2426128314
    uint16_t on_off_on_time = 0;
    bool on_off_global_scene_control = 0;
    esp_zb_cluster_list_t *cluster_list = esp_zb_ep_list_get_ep(esp_zb_dimmable_light_ep, HA_DIMMABLE_LIGHT_ENDPOINT);
    esp_zb_attribute_list_t *onoff_attr_list = esp_zb_cluster_list_get_cluster(cluster_list, ESP_ZB_ZCL_CLUSTER_ID_ON_OFF, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_on_off_cluster_add_attr(onoff_attr_list, ESP_ZB_ZCL_ATTR_ON_OFF_ON_TIME, &on_off_on_time);
    esp_zb_on_off_cluster_add_attr(onoff_attr_list, ESP_ZB_ZCL_ATTR_ON_OFF_GLOBAL_SCENE_CONTROL,&on_off_global_scene_control);    

    esp_zb_device_register(esp_zb_dimmable_light_ep);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
}

```

lpy4105, 2025-03-11:
I believe you should change 
`esp_zb_ep_list_add_ep(ep_list, esp_zb_color_dimmable_light_clusters_create(light_cfg), endpoint_config);` 
to 
`esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config);`

jaccobezemer, 2025-03-11:
@lpy4105 and @xieqinan thanks. That did the trick.

closed 2025-03-11 by jaccobezemer