espressif/esp-zigbee-sdk #567: Zigbee stack assertion failed during leave request

grebenyuk51, 2025-02-16:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### esp-zigbee-lib version.

1.6.3

### esp-zboss-lib version.

1.6.3

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Device leaves from the network after leave request receives

### What is the actual behavior?

```
I (25646) Button boot: Long press, leave
I (26226) Resteering: start steering
ZB_TRACE_LOG[0]: nwk/nwk_main.c:4511    zb_nwk_do_leave param 28 rejoin 0
ZB_TRACE_LOG[0]: common/zb_address.c:854    zb_address_unlock error: ref 44 not locked
ZB_TRACE_LOG[0]: common/zb_address.c:857    has_address_conflict 0
ZB_TRACE_LOG[0]: common/zb_address.c:858    lock_cnt 0, pending_for_delete 0
ZB_TRACE_LOG[0]: common/zb_address.c:859    clock 1
ZB_TRACE_LOG[0]: common/zb_address.c:860    redirect_type 0, redirect_ref 255
ZB_TRACE_LOG[0]: common/zb_address.c:861    short 0xb416, ieee 40:4c:ca:ff:fe:42:b5:b8
ZB_TRACE_LOG[0]: common/zb_debug.c:64    Assertion failed /builds/thread_zigbee/esp-zboss/components/zboss_stack/zboss/common/zb_address.c:863
Zigbee stack assertion failed common/zb_address.c:863

abort() was called at PC 0x42025f95 on core 0
Core  0 register dump:
MEPC    : 0x408007b0  RA      : 0x408076ce  SP      : 0x4081c370  GP      : 0x4080e6e4  
TP      : 0x4081c650  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x4081c3ac  S1      : 0x4081c3ac  A0      : 0x4081c3ac  A1      : 0x4081c38e  
A2      : 0x00000000  A3      : 0x4081c3d9  A4      : 0x00000001  A5      : 0x40816000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x4081c390  S3      : 0x4207fc00  
S4      : 0x018850a2  S5      : 0x00000028  S6      : 0x00000000  S7      : 0x0000002a  
S8      : 0x00000000  S9      : 0x40816000  S10     : 0x420d3000  S11     : 0x420d3000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
4081c370: 0x420d3000 0x40816000 0x4081c3ac 0x4080cde4 0x00000000 0x00000028 0x018850a2 0x42070030
4081c390: 0x32303234 0x35396635 0x420dc100 0x4080f538 0x4081c390 0x4080f554 0x4081c38c 0x726f6261
4081c3b0: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x35323032 0x20353966
4081c3d0: 0x63206e6f 0x2065726f 0x00000030 0x00000000 0x408114e8 0x0000035f 0x420dc191 0x42025f98
4081c3f0: 0x0000035f 0x00000001 0x0000002c 0x42022244 0x420459a8 0x408213cc 0x420dc000 0x42022250
4081c410: 0x00000040 0x0000004c 0x000000ca 0x000000ff 0x000000fe 0x00000042 0x000000b5 0x000000b8
4081c430: 0x4081c4bf 0xb4160000 0xfe42b5b8 0x404ccaff 0x4081da30 0x0000002c 0x000001e4 0x4202257a
4081c450: 0x00000000 0xffffffff 0x00000000 0x4081c47c 0x00000000 0x408114e8 0x018850a2 0x4207fc00
4081c470: 0x4081da30 0x00000004 0x00000005 0x420147f8 0x4081da30 0x408114e8 0x420d7000 0x42014886
4081c490: 0x00000000 0x408114e8 0x0000002c 0x42047064 0x0000b416 0x0000002c 0x00000005 0x4081c4cc
4081c4b0: 0x00000009 0x00000001 0x00000000 0xb416fc00 0x4081da30 0x408114e8 0x0000001c 0x42047268
4081c4d0: 0x00000000 0x00000000 0x0000001c 0x420b06dc 0x00000001 0x00000000 0x4081d164 0x2c81c50c
4081c4f0: 0x00000001 0x00000000 0x0000001c 0x4207fc18 0x00000000 0x00000000 0x01885849 0x4202f876
4081c510: 0x0000001c 0x0000002a 0x00000000 0x00000000 0x4207fc00 0x011c0000 0x00000000 0x404f4b00
4081c530: 0x4080e408 0x4081c658 0x4080e414 0x40822f50 0x420d3000 0x40822814 0x0000fc00 0x00000004
4081c550: 0x4080e408 0x4080e3a8 0x40823074 0x42072b82 0x00000000 0x00000000 0x00000000 0x420109ae
4081c570: 0x0000000c 0x00000003 0x4082308c 0x420109bc 0x4080e408 0x4080e3a8 0x40823074 0x4200a400
4081c590: 0xa5a5a5a5 0x4080e410 0x40822b70 0xa5a5a5a5 0x01a5a5a5 0x00106203 0x65777106 0xa5797472
4081c5b0: 0x001005a0 0x00000040 0x02010404 0x40800000 0x00000001 0x00000000 0x00000020 0x00000000
4081c5d0: 0x0000000b 0x01104c44 0x00000000 0x00000000 0xffffffff 0x00000000 0x00000000 0x00000000
4081c5f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c610: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c630: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081c650: 0xa5a5a5a5 0x00000150 0x4081c250 0x00000a06 0x40810918 0x40810918 0x4081c658 0x40810910
4081c670: 0x00000014 0x40817ee4 0x40817ee4 0x4081c658 0x00000000 0x00000005 0x4081a654 0x6267695a
4081c690: 0x6d5f6565 0x006e6961 0x00000000 0x4081c650 0x00000005 0x00000001 0x00000000 0x00000000
4081c6b0: 0x00000000 0x408167cc 0x40816834 0x4081689c 0x00000000 0x00000000 0x00000001 0x00000000
4081c6d0: 0x00000000 0x00000000 0x42005970 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c6f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c710: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 5cbcb5e63

Rebooting...
```

### Steps to reproduce.

My code is simple:

```
ESP_LOGI("Button boot", "Long press, leave");
esp_zb_zdo_mgmt_leave_req_param_t leave_cmd = {0};
esp_zb_get_long_address(leave_cmd.device_address);
leave_cmd.dst_nwk_addr = 0xffff;
esp_zb_lock_acquire(portMAX_DELAY);
esp_zb_zdo_device_leave_req(&leave_cmd, leave_cb, NULL);
esp_zb_lock_release();

...

void leave_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
	if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS)
	{
		ESP_LOGI("Resteering", "start steering");
		esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
	}
	else
	{
		ESP_LOGE("Leave_cb", "Error=%d", zdo_status);
	}
}
```

What am I doing wrong?

### More Information.

Fun fact! After erasing flash, there is no such error. But after several days of work, this error comes up

xieqinan, 2025-02-17:
Hi,

I think this is a usage issue. Could you try executing `esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);` after receiving the `ESP_ZB_ZDO_SIGNAL_LEAVE` signal, which indicates the device has successfully left?

grebenyuk51, 2025-03-01:
Yes, this helped. Thanks

closed 2025-03-01 by grebenyuk51