espressif/esp-zigbee-sdk #568: Handling Missing Information due to ESP_ZB_NLME_STATUS_INDICATION and Configuring Reports with ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE Cluster

VIinhQuang, 2025-02-17:
### Question

Hi, i have two questions
Question 1: 
About the ESP_ZB_NLME_STATUS_INDICATION. From what I've understood, this signal appears when I send a command to a device but don’t get a response (though I may not fully understand, please correct me if I’m wrong).

The issue here is that I need to send a read_attribute command to the sensor continuously because it only reports back to me when its state changes. However, when the signal appears right as the sensor state changes, I don't receive the report. This causes missing information. I’ve tried, but I can’t configure the sensor to report automatically. Is there a way to resolve this issue, or could I potentially remove the ESP_ZB_NLME_STATUS_INDICATION signal altogether?

Question 2:
Could someone guide me on how to configure reporting with the ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE cluster (door and window sensor of Tuya)?

i have tried config as shown below
[image]

and here is the result and it does not send information automatically when there is a state change

[image]

[image]

I would appreciate your help, and thank you so much for your response!

xieqinan, 2025-02-18:
@VIinhQuang ,

> About the ESP_ZB_NLME_STATUS_INDICATION. From what I've understood, this signal appears when I send a command to a device but don’t get a response (though I may not fully understand, please correct me if I’m wrong).

Please refer to the [comment](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zdo/esp_zigbee_zdo_common.h#L478) for parsing the payload of `ESP_ZB_NLME_STATUS_INDICATION` to understand why the signal is raised first.

> Could someone guide me on how to configure reporting with the ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE cluster (door and window sensor of Tuya)?

The access type of `ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID` is `read_only`, so it cannot be configured for reporting.

VIinhQuang, 2025-02-18:
> Please refer to the [comment](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zdo/esp_zigbee_zdo_common.h#L478) for parsing the payload of to understand why the signal is raised first.`ESP_ZB_NLME_STATUS_INDICATION`

@xieqinan Thanks for rely
In my project, I only encounter the ESP_ZB_NLME_STATUS_INDICATION signal when I send a read_attr request to another device and do not receive a response. Otherwise, this signal rarely appears. Could you guide me on how to resolve this issue?

> The issue here is that I need to send a read_attribute command to the sensor continuously because it only reports back to me when its state changes. However, when the signal appears right as the [...]

> The access type of is , so it cannot be configured for reporting.`ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID``read_only`

With Tuya's gateway, it can automatically report state changes. I'm not sure if there's a way to do the same with ESP-Zigbee. Could you help with that?

xieqinan, 2025-02-19:
> In my project, I only encounter the ESP_ZB_NLME_STATUS_INDICATION signal when I send a read_attr request to another device and do not receive a response. Otherwise, this signal rarely appears. Could you guide me on how to resolve this issue?

Could you please share me with the source code of the `read_attr` request? Let's ensure it is corrected first. If the destination address does not exist, this signal will be raised to indicate it.

> With Tuya's gateway, it can automatically report state changes. I'm not sure if there's a way to do the same with ESP-Zigbee. Could you help with that?

I’m not sure how `TuYa` has implemented it. However, the standard usage of the IAS Zone is defined in the ZCL specification, and the esp-zigbee-sdk follows this standard. You can also refer to the [discussion](https://github.com/espressif/esp-zigbee-sdk/issues/462#issuecomment-2455337059) for more details.

VIinhQuang, 2025-02-19:
@xieqinan 
I solved the problem by using esp_zb_raw_command_handler_register(rawCmdHandlerCb) 
thanks your support
Regarding managing the RSSI of devices, besides extracting it from the header of responses, are there any other methods?

VIinhQuang, 2025-02-19:
> You can also get the rssi of device from the neighbor table, refer to `esp_zb_nwk_get_next_neighbor()`.

many thanks!!!

closed 2025-02-19 by VIinhQuang